awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #860: cmake .. -G "NMake Makefiles" throwing error

AIV-GH, 2022-05-18:
A one liner description about the use case and what you are trying to achieve

Everytime I run this command, I see below error 

C:\KinesisVidRekGH\amazon-kinesis-video-streams-producer-sdk-cpp\build>cmake .. -G "NMake Makefiles"
CMake Error at CMakeLists.txt:8 (project):
  Running

   'nmake' '-?'

  failed with:

   The system cannot find the file specified

CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!
See also "C:/KinesisVidRekGH/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/CMakeOutput.log".

risho-lizhao, 2022-05-19:
You can try installing Visual Studio and installing the v142 Build Tool. Then in x64 Native Tools Command Prompt for VS 2022, executive `git config --system core.longpaths true` and then executive the above command.
Alternatively, you may also need to install perl and openssl.

disa6302, 2022-05-20:
Closing assuming answered.

closed 2022-05-20 by disa6302