awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #861: [QUESTION]Window 10 Build Error (NMAKE : fatal error U1077)

risho-lizhao, 2022-05-19:
Platform: 

- Windows 10 Pro 21H2
- x64 Native Tools Command Prompt for VS 2022(With v142 build tool)
- strawberry-perl-5.32.1.1-64bit
- OpenSSL 1.1.1o

I got error message when run `cmake .. -G "NMake Makefiles" -DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_JNI=TRUE`

**Logging**
```
[ 66%] Performing build step for 'project_libopenssl'

Microsoft(R) Program Maintenance Utility Version 14.32.31328.0
Copyright (C) Microsoft Corporation.  All rights reserved.

        "C:\Strawberry\perl\bin\perl.EXE" "-I." -Mconfigdata "util\dofile.pl"  "-omakefile" "include\crypto\bn_conf.h.in" > include\crypto\bn_conf.h
        "C:\Strawberry\perl\bin\perl.EXE" "-I." -Mconfigdata "util\dofile.pl"  "-omakefile" "include\crypto\dso_conf.h.in" > include\crypto\dso_conf.h
        "C:\Strawberry\perl\bin\perl.EXE" "-I." -Mconfigdata "util\dofile.pl"  "-omakefile" "include\openssl\opensslconf.h.in" > include\openssl\opensslconf.h
        "C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.32.31326\bin\Hostx64\x64\nmake.exe" /                   depend && "C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.32.31326\bin\Hostx64\x64\nmake.exe" /                   _all

Microsoft(R) Program Maintenance Utility Version 14.32.31328.0
Copyright (C) Microsoft Corporation.  All rights reserved.

Microsoft(R) Program Maintenance Utility Version 14.32.31328.0
Copyright (C) Microsoft Corporation.  All rights reserved.

        C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.32.31326/bin/Hostx64/x64/cl.exe  /Zi /Fdossl_static.pdb /Gs0 /GF /Gy /MD /W3 /wd4090 /nologo /O2 /I "." /I "include" -D"L_ENDIAN" -D"OPENSSL_PIC" -D"OPENSSLDIR=\"C:\\amazon-kinesis-video-streams-producer-sdk-cpp\\open-source\\local\"" -D"ENGINESDIR=\"C:\\amazon-kinesis-video-streams-producer-sdk-cpp\\open-source\\local\\lib\\engines-1_1\"" -D"OPENSSL_SYS_WIN32" -D"WIN32_LEAN_AND_MEAN" -D"UNICODE" -D"_UNICODE" -D"_CRT_SECURE_NO_DEPRECATE" -D"_WINSOCK_DEPRECATED_NO_WARNINGS" -D"OPENSSL_USE_APPLINK" -D"NDEBUG"  -c /Foapps\app_rand.obj "apps\app_rand.c"
        C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.32.31326/bin/Hostx64/x64/cl.exe  /Zi /Fdossl_static.pdb /Gs0 /GF /Gy /MD /W3 /wd4090 /nologo /O2 /I "." /I "include" -D"L_ENDIAN" -D"OPENSSL_PIC" -D"OPENSSLDIR=\"C:\\amazon-kinesis-video-streams-producer-sdk-cpp\\open-source\\local\"" -D"ENGINESDIR=\"C:\\amazon-kinesis-video-streams-producer-sdk-cpp\\open-source\\local\\lib\\engines-1_1\"" -D"OPENSSL_SYS_WIN32" -D"WIN32_LEAN_AND_MEAN" -D"UNICODE" -D"_UNICODE" -D"_CRT_SECURE_NO_DEPRECATE" -D"_WINSOCK_DEPRECATED_NO_WARNINGS" -D"OPENSSL_USE_APPLINK" -D"NDEBUG"  /Zs /showIncludes "apps\app_rand.c" 2>&1 > apps\app_rand.d
        C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.32.31326/bin/Hostx64/x64/cl.exe  /Zi /Fdossl_static.pdb /Gs0 /GF /Gy /MD /W3 /wd4090 /nologo /O2 /I "." /I "include" -D"L_ENDIAN" -D"OPENSSL_PIC" -D"OPENSSLDIR=\"C:\\amazon-kinesis-video-streams-producer-sdk-cpp\\open-source\\local\"" -D"ENGINESDIR=\"C:\\amazon-kinesis-video-streams-producer-sdk-cpp\\open-source\\local\\lib\\engines-1_1\"" -D"OPENSSL_SYS_WIN32" -D"WIN32_LEAN_AND_MEAN" -D"UNICODE" -D"_UNICODE" -D"_CRT_SECURE_NO_DEPRECATE" -D"_WINSOCK_DEPRECATED_NO_WARNINGS" -D"OPENSSL_USE_APPLINK" -D"NDEBUG"  -c /Foapps\apps.obj "apps\apps.c"
        C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.32.31326/bin/Hostx64/x64/cl.exe  /Zi /Fdossl_static.pdb /Gs0 /GF /Gy /MD /W3 /wd4090 /nologo /O2 /I "." /I "include" -D"L_ENDIAN" -D"OPENSSL_PIC" -D"OPENSSLDIR=\"C:\\amazon-kinesis-video-streams-producer-sdk-cpp\\open-source\\local\"" -D"ENGINESDIR=\"C:\\amazon-kinesis-video-streams-producer-sdk-cpp\\open-source\\local\\lib\\engines-1_1\"" -D"OPENSSL_SYS_WIN32" -D"WIN32_LEAN_AND_MEAN" -D"UNICODE" -D"_UNICODE" -D"_CRT_SECURE_NO_DEPRECATE" -D"_WINSOCK_DEPRECATED_NO_WARNINGS" -D"OPENSSL_USE_APPLINK" -D"NDEBUG"  /Zs /showIncludes "apps\apps.c" 2>&1 > apps\apps.d
        C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.32.31326/bin/Hostx64/x64/cl.exe  /Zi /Fdossl_static.pdb /Gs0 /GF /Gy /MD /W3 /wd4090 /nologo /O2 /I "." /I "include" -D"L_ENDIAN" -D"OPENSSL_PIC" -D"OPENSSLDIR=\"C:\\amazon-kinesis-video-streams-producer-sdk-cpp\\open-source\\local\"" -D"ENGINESDIR=\"C:\\amazon-kinesis-video-streams-producer-sdk-cpp\\open-source\\local\\lib\\engines-1_1\"" -D"OPENSSL_SYS_WIN32" -D"WIN32_LEAN_AND_MEAN" -D"UNICODE" -D"_UNICODE" -D"_CRT_SECURE_NO_DEPRECATE" -D"_WINSOCK_DEPRECATED_NO_WARNINGS" -D"OPENSSL_USE_APPLINK" -D"NDEBUG"  -c /Foapps\bf_prefix.obj "apps\bf_prefix.c"
        C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.32.31326/bin/Hostx64/x64/cl.exe  /Zi /Fdossl_static.pdb /Gs0 /GF /Gy /MD /W3 /wd4090 /nologo /O2 /I "." /I "include" -D"L_ENDIAN" -D"OPENSSL_PIC" -D"OPENSSLDIR=\"C:\\amazon-kinesis-video-streams-producer-sdk-cpp\\open-source\\local\"" -D"ENGINESDIR=\"C:\\amazon-kinesis-video-streams-producer-sdk-cpp\\open-source\\local\\lib\\engines-1_1\"" -D"OPENSSL_SYS_WIN32" -D"WIN32_LEAN_AND_MEAN" -D"UNICODE" -D"_UNICODE" -D"_CRT_SECURE_NO_DEPRECATE" -D"_WINSOCK_DEPRECATED_NO_WARNINGS" -D"OPENSSL_USE_APPLINK" -D"NDEBUG"  /Zs /showIncludes "apps\bf_prefix.c" 2>&1 > apps\bf_prefix.d
        C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.32.31326/bin/Hostx64/x64/cl.exe  /Zi /Fdossl_static.pdb /Gs0 /GF /Gy /MD /W3 /wd4090 /nologo /O2 /I "." /I "include" -D"L_ENDIAN" -D"OPENSSL_PIC" -D"OPENSSLDIR=\"C:\\amazon-kinesis-video-streams-producer-sdk-cpp\\open-source\\local\"" -D"ENGINESDIR=\"C:\\amazon-kinesis-video-streams-producer-sdk-cpp\\open-source\\local\\lib\\engines-1_1\"" -D"OPENSSL_SYS_WIN32" -D"WIN32_LEAN_AND_MEAN" -D"UNICODE" -D"_UNICODE" -D"_CRT_SECURE_NO_DEPRECATE" -D"_WINSOCK_DEPRECATED_NO_WARNINGS" -D"OPENSSL_USE_APPLINK" -D"NDEBUG"  -c /Foapps\opt.obj "apps\opt.c"
        C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.32.31326/bin/Hostx64/x64/cl.exe  /Zi /Fdossl_static.pdb /Gs0 /GF /Gy /MD /W3 /wd4090 /nologo /O2 /I "." /I "include" -D"L_ENDIAN" -D"OPENSSL_PIC" -D"OPENSSLDIR=\"C:\\amazon-kinesis-video-streams-producer-sdk-cpp\\open-source\\local\"" -D"ENGINESDIR=\"C:\\amazon-kinesis-video-streams-producer-sdk-cpp\\open-source\\local\\lib\\engines-1_1\"" -D"OPENSSL_SYS_WIN32" -D"WIN32_LEAN_AND_MEAN" -D"UNICODE" -D"_UNICODE" -D"_CRT_SECURE_NO_DEPRECATE" -D"_WINSOCK_DEPRECATED_NO_WARNINGS" -D"OPENSSL_USE_APPLINK" -D"NDEBUG"  /Zs /showIncludes "apps\opt.c" 2>&1 > apps\opt.d
        C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.32.31326/bin/Hostx64/x64/cl.exe  /Zi /Fdossl_static.pdb /Gs0 /GF /Gy /MD /W3 /wd4090 /nologo /O2 /I "." /I "include" -D"L_ENDIAN" -D"OPENSSL_PIC" -D"OPENSSLDIR=\"C:\\amazon-kinesis-video-streams-producer-sdk-cpp\\open-source\\local\"" -D"ENGINESDIR=\"C:\\amazon-kinesis-video-streams-producer-sdk-cpp\\open-source\\local\\lib\\engines-1_1\"" -D"OPENSSL_SYS_WIN32" -D"WIN32_LEAN_AND_MEAN" -D"UNICODE" -D"_UNICODE" -D"_CRT_SECURE_NO_DEPRECATE" -D"_WINSOCK_DEPRECATED_NO_WARNINGS" -D"OPENSSL_USE_APPLINK" -D"NDEBUG"  -c /Foapps\s_cb.obj "apps\s_cb.c"
        C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.32.31326/bin/Hostx64/x64/cl.exe  /Zi /Fdossl_static.pdb /Gs0 /GF /Gy /MD /W3 /wd4090 /nologo /O2 /I "." /I "include" -D"L_ENDIAN" -D"OPENSSL_PIC" -D"OPENSSLDIR=\"C:\\amazon-kinesis-video-streams-producer-sdk-cpp\\open-source\\local\"" -D"ENGINESDIR=\"C:\\amazon-kinesis-video-streams-producer-sdk-cpp\\open-source\\local\\lib\\engines-1_1\"" -D"OPENSSL_SYS_WIN32" -D"WIN32_LEAN_AND_MEAN" -D"UNICODE" -D"_UNICODE" -D"_CRT_SECURE_NO_DEPRECATE" -D"_WINSOCK_DEPRECATED_NO_WARNINGS" -D"OPENSSL_USE_APPLINK" -D"NDEBUG"  /Zs /showIncludes "apps\s_cb.c" 2>&1 > apps\s_cb.d
        C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.32.31326/bin/Hostx64/x64/cl.exe  /Zi /Fdossl_static.pdb /Gs0 /GF /Gy /MD /W3 /wd4090 /nologo /O2 /I "." /I "include" -D"L_ENDIAN" -D"OPENSSL_PIC" -D"OPENSSLDIR=\"C:\\amazon-kinesis-video-streams-producer-sdk-cpp\\open-source\\local\"" -D"ENGINESDIR=\"C:\\amazon-kinesis-video-streams-producer-sdk-cpp\\open-source\\local\\lib\\engines-1_1\"" -D"OPENSSL_SYS_WIN32" -D"WIN32_LEAN_AND_MEAN" -D"UNICODE" -D"_UNICODE" -D"_CRT_SECURE_NO_DEPRECATE" -D"_WINSOCK_DEPRECATED_NO_WARNINGS" -D"OPENSSL_USE_APPLINK" -D"NDEBUG"  -c /Foapps\s_socket.obj "apps\s_socket.c"
[... cut ...]
NMAKE : fatal error U1077: '"C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.32.31326\bin\HostX64\x64\nmake.exe"' : ���^�[�� �R�[�h '0x2'
Stop.
CMake Error at dependency/libkvscproducer/kvscproducer-src/CMake/Utilities.cmake:93 (message):
  CMake step for libopenssl failed: 2
Call Stack (most recent call first):
  dependency/libkvscproducer/kvscproducer-src/CMakeLists.txt:69 (build_dependency)

-- Configuring incomplete, errors occurred!
See also "C:/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/CMakeOutput.log".
```

disa6302, 2022-05-20:
@risho-lizhao , 

Not sure what is happening, but can you try the environment that is being used in our CI to see if it resolves your issue?
You can find the scripts here: 
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/.github/build_windows.bat
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/2b343c90bd9c8785d9b2b9ba321f053ebf175b5e/.github/workflows/ci.yml#L268

Another way to debug would be to try building openssl standalone on your system using MSVC and seeing if that builds successfully and then try using KVS build.

risho-lizhao, 2022-05-25:
@disa6302 , 

Thank you for your reply.
I tried building openssl (version 1.1.1i) that using MSVC and builds successfully. Then I build KVS again, but the same error still came out.

Also, I downgraded gstreamer to version 1.16.2 and installed NASM(version 2.15.05) as ci.yml showed.
In addition, I modified the path inside the build_windows.bat and ci.yml according to my computer's environment, and replaced the file downloaded from git. But the same error continues to recur.

Could you tell me which version of openssl you are using?

My build_windows.bat:
```
call "C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Auxiliary\Build\vcvars64.bat"
mkdir build
cd build
cmd.exe /c cmake -G "NMake Makefiles" ..
cmake -G "NMake Makefiles" -DBUILD_TEST=TRUE -DBUILD_GSTREAMER_PLUGIN=TRUE ..
nmake
```

My ci.yml:
```
  windows-msvc:
    runs-on: windows-2022
    env:
      AWS_KVS_LOG_LEVEL: 7
    permissions:
      id-token: write
      contents: read
    steps:
      - name: Clone repository
        uses: actions/checkout@v2
      - name: Install dependencies
        run: |
          choco install nasm strawberryperl
          choco install gstreamer --version=1.16.2
          choco install gstreamer-devel --version=1.16.2 
      - name: Build repository
        run: |
          $env:Path += ';C:\Strawberry\perl\site\bin;C:\Strawberry\perl\bin;C:\Strawberry\c\bin;C:\Program Files\NASM;C:\amazon-kinesis-video-streams-producer-sdk-cpp\open-source\local\lib;C:\amazon-kinesis-video-streams-producer-sdk-cpp\open-source\local\bin'
          git config --system core.longpaths true
          .github/build_windows.bat
      - name: Configure AWS Credentials
        uses: aws-actions/configure-aws-credentials@v1
        with:
          role-to-assume: ${{ secrets.AWS_ROLE_TO_ASSUME }}
          role-session-name: ${{ secrets.AWS_ROLE_SESSION_NAME }}
          aws-region: ${{ secrets.AWS_REGION }}
          role-duration-seconds: 10800
      - name: Run tests
        run: |
          $env:Path += ';C:\Strawberry\perl\site\bin;C:\Strawberry\perl\bin;C:\Strawberry\c\bin;C:\Program Files\NASM;C:\amazon-kinesis-video-streams-producer-sdk-cpp\open-source\local\lib;C:\amazon-kinesis-video-streams-producer-sdk-cpp\open-source\local\bin'
          & "C:\amazon-kinesis-video-streams-producer-sdk-cpp\build\tst\producerTest.exe" 
```

darek-gc, 2022-05-28:
@risho-lizhao [risho-lizhao](https://github.com/risho-lizhao) I faced the same issue. When I opened Visual Code Studio again, the CMake generation started from the beginning and the problem didn't occur.

log from the run with "fatal error LNK1181": 
[log 1 fatal error LNK1181.log](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/8791094/log.1.fatal.error.LNK1181.log)

the log without this particular error:
[log 5 after gstreamer download.log](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/8791095/log.5.after.gstreamer.download.log)

excerpt from the second log:
```
...
1> [CMake] -- Finished building dependencies.
1> [CMake] -- Found PkgConfig: C:/gstreamer/1.0/msvc_x86_64/bin/pkg-config.exe (found version "0.29.2") 
1> [CMake] -- Checking for module 'libkvspicUtils'
1> [CMake] --   No package 'libkvspicUtils' found
1> [CMake] -- Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.19044.
1> [CMake] -- Configuring done
1> [CMake] -- Generating done
1> [CMake] -- Build files have been written to: C:/kin-vid-str-prod/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic
1> [CMake] Microsoft (R) Build Engine 17.0.0+c9eb9dd64 dla platformy .NET Framework
1> [CMake] Copyright (C) Microsoft Corporation. Wszelkie prawa zastrzeżone.
1> [CMake] 
1> [CMake]   Performing update step for 'libkvspic-download'
1> [CMake]   No patch step for 'libkvspic-download'
1> [CMake]   No configure step for 'libkvspic-download'
1> [CMake]   No build step for 'libkvspic-download'
1> [CMake]   No install step for 'libkvspic-download'
1> [CMake]   No test step for 'libkvspic-download'
1> [CMake]   Completed 'libkvspic-download'
1> [CMake] -- Looking for pthread.h
1> [CMake] -- Looking for pthread.h - not found
1> [CMake] -- Found Threads: TRUE  
1> [CMake] -- Found OpenSSL: C:/kin-vid-str-prod/open-source/local/lib/libcrypto.lib (found version "1.1.1g")  
1> [CMake] -- Found CURL: C:/kin-vid-str-prod/open-source/local/lib/cmake/CURL/CURLConfig.cmake (found version "7.68.0-DEV")  
1> [CMake] -- Bitness 64 bits
1> [CMake] -- Found ZLIB: C:/gstreamer/1.0/msvc_x86_64/lib/z.lib (found suitable version "1.2.11", minimum required is "1") 
1> [CMake] -- Checking for module 'gstreamer-app-1.0'
1> [CMake] --   
1> [CMake] CMake Error at C:/Program Files/Microsoft Visual Studio/2022/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.21/Modules/FindPkgConfig.cmake:554 (message):
1> [CMake]   A required package was not found
1> [CMake] Call Stack (most recent call first):
1> [CMake]   C:/Program Files/Microsoft Visual Studio/2022/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.21/Modules/FindPkgConfig.cmake:776 (_pkg_check_modules_internal)
1> [CMake]   CMakeLists.txt:198 (pkg_check_modules)
1> [CMake] -- Configuring incomplete, errors occurred!
1> [CMake] See also "C:/kin-vid-str-prod/out/build/x64-Debug/CMakeFiles/CMakeOutput.log".
1> [CMake] See also "C:/kin-vid-str-prod/out/build/x64-Debug/CMakeFiles/CMakeError.log".
1> Wykonanie operacji „C:\WINDOWS\system32\cmd.exe” „/c "%SYSTEMROOT%\System32\chcp.com 65001 >NUL && "C:\PROGRAM FILES\MICROSOFT VISUAL STUDIO\2022\COMMUNITY\COMMON7\IDE\COMMONEXTENSIONS\MICROSOFT\CMAKE\CMake\bin\cmake.exe"  -G "Ninja"  -DCMAKE_BUILD_TYPE:STRING="Debug" -DCMAKE_INSTALL_PREFIX:PATH="C:\kin-vid-str-prod\out\install\x64-Debug" -DCMAKE_C_COMPILER:FILEPATH="C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.30.30705/bin/Hostx64/x64/cl.exe" -DCMAKE_CXX_COMPILER:FILEPATH="C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.30.30705/bin/Hostx64/x64/cl.exe" -DBUILD_GSTREAMER_PLUGIN=ON -DCMAKE_MAKE_PROGRAM="C:\PROGRAM FILES\MICROSOFT VISUAL STUDIO\2022\COMMUNITY\COMMON7\IDE\COMMONEXTENSIONS\MICROSOFT\CMAKE\Ninja\ninja.exe" "C:\kin-vid-str-prod" 2>&1"” nie powiodło się z powodu błędu: „„C:\WINDOWS\system32\cmd.exe”: element „/c "%SYSTEMROOT%\System32\chcp.com 65001 >NUL && "C:\PROGRAM FILES\MICROSOFT VISUAL STUDIO\2022\COMMUNITY\COMMON7\IDE\COMMONEXTENSIONS\MICROSOFT\CMAKE\CMake\bin\cmake.exe"  -G "Ninja"  -DCMAKE_BUILD_TYPE:STRING="Debug" -DCMAKE_INSTALL_PREFIX:PATH="C:\kin-vid-str-prod\out\install\x64-Debug" -DCMAKE_C_COMPILER:FILEPATH="C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.30.30705/bin/Hostx64/x64/cl.exe" -DCMAKE_CXX_COMPILER:FILEPATH="C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.30.30705/bin/Hostx64/x64/cl.exe" -DBUILD_GSTREAMER_PLUGIN=ON -DCMAKE_MAKE_PROGRAM="C:\PROGRAM FILES\MICROSOFT VISUAL STUDIO\2022\COMMUNITY\COMMON7\IDE\COMMONEXTENSIONS\MICROSOFT\CMAKE\Ninja\ninja.exe" "C:\kin-vid-str-prod" 2>&1"” zwrócił kod zakończenia: 1”.
```

Like it is visible in the second log, I also struggle with "CMake generation".
Has anyone idea whether I have a problem with:
1. `No package 'libkvspicUtils' found`
2. `Looking for pthread.h - not found`
3.  gstreamer-app-1.0 is the package which wasn't found and broke the process
?

risho-lizhao, 2022-05-30:
@darek-gc 
Thank you for your advice.
I used x64 Native Tools Command Prompt and when `cannot open input file 'apps\app_rand.obj'` show again, I rerun `cmake .. -G "NMake Makefiles" -DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_JNI=TRUE` command, then the error disappears. The build runs further, but another error occurs:
```
[100%] Completed 'project_libcurl'
[100%] Built target project_libcurl
-- Finished building dependencies.
-- Found PkgConfig: C:/MinGW/bin/pkg-config.exe (found version "0.26")
-- Checking for module 'libkvspicUtils'
--   No package 'libkvspicUtils' found
-- Building for: Visual Studio 17 2022
-- Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.19044.
-- Configuring done
-- Generating done
-- Build files have been written to: C:/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic
.NET Framework 向け Microsoft (R) Build Engine バージョン 17.2.1+52cd2da31
Copyright (C) Microsoft Corporation.All rights reserved.

  Checking Build System
  Creating directories for 'libkvspic-download'
  Building Custom Rule C:/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/CMakeLists.txt
  Performing download step (git clone) for 'libkvspic-download'
  Cloning into 'kvspic-src'...
  HEAD is now at c832588 Testing images stuff (#172)
  Performing update step for 'libkvspic-download'
  No patch step for 'libkvspic-download'
  No configure step for 'libkvspic-download'
  No build step for 'libkvspic-download'
  No install step for 'libkvspic-download'
  No test step for 'libkvspic-download'
  Completed 'libkvspic-download'
  Building Custom Rule C:/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/CMakeLists.txt
-- Looking for pthread.h
-- Looking for pthread.h - not found
-- Found Threads: TRUE
-- Found OpenSSL: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib/libcrypto.lib (found version "1.1.1g")
-- Found CURL: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib/cmake/CURL/CURLConfig.cmake (found version "7.68.0-DEV")
-- Bitness 64 bits
-- Found ZLIB: C:/gstreamer/1.0/x86_64/lib/z.lib (found suitable version "1.2.8", minimum required is "1")
CMake Error at C:/Program Files/CMake/share/cmake-3.21/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find JNI (missing: JAVA_AWT_LIBRARY JAVA_JVM_LIBRARY
  JAVA_INCLUDE_PATH JAVA_INCLUDE_PATH2 JAVA_AWT_INCLUDE_PATH)
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.21/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  C:/Program Files/CMake/share/cmake-3.21/Modules/FindJNI.cmake:382 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:189 (find_package)

-- Configuring incomplete, errors occurred!
See also "C:/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/CMakeOutput.log".
See also "C:/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/CMakeError.log".
``` 

I tried advice from @disa6302 , that used my `build_windows.bat` and `ci.yml` files, but the new error still occurs.

niyatim23, 2022-05-31:
Hi @risho-lizhao, looks like you do not have Java installed on your Windows system. Please install Java with JDK and set JAVA_HOME in your environment variables and try again with the steps @disa6302 has suggested

risho-lizhao, 2022-06-02:
Hello @niyatim23 , Sorry for my late reply and thank you for your  advice.
After installing JDK and setting environment variables, error message `Could NOT find JNI` solved.
But another new error occurs. 
I have installed Gstreamer 1.0 and Gstreamer 1.0 Development(version: 1.6.2) and set up environment variables.

The new error:
```
  Checking Build System
  Creating directories for 'libkvspic-download'
  Building Custom Rule C:/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/CMakeLists.txt
  Performing download step (git clone) for 'libkvspic-download'
  Cloning into 'kvspic-src'...
  HEAD is now at c832588 Testing images stuff (#172)
  Performing update step for 'libkvspic-download'
  No patch step for 'libkvspic-download'
  No configure step for 'libkvspic-download'
  No build step for 'libkvspic-download'
  No install step for 'libkvspic-download'
  No test step for 'libkvspic-download'
  Completed 'libkvspic-download'
  Building Custom Rule C:/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/CMakeLists.txt
-- Looking for pthread.h
-- Looking for pthread.h - not found
-- Found Threads: TRUE
-- Found OpenSSL: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib/libcrypto.lib (found version "1.1.1g")
-- Found CURL: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib/cmake/CURL/CURLConfig.cmake (found version "7.68.0-DEV")
-- Bitness 64 bits
-- Found ZLIB: C:/gstreamer/1.0/x86_64/lib/z.lib (found suitable version "1.2.8", minimum required is "1")
-- Found JNI: C:/Program Files/Java/jdk-18.0.1.1/lib/jawt.lib
-- Checking for module 'gstreamer-app-1.0'
--
CMake Error at C:/Program Files/CMake/share/cmake-3.21/Modules/FindPkgConfig.cmake:554 (message):
  A required package was not found
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.21/Modules/FindPkgConfig.cmake:776 (_pkg_check_modules_internal)
  CMakeLists.txt:198 (pkg_check_modules)

-- Configuring incomplete, errors occurred!
See also "C:/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/CMakeOutput.log".
See also "C:/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/CMakeError.log".
```
 Do I miss anything else?

niyatim23, 2022-06-02:
Hi @risho-lizhao, I think you should clean up your previous build with `git clean -ffxd` from the root directory and try again. Also, make sure that you set the correct value in the `$env:Path` . The last two paths in the same are specific to where your directory's `lib` and `bin` folders are:  
`$env:Path += ';C:\Strawberry\perl\site\bin;C:\Strawberry\perl\bin;C:\Strawberry\c\bin;C:\Program Files\NASM;C:\amazon-kinesis-video-streams-producer-sdk-cpp\open-source\local\lib;C:\amazon-kinesis-video-streams-producer-sdk-cpp\open-source\local\bin'`

Can you check if you are specifying the correct paths here?

risho-lizhao, 2022-06-03:
Hello @niyatim23 , I clean up my previous build, and check all paths, they are set correctly.
In addition, I was able to confirm that Gstreamer worked properly by using the command: `gst-launch-1.0 videotestsrc ! autovideosink`.
First, I executed the `cmake -G "NMake Makefiles" ..` command and no errors occurred.
However, when I execute the `cmake -G "NMake Makefiles" -DBUILD_TEST=TRUE -DBUILD_GSTREAMER_PLUGIN=TRUE ..`  command, I get the following error:
```
Microsoft(R) Program Maintenance Utility Version 14.32.31328.0
Copyright (C) Microsoft Corporation.  All rights reserved.

Consolidate compiler generated dependencies of target gtest
[ 50%] Built target gtest
Consolidate compiler generated dependencies of target gtest_main
[100%] Built target gtest_main
Install the project...
-- Install configuration: "Debug"
-- Installing: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib/cmake/GTest/GTestTargets.cmake
-- Installing: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib/cmake/GTest/GTestTargets-debug.cmake
-- Installing: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib/cmake/GTest/GTestConfigVersion.cmake
-- Installing: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib/cmake/GTest/GTestConfig.cmake
-- Up-to-date: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/include
-- Up-to-date: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/include/gtest
-- Installing: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/include/gtest/gtest-death-test.h
-- Installing: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/include/gtest/gtest-message.h
-- Installing: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/include/gtest/gtest-param-test.h
-- Installing: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/include/gtest/gtest-param-test.h.pump
-- Installing: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/include/gtest/gtest-printers.h
-- Installing: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/include/gtest/gtest-spi.h
-- Installing: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/include/gtest/gtest-test-part.h
-- Installing: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/include/gtest/gtest-typed-test.h
-- Installing: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/include/gtest/gtest.h
-- Installing: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/include/gtest/gtest_pred_impl.h
-- Installing: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/include/gtest/gtest_prod.h
-- Up-to-date: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/include/gtest/internal
-- Up-to-date: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/include/gtest/internal/custom
-- Installing: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/include/gtest/internal/custom/gtest-port.h
-- Installing: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/include/gtest/internal/custom/gtest-printers.h
-- Installing: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/include/gtest/internal/custom/gtest.h
-- Installing: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/include/gtest/internal/custom/README.md
-- Installing: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/include/gtest/internal/gtest-death-test-internal.h
-- Installing: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/include/gtest/internal/gtest-filepath.h
-- Installing: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/include/gtest/internal/gtest-internal.h
-- Installing: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/include/gtest/internal/gtest-linked_ptr.h
-- Installing: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/include/gtest/internal/gtest-param-util-generated.h
-- Installing: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/include/gtest/internal/gtest-param-util-generated.h.pump
-- Installing: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/include/gtest/internal/gtest-param-util.h
-- Installing: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/include/gtest/internal/gtest-port-arch.h
-- Installing: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/include/gtest/internal/gtest-port.h
-- Installing: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/include/gtest/internal/gtest-string.h
-- Installing: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/include/gtest/internal/gtest-tuple.h
-- Installing: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/include/gtest/internal/gtest-tuple.h.pump
-- Installing: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/include/gtest/internal/gtest-type-util.h
-- Installing: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/include/gtest/internal/gtest-type-util.h.pump
-- Installing: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib/gtestd.lib
-- Installing: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib/gtest_maind.lib
-- Installing: C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib/pkgconfig/gtest.pc
[... cut ...]
CMake Error at C:/Program Files/CMake/share/cmake-3.21/Modules/FindPkgConfig.cmake:554 (message):
  A required package was not found
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.21/Modules/FindPkgConfig.cmake:776 (_pkg_check_modules_internal)
  CMakeLists.txt:198 (pkg_check_modules)

-- Configuring incomplete, errors occurred!
See also "C:/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/CMakeOutput.log".
See also "C:/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/CMakeError.log".
```

niyatim23, 2022-06-06:
Hi @risho-lizhao, please ensure that you are using openssl installed as a dependency from the SDK CPP. Please ensure that you have GStreamer related dependencies installed mentioned in the ci build shared with you

risho-lizhao, 2022-06-07:
Hello @niyatim23 , I have installed openssl, but could you tell me how should I check that I am using openssl installed as a dependency from the SDK CPP?

In addition, GStreamer related dependencies I have installed are shown below, have I missed any dependencies that I haven't installed?
```
C:\Users\jt0200544>perl -v

This is perl 5, version 32, subversion 1 (v5.32.1) built for MSWin32-x64-multi-thread

Copyright 1987-2021, Larry Wall

Perl may be copied only under the terms of either the Artistic License or the
GNU General Public License, which may be found in the Perl 5 source kit.

Complete documentation for Perl, including FAQ lists, should be found on
this system using "man perl" or "perldoc perl".  If you have access to the
Internet, point your browser at http://www.perl.org/, the Perl Home Page.

C:\Users\jt0200544>nasm -version
NASM version 2.15.05 compiled on Aug 28 2020

C:\Users\jt0200544>java -version
java version "18.0.1.1" 2022-04-22
Java(TM) SE Runtime Environment (build 18.0.1.1+2-6)
Java HotSpot(TM) 64-Bit Server VM (build 18.0.1.1+2-6, mixed mode, sharing)

C:\Users\jt0200544>pkg-config --version
0.26

C:\Users\jt0200544>openssl version
OpenSSL 1.1.1i  8 Dec 2020
```

niyatim23, 2022-06-07:
You may be able to check it in your build logs. Also, check if gstreamer works for you as a stand-alone application. If not, please resolve that first

niyatim23, 2022-06-08:
Hi @risho-lizhao, in that case, it is difficult to point out what is going wrong. Please try a clean build with the correct paths set and refer to the ci for the instructions.

Closing assuming answered. Feel free to reopen in case of any further issues

closed 2022-06-08 by niyatim23

mattrrubino, 2022-06-16:
@risho-lizhao 
I had the same issue as you where `gstreamer-app-1.0` could not be located. I was able to successfully build the plugin by:

1. Moving gstreamer so its new path would be `C:\gstreamer\1.0\x86_64` (instead of `msvc_x86_64` for the last folder)
2. Setting `GSTREAMER_1_0_ROOT_MSVC_X86_64` environment variable to `C:\gstreamer\1.0\x86_64`
3. Updating path environment variable to include `C:\gstreamer\1.0\x86_64\bin`, `C:\gstreamer\1.0\x86_64\lib`, and `C:\gstreamer\1.0\x86_64\include`
4. Running the `build_windows.bat` file in the `.github` directory of the repository

Not sure which change fixed it for me, but the build was successful. Hope this helps!

risho-lizhao, 2022-06-17:
@mattrrubino 
Thank you very much! I build it successfully by your way!

But now I have another problem when excute `gst-inspect-1.0 kvssink`.
I set environment variables by use follow command:
```
C:\amazon-kinesis-video-streams-producer-sdk-cpp>set GST_PLUGIN_PATH=%CD%\build
C:\amazon-kinesis-video-streams-producer-sdk-cpp>set PATH=%PATH%;%CD%\open-source\local\bin;%CD%\open-source\local\lib
```

Then, if I excute `gst-inspect-1.0 kvssink`, I will get error that looklike:
```
(gst-inspect-1.0:15760): GStreamer-WARNING **: 16:45:42.123: Failed to load plugin 'C:\amazon-kinesis-video-streams-producer-sdk-cpp\build\gstkvssink.dll': 'C:\amazon-kinesis-video-streams-producer-sdk-cpp\build\gstkvssink.dll': The specified module cannot be found.
No such element or plugin 'kvssink'
```

I confirmed that the `gstkvssink.dll` file exists in the `build` folder.
Do you hava any idea about this error?

mattrrubino, 2022-06-17:
@risho-lizhao 
Glad to hear that fixed the build!

I did not encounter this new problem, but I also did not follow the build instructions exactly.
Rather than setting the environment variables through the command prompt, I set them through the Windows GUI, copying the exact paths as opposed to using %CD%.
Then, I started a new command prompt, and it worked.

Maybe this will work for you too.

risho-lizhao, 2022-06-20:
@mattrrubino 
Thank you again, I originally installed version 1.16.2 of gtreamer and after replacing it with version 1.20.2, the problem was solved. I don't know exactly what the cause was, but at least it seems to be working properly now.