awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #866: Could NOT find JNI (missing: JAVA_INCLUDE_PATH JAVA_INCLUDE_PATH2   JAVA_AWT_INCLUDE_PATH)

JacopoBandoni, 2022-05-26:
I'm trying to install the SDK on a macOS Big Sur version 11.4 on a MacBook Pro.

** Logging **
Couldn't able to build cmake with Gstreamer and JNI without any errors.

** To Reproduce **
I performed the following commands in order:
1 - `$ brew install pkg-config openssl cmake gstreamer gst-plugins-base gst-plugins-good gst-plugins-bad gst-plugins-ugly log4cplus gst-libav`
2 - `git clone https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp.git `
3 - 'mkdir -p amazon-kinesis-video-streams-producer-sdk-cpp/build
cd amazon-kinesis-video-streams-producer-sdk-cpp/build'
4 - 'cmake .. -DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_JNI=TRUE'

At this point I get the following error:

`-- Kinesis Video Cpp Producer path is /Users/jacopobandoni/Desktop/aivideo/amazon-kinesis-video-streams-producer-sdk-cpp
-- dependencies install path is /Users/jacopobandoni/Desktop/aivideo/amazon-kinesis-video-streams-producer-sdk-cpp/open-source
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/jacopobandoni/Desktop/aivideo/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer
[ 11%] Performing update step for 'libkvscproducer-download'
[ 22%] No patch step for 'libkvscproducer-download'
[ 33%] No configure step for 'libkvscproducer-download'
[ 44%] No build step for 'libkvscproducer-download'
[ 55%] No install step for 'libkvscproducer-download'
[ 66%] No test step for 'libkvscproducer-download'
[ 77%] Completed 'libkvscproducer-download'
[100%] Built target libkvscproducer-download
-- Kinesis Video Producer path is /Users/jacopobandoni/Desktop/aivideo/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src
-- Checking for module 'libkvspicUtils'
--   No package 'libkvspicUtils' found
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/jacopobandoni/Desktop/aivideo/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic
[ 11%] Performing update step for 'libkvspic-download'
[ 22%] No patch step for 'libkvspic-download'
[ 33%] No configure step for 'libkvspic-download'
[ 44%] No build step for 'libkvspic-download'
[ 55%] No install step for 'libkvspic-download'
[ 66%] No test step for 'libkvspic-download'
[ 77%] Completed 'libkvspic-download'
[100%] Built target libkvspic-download
fatal: not a git repository (or any of the parent directories): .git
-- Bitness 64 bits
found log4cplus.
CMake Error at /usr/local/Cellar/cmake/3.23.1_1/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find JNI (missing: JAVA_INCLUDE_PATH JAVA_INCLUDE_PATH2
  JAVA_AWT_INCLUDE_PATH)
Call Stack (most recent call first):
  /usr/local/Cellar/cmake/3.23.1_1/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  /usr/local/Cellar/cmake/3.23.1_1/share/cmake/Modules/FindJNI.cmake:382 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:189 (find_package)

-- Configuring incomplete, errors occurred!
See also "/Users/jacopobandoni/Desktop/aivideo/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/CMakeOutput.log".
See also "/Users/jacopobandoni/Desktop/aivideo/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/CMakeError.log".`

** Considerations **
By checking the following issue (https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/843) I noted that I should have defined the JAVA_HOME variable.

But even after setting:
export JAVA_HOME=/Library/Internet\ Plug-Ins/JavaAppletPlugin.plugin/Contents/Home

in command line or in .zshrc the previous error stay the same.

niyatim23, 2022-06-02:
Hi @JacopoBandoni, can you add check if the paths in the following section exist on your Mac and add these lines to the cmake file [here](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/CMakeLists.txt#L33) and see if that works for you?
```
if(APPLE)
      SET(JAVA_INCLUDE_PATH "$ENV{JAVA_HOME}/include")
      SET(JAVA_INCLUDE_PATH2 "/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/System/Library/Frameworks/JavaVM.framework/Versions/A/Headers/")
      SET(JAVA_AWT_LIBRARY "$ENV{JAVA_HOME}/lib/libawt.dylib")
      SET(JAVA_JVM_LIBRARY "$ENV{JAVA_HOME}/lib/server/libjvm.dylib")
      set(JAVA_AWT_INCLUDE_PATH "$ENV{JAVA_HOME}/include")
  endif()
```

niyatim23, 2022-06-15:
Closing due to no response. Feel free to reopen the issue in case of any concerns

closed 2022-06-15 by niyatim23