awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #868: [BUG] - No CMAKE_CXX_COMPILER could be found on Ubuntu 22.

igoralves1, 2022-06-10:
**Describe the bug**
Getting the following error when building

**To Reproduce**
Steps to reproduce the behavior:
1. On Cloned repository
2. Installed required libraries for Ubuntu 22 and cd to build. The cmake version is `3.22.1`
3. Built following instructions: `amazon-kinesis-video-streams-producer-sdk-cpp/build$ cmake .. -DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_JNI=TRUE`
4. Got following error:
```
-- The CXX compiler identification is unknown
CMake Error at CMakeLists.txt:8 (project):
  No CMAKE_CXX_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Configuring incomplete, errors occurred!
See also "/home/ila/Documents/vhosts/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/CMakeOutput.log".
See also "/home/ila/Documents/vhosts/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/CMakeError.log".

```
5. Checking error logs: `/Documents/vhosts/amazon-kinesis-video-streams-producer-sdk-cpp/build$ cat CMakeFiles/CMakeError.log`
```
Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND 
Build flags: 
Id flags:  

The output was:
No such file or directory

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND 
Build flags: 
Id flags: -c 

The output was:
No such file or directory

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND 
Build flags: 
Id flags: --c++ 

The output was:
No such file or directory

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND 
Build flags: 
Id flags: --ec++ 

The output was:
No such file or directory

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND 
Build flags: 
Id flags: --target=arm-arm-none-eabi;-mcpu=cortex-m3 

The output was:
No such file or directory

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND 
Build flags: 
Id flags: -c;-I__does_not_exist__ 

The output was:
No such file or directory

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND 
Build flags: 
Id flags:  

The output was:
No such file or directory

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND 
Build flags: 
Id flags: -c 

The output was:
No such file or directory

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND 
Build flags: 
Id flags: --c++ 

The output was:
No such file or directory

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND 
Build flags: 
Id flags: --ec++ 

The output was:
No such file or directory

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND 
Build flags: 
Id flags: --target=arm-arm-none-eabi;-mcpu=cortex-m3 

The output was:
No such file or directory

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND 
Build flags: 
Id flags: -c;-I__does_not_exist__ 

The output was:
No such file or directory

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND 
Build flags: 
Id flags:  

The output was:
No such file or directory

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND 
Build flags: 
Id flags: -c 

The output was:
No such file or directory

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND 
Build flags: 
Id flags: --c++ 

The output was:
No such file or directory

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND 
Build flags: 
Id flags: --ec++ 

The output was:
No such file or directory

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND 
Build flags: 
Id flags: --target=arm-arm-none-eabi;-mcpu=cortex-m3 

The output was:
No such file or directory

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND 
Build flags: 
Id flags: -c;-I__does_not_exist__ 

The output was:
No such file or directory

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND 
Build flags: 
Id flags:  

The output was:
No such file or directory

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND 
Build flags: 
Id flags: -c 

The output was:
No such file or directory

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND 
Build flags: 
Id flags: --c++ 

The output was:
No such file or directory

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND 
Build flags: 
Id flags: --ec++ 

The output was:
No such file or directory

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND 
Build flags: 
Id flags: --target=arm-arm-none-eabi;-mcpu=cortex-m3 

The output was:
No such file or directory

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND 
Build flags: 
Id flags: -c;-I__does_not_exist__ 

The output was:
No such file or directory
```

**Expected behavior**
To build the project

**Open source building**
cmake version 3.22.1

**Platform (please complete the following information):**
 - OS: [Ubuntu]
 - Processor: [Intel(R) Core(TM) i7-8550U CPU @ 1.80GHz]
 - Version [22]

niyatim23, 2022-06-14:
Hi @igoralves1,  can you please try a clean build after installing all the dependencies with `sudo apt-get install libssl-dev libcurl4-openssl-dev liblog4cplus-dev libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev gstreamer1.0-plugins-base-apps gstreamer1.0-plugins-bad gstreamer1.0-plugins-good gstreamer1.0-plugins-ugly gstreamer1.0-tools`

niyatim23, 2022-06-20:
Closing due to no response. Feel free to reopen in case of further questions

closed 2022-06-20 by niyatim23