awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #869: [QUESTION]HTTP Error 403 : Response: {"Message":"User: arn:aws:iam::454510816963:user/new_cpp_user is not authorized to perform: kinesisvideo:DescribeStream on resource: new_stream_cpp"}

BADAL244, 2022-06-20:
[INFO ] [20-06-2022 09:07:18:220.985 GMT] Using region: us-east-1
[INFO ] [20-06-2022 09:07:18:221.070 GMT] Using aws credentials for Kinesis Video Streams
[INFO ] [20-06-2022 09:07:18:221.092 GMT] No session token was detected.
[INFO ] [20-06-2022 09:07:18:222.557 GMT] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ] [20-06-2022 09:07:18:222.643 GMT] heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ] [20-06-2022 09:07:18:222.670 GMT] heapInitialize(): Creating AIV heap.
[INFO ] [20-06-2022 09:07:18:222.706 GMT] heapInitialize(): Heap is initialized OK
[DEBUG] [20-06-2022 09:07:18:222.916 GMT] getSecurityTokenHandler invoked
[DEBUG] [20-06-2022 09:07:18:223.019 GMT] Refreshing credentials. Force refreshing: 0 Now time is: 1655716038222967188 Expiration: 0
[INFO ] [20-06-2022 09:07:18:223.047 GMT] New credentials expiration is 1655719638
[INFO ] [20-06-2022 09:07:18:223.110 GMT] createDeviceResultEvent(): Create device result event.
[DEBUG] [20-06-2022 09:07:18:223.134 GMT] clientReadyHandler invoked
[DEBUG] [20-06-2022 09:07:18:223.185 GMT] Client is ready
[INFO ] [20-06-2022 09:07:18:223.293 GMT] Creating Kinesis Video Stream new_stream_cpp
[INFO ] [20-06-2022 09:07:18:223.362 GMT] createKinesisVideoStream(): Creating Kinesis Video Stream.
[INFO ] [20-06-2022 09:07:18:223.392 GMT] logStreamInfo(): SDK version: 70f74f14cf27b09f71dc1889f36eb6e04cdd90a8
[DEBUG] [20-06-2022 09:07:18:223.416 GMT] logStreamInfo(): Kinesis Video Stream Info
[DEBUG] [20-06-2022 09:07:18:223.440 GMT] logStreamInfo():      Stream name: new_stream_cpp 
[DEBUG] [20-06-2022 09:07:18:223.461 GMT] logStreamInfo():      Streaming type: STREAMING_TYPE_REALTIME 
[DEBUG] [20-06-2022 09:07:18:223.484 GMT] logStreamInfo():      Content type: video/h264,audio/aac 
[DEBUG] [20-06-2022 09:07:18:223.510 GMT] logStreamInfo():      Max latency (100ns): 600000000
[DEBUG] [20-06-2022 09:07:18:223.534 GMT] logStreamInfo():      Fragment duration (100ns): 20000000
[DEBUG] [20-06-2022 09:07:18:223.554 GMT] logStreamInfo():      Key frame fragmentation: Yes
[DEBUG] [20-06-2022 09:07:18:223.576 GMT] logStreamInfo():      Use frame timecode: Yes
[DEBUG] [20-06-2022 09:07:18:223.593 GMT] logStreamInfo():      Absolute frame timecode: Yes
[DEBUG] [20-06-2022 09:07:18:223.610 GMT] logStreamInfo():      Nal adaptation flags: 0
[DEBUG] [20-06-2022 09:07:18:223.630 GMT] logStreamInfo():      Average bandwith (bps): 4194304
[DEBUG] [20-06-2022 09:07:18:223.652 GMT] logStreamInfo():      Framerate: 100
[DEBUG] [20-06-2022 09:07:18:223.668 GMT] logStreamInfo():      Buffer duration (100ns): 1200000000
[DEBUG] [20-06-2022 09:07:18:223.687 GMT] logStreamInfo():      Replay duration (100ns): 400000000
[DEBUG] [20-06-2022 09:07:18:223.712 GMT] logStreamInfo():      Connection Staleness duration (100ns): 600000000
[DEBUG] [20-06-2022 09:07:18:223.738 GMT] logStreamInfo():      Store Pressure Policy: 1
[DEBUG] [20-06-2022 09:07:18:223.755 GMT] logStreamInfo():      View Overflow Policy: 1
[DEBUG] [20-06-2022 09:07:18:223.778 GMT] logStreamInfo():      Segment UUID: NULL
[DEBUG] [20-06-2022 09:07:18:223.801 GMT] logStreamInfo():      Frame ordering mode: 5
[DEBUG] [20-06-2022 09:07:18:223.820 GMT] logStreamInfo(): Track list
[DEBUG] [20-06-2022 09:07:18:223.841 GMT] logStreamInfo():      Track id: 1
[DEBUG] [20-06-2022 09:07:18:223.857 GMT] logStreamInfo():      Track name: kinesis_video
[DEBUG] [20-06-2022 09:07:18:223.874 GMT] logStreamInfo():      Codec id: V_MPEG4/ISO/AVC
[DEBUG] [20-06-2022 09:07:18:223.898 GMT] logStreamInfo():      Track type: TRACK_INFO_TYPE_VIDEO
[DEBUG] [20-06-2022 09:07:18:223.914 GMT] logStreamInfo():      Track cpd: NULL
[DEBUG] [20-06-2022 09:07:18:223.926 GMT] logStreamInfo():      Track id: 2
[DEBUG] [20-06-2022 09:07:18:223.937 GMT] logStreamInfo():      Track name: audio
[DEBUG] [20-06-2022 09:07:18:223.947 GMT] logStreamInfo():      Codec id: A_AAC
[DEBUG] [20-06-2022 09:07:18:223.957 GMT] logStreamInfo():      Track type: TRACK_INFO_TYPE_AUDIO
[DEBUG] [20-06-2022 09:07:18:223.968 GMT] logStreamInfo():      Track cpd: NULL
[INFO ] [20-06-2022 09:07:20:093.269 GMT] writeHeaderCallback(): RequestId: 05c00605-0bff-479b-93e4-918f5bdcf29c
[WARN ] [20-06-2022 09:07:20:093.668 GMT] curlCompleteSync(): HTTP Error 403 : Response: {"Message":"User: arn:aws:iam::454510816963:user/new_cpp_user is not authorized to perform: kinesisvideo:DescribeStream on resource: new_stream_cpp"}
Request URL: https://kinesisvideo.us-east-1.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AKIAWTUXUV3BQIIXOSHU/20220620/us-east-1/kinesisvideo/aws4_request, SignedHeaders=host;user-agent;x-amz-date, Signature=a735ca0e69fe531acf1b4986b9f62133c5333d57e2bb6a59d02a9388c
[DEBUG] [20-06-2022 09:07:20:093.814 GMT] describeStreamCurlHandler(): DescribeStream API response: {"Message":"User: arn:aws:iam::454510816963:user/new_cpp_user is not authorized to perform: kinesisvideo:DescribeStream on resource: new_stream_cpp"}
[INFO ] [20-06-2022 09:07:20:097.200 GMT] describeStreamResultEvent(): Describe stream result event.
[DEBUG] [20-06-2022 09:07:20:097.364 GMT] defaultStreamStateTransitionHook(): Stream state machine retry count: 0
[DEBUG] [20-06-2022 09:07:20:097.432 GMT] defaultStreamStateTransitionHook(): 
 KinesisVideoStream base result is [403]. Executing KVS retry handler of retry strategy type [1]
[INFO ] [20-06-2022 09:07:21:514.876 GMT] writeHeaderCallback(): RequestId: c57c3716-6dde-4bbb-b78b-80f10f8c9c89
[WARN ] [20-06-2022 09:07:21:515.244 GMT] curlCompleteSync(): HTTP Error 403 : Response: {"Message":"User: arn:aws:iam::454510816963:user/new_cpp_user is not authorized to perform: kinesisvideo:DescribeStream on resource: new_stream_cpp"}
Request URL: https://kinesisvideo.us-east-1.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AKIAWTUXUV3BQIIXOSHU/20220620/us-east-1/kinesisvideo/aws4_request, SignedHeaders=host;user-agent;x-amz-date, Signature=107ff334dc509cd8573971c6002814aa37b5e62c4605fd3826d5ca2b1
[DEBUG] [20-06-2022 09:07:21:515.379 GMT] describeStreamCurlHandler(): DescribeStream API response: {"Message":"User: arn:aws:iam::454510816963:user/new_cpp_user is not authorized to perform: kinesisvideo:DescribeStream on resource: new_stream_cpp"}
[INFO ] [20-06-2022 09:07:21:518.577 GMT] describeStreamResultEvent(): Describe stream result event.
[DEBUG] [20-06-2022 09:07:21:518.752 GMT] defaultStreamStateTransitionHook(): Stream state machine retry count: 1
[DEBUG] [20-06-2022 09:07:21:518.831 GMT] defaultStreamStateTransitionHook(): 
 KinesisVideoStream base result is [403]. Executing KVS retry handler of retry strategy type [1]
[INFO ] [20-06-2022 09:07:23:182.553 GMT] writeHeaderCallback(): RequestId: 4d8e1438-63e6-434a-bb7c-0a504b0d4d2f
[WARN ] [20-06-2022 09:07:23:182.924 GMT] curlCompleteSync(): HTTP Error 403 : Response: {"Message":"User: arn:aws:iam::454510816963:user/new_cpp_user is not authorized to perform: kinesisvideo:DescribeStream on resource: new_stream_cpp"}
Request URL: https://kinesisvideo.us-east-1.amazonaws.com/describeStream
Request Headers:
[... 38 lines cut ...]
[ERROR] [20-06-2022 09:07:32:075.034 GMT] Reporting stream error. Errored timecode: 0 Status: 1375731729
[WARN ] [20-06-2022 09:07:32:075.089 GMT] continuousRetryStreamErrorReportHandler(): Reporting stream error. Errored timecode: 0 Status: 0x52000011
[DEBUG] [20-06-2022 09:07:32:075.584 GMT] curlApiCallbacksShutdownActiveRequests(): pActiveRequests hashtable is empty
[INFO ] [20-06-2022 09:07:33:683.636 GMT] writeHeaderCallback(): RequestId: 4688f949-9df8-460a-8f24-4e22b463c906
[WARN ] [20-06-2022 09:07:33:683.809 GMT] curlCompleteSync(): HTTP Error 403 : Response: {"Message":"User: arn:aws:iam::454510816963:user/new_cpp_user is not authorized to perform: kinesisvideo:DescribeStream on resource: new_stream_cpp"}
Request URL: https://kinesisvideo.us-east-1.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AKIAWTUXUV3BQIIXOSHU/20220620/us-east-1/kinesisvideo/aws4_request, SignedHeaders=host;user-agent;x-amz-date, Signature=bdae594ad721b05b8ffc9c58e8fed8dbe9f5dd05aaf2ba1a534842370
[DEBUG] [20-06-2022 09:07:33:683.851 GMT] describeStreamCurlHandler(): DescribeStream API response: {"Message":"User: arn:aws:iam::454510816963:user/new_cpp_user is not authorized to perform: kinesisvideo:DescribeStream on resource: ne [... cut ...]
[DEBUG] [20-06-2022 09:07:46:384.529 GMT] defaultStreamStateTransitionHook(): Stream state machine retry count: 7
[DEBUG] [20-06-2022 09:07:46:384.582 GMT] defaultStreamStateTransitionHook(): 
 KinesisVideoStream base result is [403]. Executing KVS retry handler of retry strategy type [1]
[ERROR] [20-06-2022 09:07:48:224.523 GMT] createKinesisVideoStreamSync(): Failed to create Kinesis Video Stream - timed out.
[INFO ] [20-06-2022 09:07:48:224.682 GMT] freeKinesisVideoStream(): Freeing Kinesis Video stream.

niyatim23, 2022-06-20:
Hi @BADAL244, looks like your IAM user does not have the permissions to perform DescribeStream. I would suggest you to add policy `AmazonKinesisVideoStreamsFullAccess` to your user's permissions and see if it fixes this.

BADAL244, 2022-06-21:
[image]
I have added one more policy (Administration Access ) , its working perfectly now . Thanks for the suggestion.