awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #870: [QUESTION]

gi12345, 2022-06-20:
Unable to create Kinesis-video-stream

The function kinesis_video_producer->createStreamSync() fails (never completes). The code was working before with an older SDK version (~ 2 years ago).
********************************************************************************
Checked-out and built June 17, 2022:
git clone https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp.git

cd amazon-kinesis-video-streams-producer-sdk-cpp
export GST_PLUGIN_PATH=`pwd`
export LD_LIBRARY_PATH=`pwd`/open-source/local/lib
mkdir build
cd build

cmake .. -DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_DEPENDENCIES=ON >result1 2>&1
make >result2 2>&1
********************************************************************************
This is the code piece:
void CKVSVideoStreamService::kinesis_video_stream_init(CustomData *data) {
    /* create a test stream */
    map<string, string> tags;
    char tag_name[MAX_TAG_NAME_LEN];
    char tag_val[MAX_TAG_VALUE_LEN];
    SPRINTF(tag_name, "piTag");
    SPRINTF(tag_val, "piValue");

    printf("xxx-gi kinesis_video_stream_init. Point-1\n");
    STREAMING_TYPE streaming_type = DEFAULT_STREAMING_TYPE;
    data->use_absolute_fragment_times = DEFAULT_ABSOLUTE_FRAGMENT_TIMES;

    printf("xxx-gi kinesis_video_stream_init. Point-2\n");
    unique_ptr<StreamDefinition> stream_definition(new StreamDefinition(
        data->stream_name,
        hours(data->retention_hours),
        &tags,
        DEFAULT_KMS_KEY_ID,
        streaming_type,
        DEFAULT_CONTENT_TYPE,
        duration_cast<milliseconds> (seconds(DEFAULT_MAX_LATENCY_SECONDS)),
        milliseconds(DEFAULT_FRAGMENT_DURATION_MILLISECONDS),
        milliseconds(DEFAULT_TIMECODE_SCALE_MILLISECONDS),
        DEFAULT_KEY_FRAME_FRAGMENTATION,
        DEFAULT_FRAME_TIMECODES,
        data->use_absolute_fragment_times,
        DEFAULT_FRAGMENT_ACKS,
        DEFAULT_RESTART_ON_ERROR,
        DEFAULT_RECALCULATE_METRICS,
        0,
        DEFAULT_STREAM_FRAMERATE,
        DEFAULT_AVG_BANDWIDTH_BPS,
        seconds(DEFAULT_BUFFER_DURATION_SECONDS),
        seconds(DEFAULT_REPLAY_DURATION_SECONDS),
        seconds(DEFAULT_CONNECTION_STALENESS_SECONDS),
        DEFAULT_CODEC_ID,
        DEFAULT_TRACKNAME,
        nullptr,
        0));
    printf("xxx-gi kinesis_video_stream_init. Point-3 ***************************************************\n");
    data->kinesis_video_stream = data->kinesis_video_producer->createStreamSync(move(stream_definition));

    printf("xxx-gi kinesis_video_stream_init. Point-4 ***************************************************\n");
    // reset state
    data->stream_status = STATUS_SUCCESS;
    data->stream_started = false;

    printf("xxx-gi kinesis_video_stream_init. Point-5\n");
    LOG_DEBUG("Stream is ready");
}
********************************************************************************
The error seems here (entire log below):
[WARN ] [20-06-2022 15:13:36:449.385 GMT] curlCompleteSync(): curl perform failed for url https://kinesisvideo.us-east-2.amazonaws.com/describeStream with result Problem with the SSL CA cert (path? access rights?): error setting certificate verify locations:
[... 279 lines cut ...]
# cat /etc/pki/tls/certs/cert.pem
-----BEGIN CERTIFICATE-----
MIID7zCCAtegAwIBAgIBADANBgkqhkiG9w0BAQsFADCBmDELMAkGA1UEBhMCVVMx
EDAOBgNVBAgTB0FyaXpvbmExEzARBgNVBAcTClNjb3R0c2RhbGUxJTAjBgNVBAoT
HFN0YXJmaWVsZCBUZWNobm9sb2dpZXMsIEluYy4xOzA5BgNVBAMTMlN0YXJmaWVs
ZCBTZXJ2aWNlcyBSb290IENlcnRpZmljYXRlIEF1dGhvcml0eSAtIEcyMB4XDTA5
MDkwMTAwMDAwMFoXDTM3MTIzMTIzNTk1OVowgZgxCzAJBgNVBAYTAlVTMRAwDgYD
VQQIEwdBcml6b25hMRMwEQYDVQQHEwpTY290dHNkYWxlMSUwIwYDVQQKExxTdGFy
ZmllbGQgVGVjaG5vbG9naWVzLCBJbmMuMTswOQYDVQQDEzJTdGFyZmllbGQgU2Vy
dmljZXMgUm9vdCBDZXJ0aWZpY2F0ZSBBdXRob3JpdHkgLSBHMjCCASIwDQYJKoZI
hvcNAQEBBQADggEPADCCAQoCggEBANUMOsQq+U7i9b4Zl1+OiFOxHz/Lz58gE20p
OsgPfTz3a3Y4Y9k2YKibXlwAgLIvWX/2h/klQ4bnaRtSmpDhcePYLQ1Ob/bISdm2
8xpWriu2dBTrz/sm4xq6HZYuajtYlIlHVv8loJNwU4PahHQUw2eeBGg6345AWh1K
Ts9DkTvnVtYAcMtS7nt9rjrnvDH5RfbCYM8TWQIrgMw0R9+53pBlbQLPLJGmpufe
hRhJfGZOozptqbXuNC66DQO4M99H67FrjSXZm86B0UVGMpZwh94CDklDhbZsc7tk
6mFBrMnUVN+HL8cisibMn1lUaJ/8viovxFUcdUBgF4UCVTmLfwUCAwEAAaNCMEAw
DwYDVR0TAQH/BAUwAwEB/zAOBgNVHQ8BAf8EBAMCAQYwHQYDVR0OBBYEFJxfAN+q
AdcwKziIorhtSpzyEZGDMA0GCSqGSIb3DQEBCwUAA4IBAQBLNqaEd2ndOxmfZyMI
bw5hyf2E3F/YNoHN2BtBLZ9g3ccaaNnRbobhiCPPE95Dz+I0swSdHynVv/heyNXB
ve6SbzJ08pGCL72CQnqtKrcgfU28elUSwhXqvfdqlS5sdJ/PHLTyxQGjhdByPq1z
qwubdQxtRbeOlKyWN7Wg0I8VRw7j6IPdj/3vQQF3zCepYoUz8jcI73HPdwbeyBkd
iEDPfUYd/x7H4c7/I9vG+o1VTqkC50cRRj70/b17KSa7qWFiNyi2LSr2EIZkyXCn
0q23KXB56jzaYyWf/Wi3MOxw+3WKt21gZ7IeyLnp2KhvAotnDU0mV3HaIPzBSlCN
sSi6
-----END CERTIFICATE-----

disa6302, 2022-06-24:
Have you tried clean building? 
I am not sure what your older version was but our SDK went through significant folder restructuring between 2.1.0 and 3.0.0.

gi12345, 2022-06-24:
Yes. It is a  completely new clean build. It was built with the KVS SDK from June17,2022. The build machine is an AWS-EC2 with OS Ubuntu-20 LTS ARM64 preinstalled by AWS on EC2.

These are the exact build steps:
sudo apt install libssl-dev
sudo apt install libcurl4-openssl-dev
sudo apt install libgstreamer1.0-dev
sudo apt install libgstreamer-plugins-base1.0-dev
sudo apt install gstreamer1.0-plugins-base-apps
sudo apt install gstreamer1.0-plugins-bad
sudo apt install gstreamer1.0-plugins-good
sudo apt install gstreamer1.0-plugins-ugly
sudo apt install gstreamer1.0-tools

mkdir ~/kvssdk_jun17
cd ~/kvssdk_jun17
git clone https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp.git

cd amazon-kinesis-video-streams-producer-sdk-cpp
export GST_PLUGIN_PATH=`pwd`
export LD_LIBRARY_PATH=`pwd`/open-source/local/lib
mkdir build
cd build

cmake .. -DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_DEPENDENCIES=ON >result1 2>&1
make >result2 2>&1

hassanctech, 2022-06-27:
Can you please place the cert file here:  `/etc/ssl/cert.pem` and try again?  Make sure the location is correct.

gi12345, 2022-06-27:
# cat /etc/ssl/cert.pem
-----BEGIN CERTIFICATE-----
MIID7zCCAtegAwIBAgIBADANBgkqhkiG9w0BAQsFADCBmDELMAkGA1UEBhMCVVMx
EDAOBgNVBAgTB0FyaXpvbmExEzARBgNVBAcTClNjb3R0c2RhbGUxJTAjBgNVBAoT
HFN0YXJmaWVsZCBUZWNobm9sb2dpZXMsIEluYy4xOzA5BgNVBAMTMlN0YXJmaWVs
ZCBTZXJ2aWNlcyBSb290IENlcnRpZmljYXRlIEF1dGhvcml0eSAtIEcyMB4XDTA5
MDkwMTAwMDAwMFoXDTM3MTIzMTIzNTk1OVowgZgxCzAJBgNVBAYTAlVTMRAwDgYD
VQQIEwdBcml6b25hMRMwEQYDVQQHEwpTY290dHNkYWxlMSUwIwYDVQQKExxTdGFy
ZmllbGQgVGVjaG5vbG9naWVzLCBJbmMuMTswOQYDVQQDEzJTdGFyZmllbGQgU2Vy
dmljZXMgUm9vdCBDZXJ0aWZpY2F0ZSBBdXRob3JpdHkgLSBHMjCCASIwDQYJKoZI
hvcNAQEBBQADggEPADCCAQoCggEBANUMOsQq+U7i9b4Zl1+OiFOxHz/Lz58gE20p
OsgPfTz3a3Y4Y9k2YKibXlwAgLIvWX/2h/klQ4bnaRtSmpDhcePYLQ1Ob/bISdm2
8xpWriu2dBTrz/sm4xq6HZYuajtYlIlHVv8loJNwU4PahHQUw2eeBGg6345AWh1K
Ts9DkTvnVtYAcMtS7nt9rjrnvDH5RfbCYM8TWQIrgMw0R9+53pBlbQLPLJGmpufe
hRhJfGZOozptqbXuNC66DQO4M99H67FrjSXZm86B0UVGMpZwh94CDklDhbZsc7tk
6mFBrMnUVN+HL8cisibMn1lUaJ/8viovxFUcdUBgF4UCVTmLfwUCAwEAAaNCMEAw
DwYDVR0TAQH/BAUwAwEB/zAOBgNVHQ8BAf8EBAMCAQYwHQYDVR0OBBYEFJxfAN+q
AdcwKziIorhtSpzyEZGDMA0GCSqGSIb3DQEBCwUAA4IBAQBLNqaEd2ndOxmfZyMI
bw5hyf2E3F/YNoHN2BtBLZ9g3ccaaNnRbobhiCPPE95Dz+I0swSdHynVv/heyNXB
ve6SbzJ08pGCL72CQnqtKrcgfU28elUSwhXqvfdqlS5sdJ/PHLTyxQGjhdByPq1z
qwubdQxtRbeOlKyWN7Wg0I8VRw7j6IPdj/3vQQF3zCepYoUz8jcI73HPdwbeyBkd
iEDPfUYd/x7H4c7/I9vG+o1VTqkC50cRRj70/b17KSa7qWFiNyi2LSr2EIZkyXCn
0q23KXB56jzaYyWf/Wi3MOxw+3WKt21gZ7IeyLnp2KhvAotnDU0mV3HaIPzBSlCN
sSi6
-----END CERTIFICATE-----

gi12345, 2022-06-27:
The cert file /etc/ssl/cert.pem sits at this location. The output is as above.

hassanctech, 2022-06-27:
Is there a file here:  `/etc/ssl/certs/ca-certificates.crt`

gi12345, 2022-06-27:
no

hassanctech, 2022-06-27:
Can you create one there just try a symlink to `/etc/ssl/cert.pem` ?

gi12345, 2022-07-06:
I took a closer look at the error log before adding the symlink" "Can you create one there just try a symlink to /etc/ssl/cert.pem ?"
...........................................
Error in Log:
[WARN ] [20-06-2022 15:13:36:171.158 GMT] curlCompleteSync(): curl perform failed for url https://kinesisvideo.us-east-2.amazonaws.com/describeStream with result Problem with the SSL CA cert (path? access rights?): error setting certificate verify locations:
CAfile: /etc/ssl/certs/ca-certificates.crt
CApath: /etc/ssl/certs
............................................
Application code to trigger thsi error:
   data->credential.reset(new Credentials(string(accessKey),
                                          string(secretKey),
                                          sessionTokenStr,
                                          std::chrono::seconds(DEFAULT_CREDENTIAL_EXPIRATION_SECONDS)));
...........................................                                             
AWS-KVS-SDK file: dependency/libkvscproducer/kvscproducer-src/src/source/Response.c  
STATUS curlCompleteSync(PCurlResponse pCurlResponse)
{
    STATUS retStatus = STATUS_SUCCESS, status = STATUS_SUCCESS;
    PCHAR url;
    struct curl_slist* header;
    PCurlApiCallbacks pCurlApiCallbacks;                                        
    ...                                
                                               
           if (result != CURLE_OK && ATOMIC_LOAD_BOOL(&pCurlResponse->terminated)) {
        // The transmission has been force terminated.
        pCurlResponse->callInfo.httpStatus = HTTP_STATUS_CODE_REQUEST_TIMEOUT;
        pCurlResponse->callInfo.callResult = SERVICE_CALL_REQUEST_TIMEOUT;
    } else if (result != CURLE_OK) {
        curl_easy_getinfo(pCurlResponse->pCurl, CURLINFO_EFFECTIVE_URL, &url);
        DLOGW("curl perform failed for url %s with result %s: %s", url, curl_easy_strerror(result), pCurlResponse->callInfo.errorBuffer);  <====

        pCurlResponse->callInfo.callResult = getServiceCallResultFromCurlStatus(result);
    } else {
        // get the response code and note the request completion time
        curl_easy_getinfo(pCurlResponse->pCurl, CURLINFO_RESPONSE_CODE, &pCurlResponse->callInfo.httpStatus);
        pCurlResponse->callInfo.callResult = getServiceCallResultFromHttpStatus(pCurlResponse->callInfo.httpStatus);
    }        
............................................
The AWS cloud side code triggers the error; it should be possible to determine the root cause from the error string.

jdelapla, 2022-07-08:
Hey, sorry for the slow response.
I checked your requests on backend, and they're all succeeding. The error you're getting says that your arguments are invalid, but they're not, so I think something has gone wrong with your package configuration. I suspect it's something with jsmn, or the JSON parsing in general although the JSON of the response you got appears fine.

Could you apply the following patch, delete the build dir and make clean and report the logs on debug level? It should verify this is the issue. If it is, from there we need to find out why your jsmn_init or JSON parsing is failing on a valid JSON string.

```
diff --git a/CMake/Dependencies/libkvscproducer-CMakeLists.txt b/CMake/Dependencies/libkvscproducer-CMakeLists.txt
index 6c4490e..96e0a5e 100644
--- a/CMake/Dependencies/libkvscproducer-CMakeLists.txt
+++ b/CMake/Dependencies/libkvscproducer-CMakeLists.txt
@@ -7,7 +7,7 @@ include(ExternalProject)
 # clone repo only
 ExternalProject_Add(libkvscproducer-download
 	GIT_REPOSITORY    https://github.com/awslabs/amazon-kinesis-video-streams-producer-c.git
-	GIT_TAG           79da5b6d6ad67560cc08530f967e9187a6d1c594
+	GIT_TAG           d5a0455b752d1a3149b022d734dd8c7f759d308e
 	SOURCE_DIR        "${CMAKE_CURRENT_BINARY_DIR}/kvscproducer-src"
 	BINARY_DIR        "${CMAKE_CURRENT_BINARY_DIR}/kvscproducer-build"
 	CONFIGURE_COMMAND ""

gi12345, 2022-07-08:
Hello,
Thanks for you feedback. I am not sure how to apply your recommended fix:

***@***.***:~/kvssdk_jun17/amazon-kinesis-video-streams-producer-sdk-cpp$ diff --git a/CMake/Dependencies/libkvscproducer-CMakeLists.txt b/CMake/Dependencies/libkvscproducer-CMakeLists.txt
diff: unrecognized option '--git'
diff: Try 'diff --help' for more information.

-----------------------------------------------------------------------------------
Also I tried to start a videostream using IOT credentials (instead of access key, secret key);
[TRACE] [08-07-2022 19:05:08:951.959 GMT] xxx-gi iot_get_credential_endpoint: c1b70zh904bdzu.credentials.iot.us-east-2.amazonaws.com
[TRACE] [08-07-2022 19:05:08:952.000 GMT] xxx-gi cert_path                  : /mnt/plugin/.3x/certificate.pem
[TRACE] [08-07-2022 19:05:08:952.039 GMT] xxx-gi private_key_path           : /mnt/plugin/.3x/private.pem.key
[TRACE] [08-07-2022 19:05:08:952.079 GMT] xxx-gi role_alias                 : IoTDevicesRoleAlias
[TRACE] [08-07-2022 19:05:08:952.116 GMT] xxx-gi ca_cert_path               : /mnt/plugin/.3x/cacert.pem
[TRACE] [08-07-2022 19:05:08:952.156 GMT] xxx-gi data->stream_name          : 1C825919DCA2

[DEBUG] [08-07-2022 19:05:08:952.404 GMT] Creating IoT auth callbacks.
[ERROR] [08-07-2022 19:05:09:354.144 GMT] Unable to create Iot Credential provider. Error status: 0x15000027
[ERROR] [08-07-2022 19:05:09:354.520 GMT] Failed to initialize kinesis video with an exception: Unable to create Iot Credential provider. Error status: 0x15000027

And error status comes from here:
Src/credential-providers/IotCertCredentialProvider.cpp:
IotCertCredentialProvider::callback_t IotCertCredentialProvider::getCallbacks(PClientCallbacks client_callbacks)
{
    STATUS retStatus = STATUS_SUCCESS;

    LOG_DEBUG("Creating IoT auth callbacks.");
    if (STATUS_FAILED(retStatus = createIotAuthCallbacks(client_callbacks,
                           STRING_TO_PCHAR(iot_get_credential_endpoint_),
                           STRING_TO_PCHAR(cert_path_),
                           STRING_TO_PCHAR(private_key_path_),
                           STRING_TO_PCHAR(ca_cert_path_),
                           STRING_TO_PCHAR(role_alias_),
                           STRING_TO_PCHAR(stream_name_),
                           &iot_callbacks))) {
        std::stringstream status_strstrm;
        status_strstrm << std::hex << retStatus;
        LOG_AND_THROW("Unable to create Iot Credential provider. Error status: 0x" + status_strstrm.str());
    }
    return *iot_callbacks;
}

---------------------------------------------------
I have an implementation based on an “old” kvs sdk. It passes and video is successfully received on the cloud side. It uses the same camera hardware and configuration (credentials, …)

***CAUTION: EXTERNAL MESSAGE***

This message and any attachments may contain Stanley Black & Decker confidential information. Any unauthorized review, use, disclosure or distribution of any such information is prohibited. If you are not the intended recipient, please contact the sender and delete all copies.

Este mensaje y los archivos adjuntos pueden contener información confidencial de Stanley Black & Decker. Se prohíbe cualquier revisión, uso, divulgación o distribución no autorizados de dicha información. Si no es el destinatario previsto, comuníquese con el remitente y elimine todas las copias.

本邮件及其附件可能包含史丹利百得公司的保密信息。非经授权，任何人不得阅读、使用、披露或传播这种信息。若您不是收件人，请您联系发件人并删除所有内容。谢谢。

gi12345, 2022-07-08:
Below find the traces taken on the same camera platform with the same configuration with "same" application software software (only minor adjustments for KVS-SDK, KVS function calls and parameters are the same). Software uses IOT certificates.
..............................................................................
Working example with "older ~ 2years" KVS-SDK version:

[INFO ] [08-07-2022 16:16:22:224.644 GMT] Using region: us-east-2
[INFO ] [08-07-2022 16:16:22:225.107 GMT] Using IoT credentials for Kinesis Video Streams
[DEBUG] [08-07-2022 16:16:22:225.654 GMT] Creating IoT auth callbacks.
1C825919DCA2

* Connected to c1b70zh904bdzu.credentials.iot.us-east-2.amazonaws.com (13.58.55.139) port 443 (#0)
* ALPN, offering http/1.1
* successfully set certificate verify locations:
*   CAfile: /mnt/plugin/.3x/cacert.pem
  CApath: none
* SSL connection using TLSv1.2 / ECDHE-RSA-AES256-GCM-SHA384
* ALPN, server accepted to use http/1.1
* Server certificate:
*  subject: CN=*.credentials.iot.us-east-2.amazonaws.com
*  start date: Oct 30 00:00:00 2021 GMT
*  expire date: Oct  6 23:59:59 2022 GMT
*  subjectAltName: host "c1b70zh904bdzu.credentials.iot.us-east-2.amazonaws.com" matched cert's "*.credentials.iot.us-east-2.amazonaws.com"
*  issuer: C=US; O=Amazon; OU=Server CA 1B; CN=Amazon
*  SSL certificate verify ok.
> GET /role-aliases/IoTDevicesRoleAlias/credentials HTTP/1.1
Host: c1b70zh904bdzu.credentials.iot.us-east-2.amazonaws.com
Accept: */*
x-amzn-iot-thingname:1C825919DCA2

* Mark bundle as not supporting multiuse
< HTTP/1.1 200 OK
< content-type: application/json
< content-length: 1305
< date: Fri, 08 Jul 2022 16:16:22 GMT
< x-amzn-RequestId: 899f53e6-cbd3-4916-2596-e158a672a464
< 
{"credentials":{"accessKeyId":"ASIAWY3BNF3KL6WIDHSM","secretAccessKey":"4aGQ29Z9QSjphDVhyQOCLLk4/stJ/45PTae0+5Bj","sessionToken":"IQoJb3JpZ2luX2VjEMj//////////wEaCXVzLWVhc3QtMiJGMEQCIBsIoV1dIbdU4vvJ56stWeGCEXl3vR/6f3/NiJ7u+S/eAiBAnwTW8bB6QmZtxwBP7wBcTiA0+5o6Skmjw4nSmfEFliqWBAjx//////////8BEAIaDDQ2NTY3MTM2ODQwNCIMfbn8XXyJu0aKWXTrKuoDI61V6RvZ5wBTtgqO6DNc5ohfI3HtzrnBGK2nqa422thGyBKqFGE+lzniTogLtgaCEnbUht+Km+nIb8NuivdH2I/i4/eanSL6bEOrMs/RhE2uogm90TpVzxubk4DM5KVBk5yVQEYrYALndJHfbRJjF0RrWkOJ5lErhLtaxFiYFZ2bwkiYIvGoiBr7kNDZpUt9ayaPSAroQa36f/DSH5Eb1BToo7ezkmj7s/GMHr+rswgQzQMo/HO6S6qMvF9Kr8TPvo/LhtuwpfDbI+9OO09SrPUXRBmrDPUbZwF6DgTcRo9ZxW//OBYP7/NKtUc1fq3MAWw0osrQ4y+sWa8mLsr/lrw+UexAQxXC9eod7QilSUuM/8NdGlbDFHgQAxRpEtwnpzAz5ZFTctZyA27KYvk65yFU8O6tiWDPB9pUKk+2vhhZnLqCM/NvLHENEhuh4EaXpuKgLAj2o5O2ZDhPCZY3AxZuS2N4DjZg4VbzyUzdu6dlTr1Kk+D3rO6xZONVAiL2S8OHOup/W8qNqI+zJnNeFVq29YzBsvbNxOXnMPFIbSw4fm1T3/NxN7jPPkolfiuQX3WyY7bb5mKjNCO2nxAzNGsOdWVMWYJKO8MqSbvFxXrRa1HukpczzX5TaKH4zpA0qNRTlJCxw6F9jDDWsKGWBjrHARaCFHFkD/+T/8YjEPkEFQ9W9S* Connection #0 to host c1b70zh904bdzu.credentials.iot.us-east-2.amazonaws.com left intact
100  1305  100  1305    0     0   2685      0 --:--:-- --:--:-- --:--:--  2685
x3ZRZM1jadRkB2+iL8mUysqE5z28vR3CEhSS39gKoKrdYkkBC+FchrFypIM4lNMxYloMgA9AcYHMbs2hXQa6lBeO7Ea8kx5IUeJ38vjiiXpoyHtTI5lDDRYtg6u4hjflVbGCkTNReolDGAD028miu5NrQS52XfcRT7rV6QIo80etY/ATGrC6jTmyCG60g5Pt3YEREXG7SV2e1HVmkggMtPR9bhb4pNcPV+xNsBBSz3vR/Kxos=","expiration":"2022-07-08T17:16:22Z"}}

[INFO ] [08-07-2022 16:16:22:740.089 GMT] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ] [08-07-2022 16:16:22:740.237 GMT] heapInitialize(): Initializing native heap with limit size 8388608, spill ratio 0% and flags 0x00000001
[INFO ] [08-07-2022 16:16:22:740.281 GMT] heapInitialize(): Creating AIV heap.
[INFO ] [08-07-2022 16:16:22:740.377 GMT] heapInitialize(): Heap is initialized OK
[DEBUG] [08-07-2022 16:16:22:740.454 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[DEBUG] [08-07-2022 16:16:22:740.510 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
[INFO ] [08-07-2022 16:16:22:740.557 GMT] createDeviceResultEvent(): Create device result event.
[DEBUG] [08-07-2022 16:16:22:740.650 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
[DEBUG] [08-07-2022 16:16:22:740.701 GMT] clientReadyHandler invoked
[TRACE] [08-07-2022 16:16:22:740.747 GMT] createKinesisVideoClientSync(): Awaiting for the Kinesis Video Client to become ready...
[TRACE] [08-07-2022 16:16:22:740.785 GMT] createKinesisVideoClientSync(): Kinesis Video Client is Ready.
DEBUG] [08-07-2022 16:16:22:740.842 GMT] Client is ready

..............................................................................
Failure example with "newer ~ June 17,2022" KVS-SDK version:

[INFO ] [08-07-2022 19:04:38:509.342 GMT] Using region: us-east-2
[INFO ] [08-07-2022 19:04:38:509.560 GMT] Using IoT credentials for Kinesis Video Streams
xxx-gi kinesis_video_init Using IOT credentials Point10
[TRACE] [08-07-2022 19:04:38:509.614 GMT] xxx-gi iot_get_credential_endpoint: c1b70zh904bdzu.credentials.iot.us-east-2.amazonaws.com
[TRACE] [08-07-2022 19:04:38:509.656 GMT] xxx-gi cert_path                  : /mnt/plugin/.3x/certificate.pem
[TRACE] [08-07-2022 19:04:38:509.695 GMT] xxx-gi private_key_path           : /mnt/plugin/.3x/private.pem.key
[TRACE] [08-07-2022 19:04:38:509.736 GMT] xxx-gi role_alias                 : IoTDevicesRoleAlias
[TRACE] [08-07-2022 19:04:38:509.775 GMT] xxx-gi ca_cert_path               : /mnt/plugin/.3x/cacert.pem
[TRACE] [08-07-2022 19:04:38:509.813 GMT] xxx-gi data->stream_name          : 1C825919DCA2
[... cut ...]
                           STRING_TO_PCHAR(private_key_path_),
                           STRING_TO_PCHAR(ca_cert_path_),
                           STRING_TO_PCHAR(role_alias_),
                           STRING_TO_PCHAR(stream_name_),
                           &iot_callbacks))) {
        std::stringstream status_strstrm;
        status_strstrm << std::hex << retStatus;
        LOG_AND_THROW("Unable to create Iot Credential provider. Error status: 0x" + status_strstrm.str());
    }
    return *iot_callbacks;
}

jdelapla, 2022-07-10:
> Hello, Thanks for you feedback. I am not sure how to apply your recommended fix: ***@***.***:~/kvssdk_jun17/amazon-kinesis-video-streams-producer-sdk-cpp$ diff --git [...]

github doesn't allow me to directly attach a .patch file. Copy the code section in my comment and put it into a your.patch file.
From the top of master enter the following:
`git apply path/to/your.patch`
Substituting path/to/your.patch with the actual path to the patch file you've created.

Please apply this patch and try again with debug level logging and post the logs here.
I understand an older version work well for you, however I suspect in the current version your setup is having errors with jsmn, these logs would confirm that and we can proceed from there.

gi12345, 2022-07-11:
Hello,
I can’t see your patch modifications.
If you please could send me the source code changes and I will apply hem manually.
For building I hope a make clean; make will do.
Thanks

jdelapla, 2022-07-11:
```
diff --git a/CMake/Dependencies/libkvscproducer-CMakeLists.txt b/CMake/Dependencies/libkvscproducer-CMakeLists.txt
index 6c4490e..96e0a5e 100644
--- a/CMake/Dependencies/libkvscproducer-CMakeLists.txt
+++ b/CMake/Dependencies/libkvscproducer-CMakeLists.txt
@@ -7,7 +7,7 @@ include(ExternalProject)
 # clone repo only
 ExternalProject_Add(libkvscproducer-download
 	GIT_REPOSITORY    https://github.com/awslabs/amazon-kinesis-video-streams-producer-c.git
-	GIT_TAG           79da5b6d6ad67560cc08530f967e9187a6d1c594
+	GIT_TAG           d5a0455b752d1a3149b022d734dd8c7f759d308e
 	SOURCE_DIR        "${CMAKE_CURRENT_BINARY_DIR}/kvscproducer-src"
 	BINARY_DIR        "${CMAKE_CURRENT_BINARY_DIR}/kvscproducer-build"
 	CONFIGURE_COMMAND ""
```

in CMake/Dependencies/libkvscproducer-CMakelists.txt change GIT_TAG to `d5a0455b752d1a3149b022d734dd8c7f759d308e`

gi12345, 2022-07-11:
diff --git a/CMake/Dependencies/libkvscproducer-CMakeLists.txt b/CMake/Dependencies/libkvscproducer-CMakeLists.txt
diff: unrecognized option '--git'
diff: Try 'diff --help' for more information.
--------------------------------
This does not work for my environment; I don’t have such a setup.
If you give the code to be changed I can rebuild that manually.
Thanks,

gi12345, 2022-07-12:
1. I applied the patch you sent and modified the stream_definition code:

    unique_ptr<StreamDefinition> stream_definition(new StreamDefinition(
         data->stream_name,
         hours(data->retention_hours),
         &tags,
         DEFAULT_KMS_KEY_ID,
         streaming_type,
         DEFAULT_CONTENT_TYPE,
         duration_cast<milliseconds> (seconds(DEFAULT_MAX_LATENCY_SECONDS)),
         milliseconds(DEFAULT_FRAGMENT_DURATION_MILLISECONDS),
         milliseconds(DEFAULT_TIMECODE_SCALE_MILLISECONDS),
         DEFAULT_KEY_FRAME_FRAGMENTATION,
         DEFAULT_FRAME_TIMECODES,
         data->use_absolute_fragment_times,
         DEFAULT_FRAGMENT_ACKS,
         DEFAULT_RESTART_ON_ERROR,
         DEFAULT_RECALCULATE_METRICS,
-        0,
+        NAL_ADAPTATION_FLAG_NONE,
         DEFAULT_STREAM_FRAMERATE,
         DEFAULT_AVG_BANDWIDTH_BPS,
         seconds(DEFAULT_BUFFER_DURATION_SECONDS),
         seconds(DEFAULT_REPLAY_DURATION_SECONDS),
         seconds(DEFAULT_CONNECTION_STALENESS_SECONDS),
         DEFAULT_CODEC_ID,
         DEFAULT_TRACKNAME,
         nullptr,
-        0));
+        0,
+        MKV_TRACK_INFO_TYPE_VIDEO,
+        vector<uint8_t>(),
+        1));
+

Error-Code 0x52000014 no longer shows up.

......................................................................................

2. When I use IOT credential for authentication I still get this error: "Failed to initialize kinesis video with an exception: Unable to create Iot Credential provider. Error status: 0x15000027".

Code section:
    data->kinesis_video_producer = KinesisVideoProducer::createSync(move(device_info_provider),
                                                                    move(client_callback_provider),
                                                                    move(stream_callback_provider),
                                                                    move(credential_provider),
                                                                    defaultRegionStr);

Log:
[INFO ] [12-07-2022 14:58:49:872.080 GMT] Using region: us-east-2
[INFO ] [12-07-2022 14:58:49:872.274 GMT] Using IoT credentials for Kinesis Video Streams
[TRACE] [12-07-2022 14:58:49:872.467 GMT] xxx-gi iot_get_credential_endpoint: c1b70zh904bdzu.credentials.iot.us-east-2.amazonaws.com
[TRACE] [12-07-2022 14:58:49:872.510 GMT] xxx-gi cert_path                  : /mnt/plugin/.3x/certificate.pem
[TRACE] [12-07-2022 14:58:49:874.059 GMT] xxx-gi private_key_path           : /mnt/plugin/.3x/private.pem.key
[TRACE] [12-07-2022 14:58:49:874.110 GMT] xxx-gi role_alias                 : IoTDevicesRoleAlias
[TRACE] [12-07-2022 14:58:49:874.152 GMT] xxx-gi ca_cert_path               : /mnt/plugin/.3x/cacert.pem
[TRACE] [12-07-2022 14:58:49:874.193 GMT] xxx-gi data->stream_name          : 1C825919DCA2
1C825919DCA2
Trying 3.130.143.195:443...
* TCP_NODELAY set
* Connected to c1b70zh904bdzu.credentials.iot.us-east-2.amazonaws.com (3.130.143.195) port 443 (#0)
* ALPN, offering http/1.1
* successfully set certificate verify locations:
*   CAfile: /mnt/plugin/.3x/cacert.pem
  CApath: /etc/ssl/certs
* SSL connection using TLSv1.2 / ECDHE-RSA-AES256-GCM-SHA384
* ALPN, server accepted to use http/1.1
* Server certificate:
*  subject: CN=*.credentials.iot.us-east-2.amazonaws.com
*  start date: Oct 30 00:00:00 2021 GMT
*  expire date: Oct  6 23:59:59 2022 GMT
*  subjectAltName: host "c1b70zh904bdzu.credentials.iot.us-east-2.amazonaws.com" matched cert's "*.credentials.iot.us-east-2.amazonaws.com"
*  issuer: C=US; O=Amazon; OU=Server CA 1B; CN=Amazon
*  SSL certificate verify ok.
> GET /role-aliases/IoTDevicesRoleAlias/credentials HTTP/1.1
Host: c1b70zh904bdzu.credentials.iot.us-east-2.amazonaws.com
Accept: */*
x-amzn-iot-thingname:1C825919DCA2

* Mark bundle as not supporting multiuse
< HTTP/1.1 200 OK
< content-type: application/json
< content-length: 1281
< date: Tue, 12 Jul 2022 14:58:50 GMT
< x-amzn-RequestId: 35a5d8cf-8ec8-ae4f-4d1a-31acce9630c4
< 
{"credentials":{"accessKeyId":"ASIAWY3BNF3KNTXL4WBX","secretAccessKey":"aj9ekILbpFMD3CTD42yZGjAAjFbCmSrnhV8J1uTN","sessionToken":"IQoJb3JpZ2luX2VjECcaCXVzLWVhc3QtMiJHMEUCIQC8vyxIjrvC6deHbF/oGgjyMM5YEGkpcpbeKED1Reb+8QIgGkOvBTfr2CHoANA4EMlYEDlWNjKUYPtf2v5jLkD0gZEqjQQIYBACGgw0NjU2NzEzNjg0MDQiDKTbyENLjgbHRkZETSrqA77Y5idoBRA2TQwYUeC9AZz8v+TRxcrg8jPx8M4N/EMGlB1F+XPy16/hVgpoOxmM54oBeejR6R1IajSuS7nFj61wYG7Fz3i2z4BwS5zAH0+d7w0YeaW48fv8HMoNffRGN7UzEhyGitw2CTy7BTBZsBafHabdVHTyD8nBxOEbrJ2uHgaTA1rKen5uRoVlGVCgwkDIjCgKzMKAn69ohEg57JBlGJ/F5LKj6Bu9fZOpzyvs7gXWd1oVLc1ZuOKfw19cBetNZ/GTO6RUM+c322SWCeZPohXucJaBJsLYh/wYGArqHbjExXka+NNsUv4MZf5n9rn7EzZB8cs9r2J4cUEBRF0J0ApIw1rMp+1hbBSgI8ha5IDvpem+yQ7eHHicdOnpCM1FhE6NfEf85onxT+JEp5QxPChpKIXGwP4KKbGDD/tkfakKs55VS7jf9dj6CWa4fsiGgrk/ABT55pVJjyvomIzeWvB+womVkWuqMpREr9fLTV+ybtRr7kuipM5htMiScMiM4Fbo8cQvDZmEPm9oMrhW+PUoyMhEQf9dXTa9+fW/wO9Ys32RhXN0GyeQwcvqjo+xS0g9r/Ebr7PStxlFNPr+BDNljp6/SNFsybHfSYqjF59uutv/tCFGU0FzqqLM70MpLGzLuKzQmCAwqpi2lgY6xgETIHGBWHkn/e41v6CPH96qiev6ciGPXJTyPrF80D11lDomi3* Connection #0 to host c1b70zh904bdzu.credentials.iot.us-east-2.amazonaws.com left intact
xW+Vgpsrpuon7Tl4Cs07Gn9qqUUqNoFdvmKC+kOTRxisbfs+5LcyeRPeRwUkpFuK17AJxJgbhxo/qgBeRDhhFN1f43kJIruDA73Suz9eSmuZkELtGXPcMQbRgtl/vRyndVfUXM6qf5I4IiMqsAtmu1gpeCJ+OVfCKVNsYuxIrDrV8+VVUU9vy1ZcpH6WRM3lLcVsDLz8FAGmm03gBv7pbp6UM=","expiration":"2022-07-12T15:58:50Z"}}
[ERROR] [12-07-2022 14:58:50:353.840 GMT] Unable to create Iot Credential provider. Error status: 0x15000027
[ERROR] [12-07-2022 14:58:50:360.160 GMT] Failed to initialize kinesis video with an exception: Unable to create Iot Credential provider. Error status: 0x15000027

hassanctech, 2022-07-18:
The error you see is `STATUS_CLIENT_BEING_SHUTDOWN` can you please provide complete logging, **please put in a file and attach, do not put in the comments as it is very difficult to read**

gi12345, 2022-07-18:
requested file atached.
[qqq1.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/9135868/qqq1.txt)

gi12345, 2022-07-20:
Any update on that? Thanks.

gi12345, 2022-07-20:
I agree with your assessment: "This is indicative of an invalid IoT response being generated or there is a parser error."
I ran the response through a json-online parser and it signaled failure in session key.
I have run my software on 2 devices located in 2 different cities (so different networks), both create the same error.
I have an older software running with an older kvs-sdk (>2 years) and it works. But I checked the json-response and the online parser also flagged error in the session key. 
So, it seems, either the session key is invalid json or the parser cannot parse the key.

MushMal, 2022-07-21:
Is there a way you can obfuscate some of the A-Za-z characters in the JSON token and paste it here? AWS folks might want to check whether it's something funny returned from the IoT

gi12345, 2022-07-21:
I have not found any specific pattern when I tried to dissent the credential reply. The sessionToken seems to use / to "concatenate" different pieces, but nothing special I can derive from the pieces.

Here is the failed credential:
{"credentials":
   {"accessKeyId":"ASIAWY3BNF3KORCD42GD",
    "secretAccessKey":"zUdjYPTTDnbKWZZinLecMS3riZ/3pbDPEiwbOlGv", 
"sessionToken":"IQoJb3JpZ2luX2VjEL3//////////wEaCXVzLWVhc3QtMiJGMEQCIHEwWc8lUIvTMEdXc9jdmKFJuxqyEt7jnzPCrLuRXAbhAiB+HFJwSrlcpI4Upao8qwQsPCA7M9uQoUyKExn8eZVubiqWBAj2//////////
8BEAIaDDQ2NTY3MTM2ODQwNCIMEMO22EmEKTedt4uIKuoDRXCpHhZsjnm77M30KGCntIdmB25z8HQvtM5ogu5AQtSewFTOQtJSaSVKhz9Rw2rtgv6YEBE+8s1NdPCoheHeLk1q0OaS/
lL5ss9JbmVQy3/
JuipU5mKSx9kkG0g79V4AJskxNM/
misyhOMLTycGeFtq+Ps/
Pur8YOO8wvNz107U+jfbq7490y29rQPTigXtQWajl1vE4Y9GRpHd8XtifByYiE70cKHZhWlqebpvb+c+bg3UX/
QRPrOdim/
cUtUOVEkpPvxNf4cf7QhLIpXL3Yrr/
PPG6CBCuuyvjBZUj+Notw6io1GAVySShTUAPlQxgWf0cX1KnUBXD52/
FzfPPjpS0ouL4xnj+84hh0W/
2EE3+fPu3cYjOmyHjjAibE/
bKHk2sydbQtNvvTb6eBRxDSbBAL2gFXRJnR3+bVmkXv8KULFSIlzpB4+e5nugrLhdVTviwCDX9IcF5w5fVuQDFI6JLe2PAP6T3PEhYvC02n0mXQt/
0BiDjGbmIuSZjVeU89Yfi/
U/
CRu6VQdNsHEE/
Chz2kUOB4T9Nl9I6L8Ky1x5p3LrsRIadqucY3gMVrKkNNk7V4keui24rQqjbk3sfgOLmn3jOH/
WZuhKg9hQ3DjhGEtD2u3HjuRb2N8ruooBuGsOmDKO3l9f9aTC6mNeWBjrHAR0JUAZO1xStgWi4zvHhkJonhrlQiMisKzATgfqjf5u1J1jzcFMw1LcB46Lhbn0OTTzYoAvjro+6FBVyahYJw1tmybMn/I9CqoiYzuzCRM4itQPMNhoYEYq96XOo+eFrGi4Dm2vNCzhuN+YWzqWLXHqicjbrXPjVG9Jv0wQlE6+6ltZ62fGoBq59o95rF8iER07IT4PXI+ZN/
BQhOA5+hjaN+2Te9q9uPCcubu8fizj/Wbd4cbSiWsgQGVDZhmqkAbeOueZi4Dc=",
    "expiration":"2022-07-18T22:10:18Z"}}

And here is one which passed in an old version. Again, slash seems to be used as some "concatenator" and the pieces seem to be similar to the pieces from the failed token above; but the contents is all different:
{"credentials":
   {"accessKeyId":"ASIAWY3BNF3KL6WIDHSM",
    "secretAccessKey":"4aGQ29Z9QSjphDVhyQOCLLk4/stJ/45PTae0+5Bj",
    "sessionToken":"IQoJb3JpZ2luX2VjEMj//////////
wEaCXVzLWVhc3QtMiJGMEQCIBsIoV1dIbdU4vvJ56stWeGCEXl3vR/
6f3/
NiJ7u+S/
eAiBAnwTW8bB6QmZtxwBP7wBcTiA0+5o6Skmjw4nSmfEFliqWBAjx//////////8BEAIaDDQ2NTY3MTM2ODQwNCIMfbn8XXyJu0aKWXTrKuoDI61V6RvZ5wBTtgqO6DNc5ohfI3HtzrnBGK2nqa422thGyBKqFGE+lzniTogLtgaCEnbUht+Km+nIb8NuivdH2I/
i4/
eanSL6bEOrMs/
RhE2uogm90TpVzxubk4DM5KVBk5yVQEYrYALndJHfbRJjF0RrWkOJ5lErhLtaxFiYFZ2bwkiYIvGoiBr7kNDZpUt9ayaPSAroQa36f/
DSH5Eb1BToo7ezkmj7s/
GMHr+rswgQzQMo/
HO6S6qMvF9Kr8TPvo/
LhtuwpfDbI+9OO09SrPUXRBmrDPUbZwF6DgTcRo9ZxW//
OBYP7/
NKtUc1fq3MAWw0osrQ4y+sWa8mLsr/
lrw+UexAQxXC9eod7QilSUuM/
8NdGlbDFHgQAxRpEtwnpzAz5ZFTctZyA27KYvk65yFU8O6tiWDPB9pUKk+2vhhZnLqCM/
NvLHENEhuh4EaXpuKgLAj2o5O2ZDhPCZY3AxZuS2N4DjZg4VbzyUzdu6dlTr1Kk+D3rO6xZONVAiL2S8OHOup/
W8qNqI+zJnNeFVq29YzBsvbNxOXnMPFIbSw4fm1T3/
NxN7jPPkolfiuQX3WyY7bb5mKjNCO2nxAzNGsOdWVMWYJKO8MqSbvFxXrRa1HukpczzX5TaKH4zpA0qNRTlJCxw6F9jDDWsKGWBjrHARaCFHFkD/
+T/
8YjEPkEFQ9W9Sx3ZRZM1jadRkB2+iL8mUysqE5z28vR3CEhSS39gKoKrdYkkBC+FchrFypIM4lNMxYloMgA9AcYHMbs2hXQa6lBeO7Ea8kx5IUeJ38vjiiXpoyHtTI5lDDRYtg6u4hjflVbGCkTNReolDGAD028miu5NrQS52XfcRT7rV6QIo80etY/ATGrC6jTmyCG60g5Pt3YEREXG7SV2e1HVmkggMtPR9bhb4pNcPV+xNsBBSz3vR/
Kxos=",
    "expiration":"2022-07-08T17:16:22Z"}
}

MushMal, 2022-07-21:
JSON validators fail to parse this excerpt on LN 4. It seems that the multiple '/' backslashes break the parsers.

I think AWS IoT folks should comment on this.

gi12345, 2022-07-27:
Any progress on this. Neither IOT certificate nor private key is working. It's a showstopper for the entre project.

gi12345, 2022-08-02:
Still nothing working. Any update o this please?

xiaoyaodingchunqiu, 2022-08-23:
> Still nothing working. Any update o this please?

Is your problem solved？
I have the same problem after compiling and running on rock pi
If there is any progress, please reply
Thanks

gi12345, 2022-08-23:
[ERROR] [12-07-2022 14:58:50:353.840 GMT] Unable to create Iot Credential provider. Error status: 0x15000027
[ERROR] [12-07-2022 14:58:50:360.160 GMT] Failed to initialize kinesis video with an exception: Unable to create Iot Credential provider. Error status: 0x15000027

The problem still exists and is a showstopper for us. No reply from AWS.
I added extensive tracing to AWS SDK and the json buffer format and their parameters seem fine. Even the reply from the cloud side seems fine, it's the error code which is interpreted and causes error handling logic.

The curl request looks like this:
[DEBUG] [22-08-2022 22:12:19:853.667 GMT] describeStreamCurlHandler(): xxx-gi executing curl requests. Len=33
[DEBUG] [22-08-2022 22:12:19:853.816 GMT] describeStreamCurlHandler(): xxx-gi Curl-Buf={
	"StreamName": "1C825919DCA2"
}
[DEBUG] [22-08-2022 22:12:19:854.009 GMT] describeStreamCurlHandler(): xxx-gi Curl-HEX=7B0A0922.53747265.616D4E61.6D65223A.20223143.38323539.31394443.4132220A.7D

And the curl reply like this:
[DEBUG] [22-08-2022 22:12:20:241.871 GMT] describeStreamCurlHandler(): xxx-gi DescribeStream API response: Len=391 Val={"StreamInfo":{"CreationTime":1.657031570327E9,"DataRetentionInHours":8760,"DeviceName":null,"IngestionConfiguration":null,"KmsKeyId":"arn:aws:kms:us-east-2:465671368404:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-east-2:465671368404:stream/1C825919DCA2/1657031570327","StreamName":"1C825919DCA2","Version":"M3HjvO1mtiB53dBxOE23"}}
[DEBUG] [22-08-2022 22:12:20:242.118 GMT] describeStreamCurlHandler(): xxx-gi DescribeStream API response: TokenCount=23

The parser detects 23 tokens which seems correct. 

It enters the cleanup handler like this:
[INFO ] [22-08-2022 22:12:20:243.621 GMT] describeStreamResultEvent(): Describe stream result event.
[ERROR] [22-08-2022 22:12:20:243.837 GMT] describeStreamResultEvent(): operation returned status code: 0x52000014

I don't think the error is json parser related. It looks more like an error between network buffer and json layer on AWS cloud side.

Error code 0x52000014 (AWS error description):
The backend returns this error when a service call argument is not valid or when the SDK encounters an error that it can't interpret.

jdelapla, 2023-01-20:
We cannot reproduce your issue. We recommend you do a complete clean of all open source dependencies are try again. There is not a JSON parser error on the cloud side.

closed 2023-01-20 by jdelapla