awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #875: rtsp streaming not working 0:00:17.400255646  4794 0x7f7a38012000 INFO               rtpsource rtpsource.c:1183:update_receiver_stats: duplicate or reordered packet (seqnr 59053, expected 59057)

BADAL244, 2022-06-30:
0:00:03.987585767  4794 0x561e8f6feb60 INFO              GST_STATES gstelement.c:2575:_priv_gst_element_state_changed:<udpsrc1> notifying about state-changed PAUSED to PLAYING (VOID_PENDING pending)
0:00:03.987630824  4794 0x561e8f6feb60 INFO                 basesrc gstbasesrc.c:1373:gst_base_src_do_seek:<udpsrc2> seeking: time segment start=0:00:00.000000000, offset=0:00:00.000000000, stop=99:99:99.999999999, rate=1.000000, applied_rate=1.000000, flags=0x00, time=0:00:00.000000000, base=0:00:00.000000000, position 0:00:00.000000000, duration 99:99:99.999999999
0:00:03.987657318  4794 0x561e8f6feb60 INFO                    task gsttask.c:453:gst_task_set_lock: setting stream lock 0x561e8fb458d0 on task 0x561e8fb46950
0:00:03.987672548  4794 0x561e8f6feb60 INFO                GST_PADS gstpad.c:6159:gst_pad_start_task:<udpsrc2:src> created task 0x561e8fb46950
0:00:03.987673770  4794 0x7f7a38012000 INFO               GST_EVENT gstevent.c:820:gst_event_new_caps: creating caps event application/x-rtp, media=(string)video, payload=(int)96, clock-rate=(int)90000, encoding-name=(string)H264, packetization-mode=(string)1, a-recvonly=(string)"", npt-start=(guint64)0, play-speed=(double)1, play-scale=(double)1
0:00:03.987759513  4794 0x561e8f6feb60 INFO              GST_STATES gstelement.c:2647:gst_element_continue_state:<udpsrc2> committing state from READY to PAUSED, pending PLAYING, next PLAYING
0:00:03.987778841  4794 0x561e8f6feb60 INFO              GST_STATES gstelement.c:2575:_priv_gst_element_state_changed:<udpsrc2> notifying about state-changed READY to PAUSED (PLAYING pending)
0:00:03.987859413  4794 0x561e8f6feb60 INFO              GST_STATES gstelement.c:2655:gst_element_continue_state:<udpsrc2> continue state change PAUSED to PLAYING, final PLAYING
0:00:03.987879758  4794 0x561e8f6feb60 INFO              GST_STATES gstelement.c:2675:gst_element_continue_state:<udpsrc2> completed state change to PLAYING
0:00:03.987896229  4794 0x561e8f6feb60 INFO              GST_STATES gstelement.c:2575:_priv_gst_element_state_changed:<udpsrc2> notifying about state-changed PAUSED to PLAYING (VOID_PENDING pending)
Progress: (request) Sent PLAY request
0:00:03.988177605  4794 0x7f7a38012060 INFO        GST_ELEMENT_PADS gstelement.c:912:gst_element_get_static_pad: no such pad 'sink' in element "udpsrc2"
0:00:03.988283199  4794 0x7f7a38012060 INFO               GST_EVENT gstevent.c:820:gst_event_new_caps: creating caps event application/x-rtcp
0:00:04.566558202  4794 0x7f7a38012000 INFO               GST_EVENT gstevent.c:900:gst_event_new_segment: creating segment event time segment start=0:00:00.000000000, offset=0:00:00.000000000, stop=99:99:99.999999999, rate=1.000000, applied_rate=1.000000, flags=0x00, time=0:00:00.000000000, base=0:00:00.000000000, position 0:00:00.000000000, duration 99:99:99.999999999
0:00:04.566705575  4794 0x7f7a38012000 INFO                 basesrc gstbasesrc.c:2962:gst_base_src_loop:<udpsrc1> marking pending DISCONT
0:00:04.567400599  4794 0x7f7a38012000 INFO               GST_EVENT gstevent.c:820:gst_event_new_caps: creating caps event application/x-rtp, media=(string)video, payload=(int)96, clock-rate=(int)90000, encoding-name=(string)H264, packetization-mode=(string)1, a-recvonly=(string)"", npt-start=(guint64)0, play-speed=(double)1, play-scale=(double)1, ssrc=(uint)1072899299
0:00:04.567468984  4794 0x7f7a38012000 INFO               GST_EVENT gstevent.c:820:gst_event_new_caps: creating caps event application/x-rtcp, ssrc=(uint)1072899299
0:00:04.567511108  4794 0x7f7a38012000 INFO        GST_ELEMENT_PADS gstelement.c:668:gst_element_add_pad:<rtpssrcdemux0> adding pad 'src_1072899299'
0:00:04.567553948  4794 0x7f7a38012000 INFO        GST_ELEMENT_PADS gstelement.c:668:gst_element_add_pad:<rtpssrcdemux0> adding pad 'rtcp_src_1072899299'
0:00:04.567601000  4794 0x7f7a38012000 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:361:gst_element_factory_create: creating element "rtpjitterbuffer"
0:00:04.568038164  4794 0x7f7a38012000 INFO        GST_ELEMENT_PADS gstelement.c:668:gst_element_add_pad:<GstRtpJitterBuffer@0x7f7a2c01de30> adding pad 'src'
0:00:04.568076861  4794 0x7f7a38012000 INFO        GST_ELEMENT_PADS gstelement.c:668:gst_element_add_pad:<GstRtpJitterBuffer@0x7f7a2c01de30> adding pad 'sink'
0:00:04.568115668  4794 0x7f7a38012000 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:361:gst_element_factory_create: creating element "rtpptdemux"
0:00:04.568339556  4794 0x7f7a38012000 INFO        GST_ELEMENT_PADS gstelement.c:668:gst_element_add_pad:<GstRtpPtDemux@0x7f7a2c024070> adding pad 'sink'
0:00:04.568648265  4794 0x7f7a38012000 INFO        GST_ELEMENT_PADS gstutils.c:1771:gst_element_link_pads_full: trying to link element rtpjitterbuffer0:src to element rtpptdemux0:sink
0:00:04.568685751  4794 0x7f7a38012000 INFO        GST_ELEMENT_PADS gstelement.c:915:gst_element_get_static_pad: found pad rtpjitterbuffer0:src
0:00:04.568709116  4794 0x7f7a38012000 INFO        GST_ELEMENT_PADS gstelement.c:915:gst_element_get_static_pad: found pad rtpptdemux0:sink
0:00:04.568742137  4794 0x7f7a38012000 INFO                GST_PADS gstutils.c:1587:prepare_link_maybe_ghosting: rtpjitterbuffer0 and rtpptdemux0 in same bin, no need for ghost pads
0:00:04.568783307  4794 0x7f7a38012000 INFO                GST_PADS gstpad.c:2377:gst_pad_link_prepare: trying to link rtpjitterbuffer0:src and rtpptdemux0:sink
0:00:04.568829530  4794 0x7f7a38012000 INFO                GST_PADS gstpad.c:2585:gst_pad_link_full: linked rtpjitterbuffer0:src and rtpptdemux0:sink, successful
0:00:04.568858011  4794 0x7f7a38012000 INFO               GST_EVENT gstevent.c:1579:gst_event_new_reconfigure: creating reconfigure event
0:00:04.568882312  4794 0x7f7a38012000 INFO               GST_EVENT gstpad.c:5812:gst_pad_send_event_unchecked:<rtpjitterbuffer0:src> Received event on flushing pad. Discarding
0:00:04.568943885  4794 0x7f7a38012000 INFO              GST_STATES gstelement.c:2647:gst_element_continue_state:<rtpptdemux0> committing state from NULL to READY, pending PLAYING, next PAUSED
0:00:04.568977384  4794 0x7f7a38012000 INFO              GST_STATES gstelement.c:2575:_priv_gst_element_state_changed:<rtpptdemux0> notifying about state-changed NULL to READY (PLAYING pending)
0:00:04.569051434  4794 0x7f7a38012000 INFO              GST_STATES gstelement.c:2655:gst_element_continue_state:<rtpptdemux0> continue state change READY to PAUSED, final PLAYING
0:00:04.569101354  4794 0x7f7a38012000 INFO              GST_STATES gstelement.c:2647:gst_element_continue_state:<rtpptdemux0> committing state from READY to PAUSED, pending PLAYING, next PLAYING
0:00:04.569163672  4794 0x7f7a38012000 INFO              GST_STATES gstelement.c:2575:_priv_gst_element_state_changed:<rtpptdemux0> notifying about state-changed READY to PAUSED (PLAYING pending)
0:00:04.569227874  4794 0x7f7a38012000 INFO              GST_STATES gstelement.c:2655:gst_element_continue_state:<rtpptdemux0> continue state change PAUSED to PLAYING, final PLAYING
0:00:04.569272544  4794 0x7f7a38012000 INFO              GST_STATES gstelement.c:2675:gst_element_continue_state:<rtpptdemux0> completed state change to PLAYING
0:00:04.569300639  4794 0x7f7a38012000 INFO              GST_STATES gstelement.c:2575:_priv_gst_element_state_changed:<rtpptdemux0> notifying about state-changed PAUSED to PLAYING (VOID_PENDING pending)
0:00:04.569363135  4794 0x7f7a38012000 INFO              GST_STATES gstelement.c:2647:gst_element_continue_state:<rtpjitterbuffer0> committing state from NULL to READY, pending PLAYING, next PAUSED
0:00:04.569403311  4794 0x7f7a38012000 INFO              GST_STATES gstelement.c:2575:_priv_gst_element_state_changed:<rtpjitterbuffer0> notifying about state-changed NULL to READY (PLAYING pending)
0:00:04.569465096  4794 0x7f7a38012000 INFO              GST_STATES gstelement.c:2655:gst_element_continue_state:<rtpjitterbuffer0> continue state change READY to PAUSED, final PLAYING
0:00:04.569662477  4794 0x7f7a38012000 INFO                    task gsttask.c:453:gst_task_set_lock: setting stream lock 0x7f7a2c01e100 on task 0x7f7a340064d0
0:00:04.569699783  4794 0x7f7a38012000 INFO                GST_PADS gstpad.c:6159:gst_pad_start_task:<rtpjitterbuffer0:src> created task 0x7f7a340064d0
0:00:04.569969148  4794 0x7f7a38012000 INFO              GST_STATES gstelement.c:2647:gst_element_continue_state:<rtpjitterbuffer0> committing state from READY to PAUSED, pending PLAYING, next PLAYING
[... cut ...]
0:00:27.531102278  4794 0x561e8f70d800 INFO              GST_STATES gstelement.c:2675:gst_element_continue_state:<rtspsrc0> completed state change to NULL
0:00:27.531119715  4794 0x561e8f70d800 INFO              GST_STATES gstelement.c:2575:_priv_gst_element_state_changed:<rtspsrc0> notifying about state-changed READY to NULL (VOID_PENDING pending)
0:00:27.531137796  4794 0x561e8f70d800 INFO              GST_STATES gstbin.c:2952:gst_bin_change_state_func:<pipeline0> child 'rtspsrc0' changed state to 1(NULL) successfully
0

BADAL244, 2022-07-01:
0:00:38.922631461  4064 0x7f5f5c012000 INFO               rtpsource rtpsource.c:1183:update_receiver_stats: duplicate or reordered packet (seqnr 5165, expected 5173)
0:00:38.922773452  4064 0x7f5f5c012000 INFO               rtpsource rtpsource.c:1183:update_receiver_stats: duplicate or reordered packet (seqnr 5166, expected 5173)
0:00:38.922869248  4064 0x7f5f5c012000 INFO               rtpsource rtpsource.c:1183:update_receiver_stats: duplicate or reordered packet (seqnr 5167, expected 5173)
0:00:38.922955215  4064 0x7f5f5c012000 INFO               rtpsource rtpsource.c:1183:update_receiver_stats: duplicate or reordered packet (seqnr 5168, expected 5173)
0:00:38.928532442  4064 0x7f5f5c012000 INFO               rtpsource rtpsource.c:1183:update_receiver_stats: duplicate or reordered packet (seqnr 5169, expected 5173)
0:00:38.928684265  4064 0x7f5f5c012000 INFO               rtpsource rtpsource.c:1183:update_receiver_stats: duplicate or reordered packet (seqnr 5170, expected 5173)
0:00:38.928763819  4064 0x7f5f5c012000 INFO               rtpsource rtpsource.c:1183:update_receiver_stats: duplicate or reordered packet (seqnr 5171, expected 5173)
0:00:41.920779012  4064 0x7f5f5c012000 INFO               rtpsource rtpsource.c:1183:update_receiver_stats: duplicate or reordered packet (seqnr 5747, expected 5751)
0:00:41.920949684  4064 0x7f5f5c012000 INFO               rtpsource rtpsource.c:1183:update_receiver_stats: duplicate or reordered packet (seqnr 5748, expected 5751)
0:00:41.921075816  4064 0x7f5f5c012000 INFO               rtpsource rtpsource.c:1183:update_receiver_stats: duplicate or reordered packet (seqnr 5749, expected 5751)
0:00:52.480258204  4064 0x7f5f5c012000 INFO               rtpsource rtpsource.c:1183:update_receiver_stats: duplicate or reordered packet (seqnr 7811, expected 7816)
0:00:52.480479033  4064 0x7f5f5c012000 INFO               rtpsource rtpsource.c:1183:update_receiver_stats: duplicate or reordered packet (seqnr 7812, expected 7816)
0:00:52.480645697  4064 0x7f5f5c012000 INFO               rtpsource rtpsource.c:1183:update_receiver_stats: duplicate or reordered packet (seqnr 7813, expected 7816)
0:00:54.729681155  4064 0x7f5f5c012000 INFO               rtpsource rtpsource.c:1183:update_receiver_stats: duplicate or reordered packet (seqnr 8275, expected 8279)
0:00:54.729877204  4064 0x7f5f5c012000 INFO               rtpsource rtpsource.c:1183:update_receiver_stats: duplicate or reordered packet (seqnr 8276, expected 8279)
0:00:57.381608189  4064 0x7f5f5c012000 INFO               rtpsource rtpsource.c:1183:update_receiver_stats: duplicate or reordered packet (seqnr 8755, expected 8759)
0:00:57.381969192  4064 0x7f5f5c012000 INFO               rtpsource rtpsource.c:1183:update_receiver_stats: duplicate or reordered packet (seqnr 8756, expected 8759)
0:00:57.919540869  4064 0x7f5f5c012000 INFO               rtpsource rtpsource.c:1183:update_receiver_stats: duplicate or reordered packet (seqnr 8860, expected 8863)
0:00:57.919691454  4064 0x7f5f5c012000 INFO               rtpsource rtpsource.c:1183:update_receiver_stats: duplicate or reordered packet (seqnr 8861, expected 8863)
0:00:58.046033760  4064 0x7f5f5c012000 INFO               rtpsource rtpsource.c:1183:update_receiver_stats: duplicate or reordered packet (seqnr 8883, expected 8886)
0:00:58.046171760  4064 0x7f5f5c012000 INFO               rtpsource rtpsource.c:1183:update_receiver_stats: duplicate or reordered packet (seqnr 8884, expected 8886)
0:00:58.257952402  4064 0x7f5f5c012000 INFO               rtpsource rtpsource.c:1183:update_receiver_stats: duplicate or reordered packet (seqnr 8922, expected 8926)
0:00:58.258141742  4064 0x7f5f5c012000 INFO               rtpsource rtpsource.c:1183:update_receiver_stats: duplicate or reordered packet (seqnr 8923, expected 8926)
0:00:58.493329406  4064 0x7f5f5c012000 INFO               rtpsource rtpsource.c:1183:update_receiver_stats: duplicate or reordered packet (seqnr 8973, expected 8978)
0:00:58.493538579  4064 0x7f5f5c012000 INFO               rtpsource rtpsource.c:1183:update_receiver_stats: duplicate or reordered packet (seqnr 8974, expected 8978)
0:00:58.751900961  4064 0x7f5f5c012000 INFO               rtpsource rtpsource.c:1183:update_receiver_stats: duplicate or reordered packet (seqnr 9020, expected 9024)
0:00:58.930885964  4064 0x7f5f5c012000 INFO               rtpsource rtpsource.c:1183:update_receiver_stats: duplicate or reordered packet (seqnr 9054, expected 9065)
0:00:58.931051332  4064 0x7f5f5c012000 INFO               rtpsource rtpsource.c:1183:update_receiver_stats: duplicate or reordered packet (seqnr 9055, expected 9065)
0:00:58.931144167  4064 0x7f5f5c012000 INFO               rtpsource rtpsource.c:1183:update_receiver_stats: duplicate or reordered packet (seqnr 9056, expected 9065)
0:01:01.350502438  4064 0x7f5f5c012000 INFO               rtpsource rtpsource.c:1183:update_receiver_stats: duplicate or reordered packet (seqnr 9534, expected 9537)
0:01:09.420984931  4064 0x7f5f5c012000 INFO               rtpsource rtpsource.c:1183:update_receiver_stats: duplicate or reordered packet (seqnr 11152, expected 11155)
0:01:09.421217238  4064 0x7f5f5c012000 INFO               rtpsource rtpsource.c:1183:update_receiver_stats: duplicate or reordered packet (seqnr 11153, expected 11155)
0:01:09.517055738  4064 0x7f5f5c012000 INFO               rtpsource rtpsource.c:1183:update_receiver_stats: duplicate or reordered packet (seqnr 11170, expected 11173)
[... cut ...]
0:01:18.324114962  4064 0x7f5f5c012000 INFO               rtpsource rtpsource.c:1183:update_receiver_stats: duplicate or reordered packet (seqnr 12853, expected 12855)
0:01:19.450471649  4064 0x7f5f5c012000 INFO               rtpsource rtpsource.c:1183:update_receiver_stats: duplicate or reordered packet (seqnr 13116, expected 13120)
0:01:19.450744409  4064 0x7f5f5c012000 INFO               rtpsource rtpsource.c:1183:update_receiver_stats: duplicate or reordered packet (seqnr 13117, expected 13120)

jdelapla, 2022-07-10:
Hi, could you please provide some context to these logs? What's the environment, what sample are you running?

BADAL244, 2022-07-11:
Hello , I have one rtsp camera running on vlc and i can see the rtsp stream , but trying sending this stream over kinesis giving me this logs . I am running a cpp gstramer for rtsp on linux device .

disa6302, 2022-07-26:
Closing due to no response. Feel free to reopen when you have more details.

closed 2022-07-26 by disa6302