awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #876: [BUG]

rifkhananoor, 2022-07-06:
**Logging**
b0be8373643b:amazon-kinesis-video-streams-producer-sdk-cpp rifkhata$ pwd
/Users/rifkhata/amazon-kinesis-video-streams-producer-sdk-cpp
b0be8373643b:amazon-kinesis-video-streams-producer-sdk-cpp rifkhata$ printenv | grep JAVA
JAVA_AWT_INCLUDE_PATH=/Library/Java/JavaVirtualMachines/amazon-corretto-18.jdk/Contents/Home/include
JAVA_HOME=/Library/Java/JavaVirtualMachines/amazon-corretto-18.jdk/Contents/Home
JAVA_INCLUDE_PATH2=/Library/Java/JavaVirtualMachines/amazon-corretto-18.jdk/Contents/Home/include
JAVA_TOOL_OPTIONS=-Dlog4j2.formatMsgNoLookups=true
JAVA_INCLUDE_PATH=/Library/Java/JavaVirtualMachines/amazon-corretto-18.jdk/Contents/Home/include
b0be8373643b:amazon-kinesis-video-streams-producer-sdk-cpp rifkhata$ cmake  -DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_DEPENDENCIES=OFF
CMake Warning:
  No source or binary directory provided.  Both will be assumed to be the
  same as the current working directory, but note that this warning will
  become a fatal error in future CMake releases.

-- Kinesis Video Cpp Producer path is /Users/rifkhata/amazon-kinesis-video-streams-producer-sdk-cpp
-- dependencies install path is /Users/rifkhata/amazon-kinesis-video-streams-producer-sdk-cpp/open-source
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/rifkhata/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer
[ 11%] Performing update step for 'libkvscproducer-download'
[ 22%] No patch step for 'libkvscproducer-download'
[ 33%] No configure step for 'libkvscproducer-download'
[ 44%] No build step for 'libkvscproducer-download'
[ 55%] No install step for 'libkvscproducer-download'
[ 66%] No test step for 'libkvscproducer-download'
[ 77%] Completed 'libkvscproducer-download'
[100%] Built target libkvscproducer-download
-- Kinesis Video Producer path is /Users/rifkhata/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src
-- Checking for module 'libkvspicUtils'
--   No package 'libkvspicUtils' found
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/rifkhata/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic
[ 11%] Performing update step for 'libkvspic-download'
[ 22%] No patch step for 'libkvspic-download'
[ 33%] No configure step for 'libkvspic-download'
[ 44%] No build step for 'libkvspic-download'
[ 55%] No install step for 'libkvspic-download'
[ 66%] No test step for 'libkvspic-download'
[ 77%] Completed 'libkvspic-download'
[100%] Built target libkvspic-download
-- Bitness 64 bits
**found log4cplus.
CMake Error at /opt/homebrew/Cellar/cmake/3.23.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find JNI (missing: JAVA_INCLUDE_PATH JAVA_INCLUDE_PATH2
  JAVA_AWT_INCLUDE_PATH)
Call Stack (most recent call first):
  /opt/homebrew/Cellar/cmake/3.23.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  /opt/homebrew/Cellar/cmake/3.23.2/share/cmake/Modules/FindJNI.cmake:367 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:189 (find_package)**

-- Configuring incomplete, errors occurred!
See also "/Users/rifkhata/amazon-kinesis-video-streams-producer-sdk-cpp/CMakeFiles/CMakeOutput.log".
See also "/Users/rifkhata/amazon-kinesis-video-streams-producer-sdk-cpp/CMakeFiles/CMakeError.log".
b0be8373643b:amazon-kinesis-video-streams-producer-sdk-cpp rifkhata$

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. git clone --recursive  https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp.git
2. mkdir -p amazon-kinesis-video-streams-producer-sdk-cpp/build
3. brew install openssl
4. cmake  -DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_DEPENDENCIES=OFF

**Expected behavior**
compile successfully.

**SDK version number**
Release 3.3.1 of the Amazon Kinesis Video C++ Producer SDK

**Open source building**
brew install pkg-config openssl cmake gstreamer gst-plugins-base gst-plugins-good gst-plugins-bad gst-plugins-ugly log4cplus gst-libav 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Platform (please complete the following information):**
 - MacOS 12.3.1
 - M1
 - 

**Additional context**
Add any other context about the problem here.

jdelapla, 2022-07-06:
It appears your Cmake is trying to build the JNI, even though you are not explicitly stating so in your dependencies. This can happen if you did a cmake call earlier with the JNI dependency. Please delete the build directory, remake it, and try again with the Cmake command you gave,
```
cd ..
rm -rf build
mkdir -p build && cd build
cmake .. -DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_DEPENDENCIES=OFF
```

rifkhananoor, 2022-07-06:
Once i removed main folder and tried the steps .it worked

gst-inspect-1.0 kvssink
Factory Details:
  Rank                     primary + 10 (266)
  Long-name                KVS Sink
  Klass                    Sink/Video/Network
  Description              GStreamer AWS KVS plugin
  Author                   AWS KVS <kinesis-video-support@amazon.com>

Plugin Details:
  Name                     kvssink
  Description              GStreamer AWS KVS plugin
  Filename                 /Users/rifkhata/amazon-kinesis-video-streams-producer-sdk-cpp/build/libgstkvssink.so
  Version                  1.0
  License                  Proprietary
  Source module            kvssinkpackage
  Binary package           GStreamer
  Origin URL               http://gstreamer.net/

closed 2022-07-10 by jdelapla