awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #880: [QUESTION] How to pass AWS_KINESISVIDEO_IMAGE_PREFIX Tag in Gstreamer command

dharmik-dalwadi-seaflux, 2022-07-12:
## Scenario
- I am currently sending one RTSP Stream to Kinesis Video Stream using the Gstreamer command, Now my aim is to implement the [Automated Image Generation (S3 Delivery)](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/gs-s3Delivery.html) API to generate and store images in s3. But I don't know how can I add **AWS_KINESISVIDEO_IMAGE_PREFIX** tag in the Gstreamer command to start producing images as given in [Producer MKV Tags](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/gs-s3Delivery.html#puteventmetadata). 
- Is there any way to add **Tags** in Gstreamer command? 
- I am using the below-given command to Send the RTSP stream to KVS.

## Command
`gst-launch-1.0 -v rtspsrc location=<rtsp-url> ! rtph264depay ! h264parse ! kvssink name=sink stream-name=<stream-name> access-key=<access-key> secret-key=<secret-key> aws-region=<region>`

Thanks.

hassanctech, 2022-07-12:
A follow up question I have is do you want the ability to be able to start and stop image generation or are you only interested in the feature where you specify image generation at the start of the pipeline and then you keep generating images until you stop the pipeline.

Today we do not expose this feature via the `kvssink` plugin but I can see a few different ways we can do this.  

1. We can expose an option in `kvssink` to control yes/no if you want image generation.  In this case when the pipeline starts so will image generation and there will be no way to turn it off you would need to stop the pipeline.
2. We can expose a callback say every keyframe and at that point you would have the opportunity to specify such tags.

Clearly (2) is more robust but (1) is trivial to implement so I'm curious if that meets your needs?

dharmik-dalwadi-seaflux, 2022-07-12:
@hassanctech  Right now we are currently testing the service, so (1) would give us the ability to do so in a basic way, and we would be really happy with it.

About (2) Currently the default samplingInterval of s3 delivery is 3 seconds, so i'm not sure, even if you put the metadata tag on it, that you will be able to extract more than 1 frame every 3 seconds (or maybe i'm missing something about the samplingInterval)

Our current business need is to have 5fps frames on S3 buckets, we are testing s3 delivery with kvs video stream to test how feasible it is, if you have more idea on how to deal with it, please share you thoughts.