awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #882: [QUESTION] Getting error running cmake command

dharshinithilagar, 2022-07-26:


disa6302, 2022-07-27:
Please follow the steps in the CI
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/70f74f14cf27b09f71dc1889f36eb6e04cdd90a8/.github/workflows/ci.yml#L271

Script: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/.github/build_windows.bat

weberdla, 2022-08-17:
Following the above steps fails with cmake...

Build on MS VS fails with LINK : fatal error LNK1181: cannot open input file 'apps\app_rand.obj'

niyatim23, 2022-08-17:
Our CI is able to successfully build the SDK with the steps shared so this is not an issue with the SDK. Please make sure you're trying with the correct version of MS VS. Try a fresh build with these steps

weberdla, 2022-08-17:
Ran fresh build and encountered same problem.  Re-ran same build as indicated in #715 and then it works.