awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #885: [BUG] gst-inspect-1.0.exe kvssink crashes on Windows 10

bmackindtiq, 2022-07-27:
**Describe the bug**
I built master on Windows 10, with gstreamer 1.16.2 using the following cmake options:

cmake -G "NMake Makefiles" -DBUILD_STATIC=TRUE -DCMAKE_BUILD_TYPE=Release -DCMAKE_CONFIGURATION_TYPES=Release -DBUILD_TEST=FALSE -DBUILD_GSTREAMER_PLUGIN=TRUE -DCMAKE_VERBOSE_MAKEFILE:BOOL=ON -DPKG_CONFIG_EXECUTABLE=C:/gstreamer/1.0/x86_64/bin/pkg-config.exe .
nmake

gstvkssink.dll builds successfully

I loaded this into my gstreamer installation into the directory specified by GST_PLUGIN_PATH.
I also put the log4cplusD.dll produced by the build into amazon-kinesis-video-streams-producer-sdk-cpp\open-source\local\bin

The following call crashes (it works for all other plugins I tried)

gst-inspect-1.0.exe kvssink 

I set my system go generate a dump on crash. Following is the stack trace. You can see it crashes in log4cplus.
I couldn't find isBigEndian() in and of the source, so I was stuck there

**Log**

Last gst log:
0:00:00.152928000 29412 000001AF8A1D7000 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:361:gst_element_factory_create: creating element "kvssink"

**Stack Trace**
0:000> !analyze -v
*******************************************************************************
*                                                                             *
*                        Exception Analysis                                   *
*                                                                             *
*******************************************************************************

KEY_VALUES_STRING: 1

    Key  : Analysis.CPU.mSec
    Value: 577

    Key  : Analysis.DebugAnalysisManager
    Value: Create

    Key  : Analysis.Elapsed.mSec
    Value: 581

    Key  : Analysis.Init.CPU.mSec
    Value: 311

    Key  : Analysis.Init.Elapsed.mSec
    Value: 145125

    Key  : Analysis.Memory.CommitPeak.Mb
    Value: 89

    Key  : Timeline.OS.Boot.DeltaSec
    Value: 106910

    Key  : Timeline.Process.Start.DeltaSec
    Value: 1

    Key  : WER.OS.Branch
    Value: vb_release

    Key  : WER.OS.Timestamp
    Value: 2019-12-06T14:06:00Z

    Key  : WER.OS.Version
    Value: 10.0.19041.1

FILE_IN_CAB:  gst-inspect-1.0.exe_220727_130707.dmp

COMMENT:  
*** "c:\Users\bmackin\SysinternalsSuite\procdump.exe" -accepteula -ma -j "c:\dumps" 28588 520 00000228BD290000
[... 74 lines cut ...]
FAULTING_SOURCE_LINE:  minkernel\crts\ucrt\src\appcrt\heap\free_base.cpp

FAULTING_SOURCE_FILE:  minkernel\crts\ucrt\src\appcrt\heap\free_base.cpp

FAULTING_SOURCE_LINE_NUMBER:  105

FAULTING_SOURCE_CODE:  
No source found for 'minkernel\crts\ucrt\src\appcrt\heap\free_base.cpp'

**Describe the bug**
See above

**To Reproduce**
run 
gst-inspect-1.0.exe kvssink 

There is no output, as it crashes before printing output

**Platform (please complete the following information):**
 - OS: WIndows 10 Pro
 - Processor: i7-11800H @2.30GHz
 - Version: 21H2

**Software version**
commit 70f74f14cf27b09f71dc1889f36eb6e04cdd90a8 (HEAD -> master, tag: v3.3.1, origin/master, origin/HEAD)

bmackindtiq, 2022-07-28:
It crashed building and running with the latest gstreamer (1.20.3). Trace wasn't identical, but again it crashed in log4cplus:

STACK_TEXT:  
00000059`621de940 00007ffd`99daf213     : 00007ffd`99df6630 00000059`621dfb40 00000000`0019001b 00000000`0000002c : ntdll!RtlReportFatalFailure+0x9
00000059`621de990 00007ffd`99db7f92     : 000001d5`4cd340d0 00007ffd`99e187f0 00000000`00000003 000001d5`4cc40000 : ntdll!RtlReportCriticalFailure+0x97
00000059`621dea80 00007ffd`99db827a     : 00000000`00000003 00000000`00000000 000001d5`4cc40000 00007ffd`99d0e19b : ntdll!RtlpHeapHandleError+0x12
00000059`621deab0 00007ffd`99dbdf01     : 000001d5`4cc40000 000001d5`4cc40000 000001d5`4f0269a0 00000000`00000000 : ntdll!RtlpHpHeapHandleError+0x7a
00000059`621deae0 00007ffd`99d57142     : 00000059`621dee40 000001d5`4cc40000 00000059`621df499 00000000`00000000 : ntdll!RtlpLogHeapFailure+0x45
00000059`621deb10 00007ffd`99cd47b1     : 00000059`621def00 000001d5`4cc40000 00000059`621df499 00000000`00000000 : ntdll!RtlpFreeHeapInternal+0x81a32
00000059`621debd0 00007ffc`eb3109ed     : 000001d5`4f1d5e80 000001d5`4f1fb0a0 00000000`00000008 00007ffc`00000000 : ntdll!RtlFreeHeap+0x51
00000059`621dec10 00007ffc`eb30d07a     : 000001d5`4f1d5ed0 000001d5`4f1d5ffd 00000000`00000004 00007ffc`00000001 : ucrtbased!_free_base+0x2d
00000059`621dec50 00007ffc`eb310275     : 000001d5`4f1d5f00 00007ffc`00000001 00000059`621dee10 00007ffd`3bc813f4 : ucrtbased!free_dbg_nolock+0x72a
00000059`621ded50 00007ffc`ebe06df8     : 000001d5`4f1d5f00 00007ffd`ffffffff 00007ffd`3bcf5258 000001d5`4f1fb0a0 : ucrtbased!_free_dbg+0x55
00000059`621ded90 00007ffc`ebe054b8     : 000001d5`4f1d5f00 00007ffc`ebd88af4 00000059`621dee10 000001d5`4f1d5e80 : log4cplusD!log4cplus::thread::FairMutex::FairMutex+0x9174e
00000059`621dedc0 00007ffc`ebd7f5b8     : 000001d5`4f1d5f00 00000000`00000010 cccccccc`cccccccc cccccccc`cccccccc : log4cplusD!log4cplus::thread::FairMutex::FairMutex+0x8fe0e
00000059`621dedf0 00007ffc`ebd8a327     : 000001d5`4f1d5f00 00000000`00000010 00007ffd`3bcf5258 000001d5`4f1d5e80 : log4cplusD!log4cplus::thread::FairMutex::FairMutex+0x9f0e
00000059`621dee20 00007ffc`ebd7f5f4     : 00000059`621deee4 000001d5`4f1d5f00 00000000`00000001 000001d5`4f1d5e80 : log4cplusD!log4cplus::thread::FairMutex::FairMutex+0x14c7d
00000059`621dee50 00007ffc`ebd7f62e     : 00000059`621deee4 000001d5`4f1d5f00 000001d5`4f1d5f00 cccccccc`cccccccc : log4cplusD!log4cplus::thread::FairMutex::FairMutex+0x9f4a
00000059`621dee80 00007ffc`ebda0e33     : 00000059`621deee4 000001d5`4f1d5f00 00000059`621df100 00007ffc`ebda139e : log4cplusD!log4cplus::thread::FairMutex::FairMutex+0x9f84
00000059`621deeb0 00007ffc`ebd99d59     : 00000059`621df008 00000059`621df0b8 00000000`00000050 00007ffc`00000001 : log4cplusD!log4cplus::thread::FairMutex::FairMutex+0x2b789
00000059`621def10 00007ffc`ebd9993c     : 00000059`621df370 cccccccc`cccccccc 00000059`621df410 00007ffc`ebd9fe45 : log4cplusD!log4cplus::thread::FairMutex::FairMutex+0x246af
00000059`621df110 00007ffc`ebd99475     : 00000059`621df370 00000059`621df200 cccccccc`00000000 cccccccc`cccccccc : log4cplusD!log4cplus::thread::FairMutex::FairMutex+0x24292
00000059`621df1b0 00007ffc`ebd9ad78     : 00000059`621df370 00000059`621df200 000001d5`4f0c7040 000001d5`00000000 : log4cplusD!log4cplus::thread::FairMutex::FairMutex+0x23dcb
00000059`621df1e0 00007ffc`ebd9afa7     : 00000059`621df370 000001d5`4f0c7040 00000059`621df400 00007ffc`ebd8b777 : log4cplusD!log4cplus::thread::FairMutex::FairMutex+0x256ce
00000059`621df340 00007ffc`f771aa5f     : 000001d5`4f0c7040 00000059`621df400 000001d5`4cc7a500 00007ffc`00000001 : log4cplusD!log4cplus::thread::FairMutex::FairMutex+0x258fd
00000059`621df420 00007ffd`23a82792     : 00000000`00000003 00000000`00000000 000001d5`4f1fb0a0 00000000`ffffffff : gstkvssink!kinesisVideoStreamDefaultStreamShutdown+0x3b8f
00000059`621df500 00007ffd`23a6ee3a     : 000001d5`4f046b70 00000000`00000000 00000000`00000000 00007ffd`23a856a3 : gobject_2_0_0!g_type_create_instance+0x152
00000059`621df540 00007ffd`23a6fe01     : 00000000`00000000 000001d5`4f0269a0 000001d5`4f0269a0 00000000`00000000 : gobject_2_0_0!g_object_new+0xaa
00000059`621df5f0 00007ffd`227817c4     : 000001d5`4f0269a0 00000000`00000000 000001d5`4f0269a0 00007ffd`2274e548 : gobject_2_0_0!g_object_new_with_properties+0x231
00000059`621df680 00007ff7`2bb72b71     : 00000000`00000000 00000059`621df880 000001d5`4cc5a9cc 00000000`00000000 : gstreamer_1_0_0!gst_element_factory_create_with_properties+0x114
00000059`621df6d0 00007ff7`2bb71cbd     : 00000000`00000000 000001d5`4cc5a9cc 00000059`621df880 00000000`00000000 : gst_inspect_1_0+0x2b71
00000059`621df780 00007ff7`2bb77f64     : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : gst_inspect_1_0+0x1cbd
00000059`621dfad0 00007ffd`99467034     : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : gst_inspect_1_0+0x7f64
00000059`621dfb10 00007ffd`99d02651     : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : kernel32!BaseThreadInitThunk+0x14
00000059`621dfb40 00000000`00000000     : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : ntdll!RtlUserThreadStart+0x21

bmackindtiq, 2022-08-05:
From my investigation, it appears that this is due to log4cplus issues when built with Microsoft's toolchain.  See "Windows and Visual Studio" here: https://github.com/log4cplus/log4cplus. Log4cplus needs to be initialized in main() when using Microsoft's toolchain. No gstreamer executable does this, as log4cplus isn't used anywhere in gstreamer. I did a test using a simple mock log4cplus I created that does nothing in the interfaces that are actually used by the sdk or plugin, and it does not crash anymore when running "gst-inspect-1.0.exe kvssink". Unfortunately, it eliminates all logging.

hassanctech, 2022-08-08:
I see yeah I just looked at the link you shared.  Do you have any ideas for a way forward, I mean since the GStreamer elements themselves do not have a main() it seems like this would prevent any element from using log4cplus which seems unlikely I'll look to see if there is really no way for Windows.

bmackindtiq, 2022-08-09:
This is a good question. The only ideas I have aren't so well thought out, and may require too much effort. 
1) Alternate c++ versions of the gstreamer exe's (gst-launch-1.0, gst-inspect-1.0) to use with your kinesis plugin. Probably don't even need to be full featured
2) Wrap your log4cplus calls in a lightweight wrapper. You don't seem to use every feature of it. Offer a build option where you can choose whether to use log4cplus or something simpler with fewer features (maybe even internally just console logging eg cout <<). I understand you can't just switch to gstreamer's logging, as your producer SDK doesn't use glib, and someone using 

I'd be concerned about licensing issues if you did #1, as well as the "forever" technical debt. #2 doesn't seem so bad. If log4cplus ever fixes the issue, you can even abandon it

niyatim23, 2023-09-05:
Hi @bmackindtiq, please refer to the steps of a working Windows build in our CI:
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/4c4afff63f3371a69be2472bb0bee654ef09abd3/.github/workflows/ci.yml#L278-L297

Be sure to modify the folder paths as per the location on your system. Additionally, you may also have to modify the bat file as per MSVC's path on your system and export the paths as follows:
```
set GST_PLUGIN_PATH=%CD%\build
set PATH=%PATH%;%CD%\open-source\local\bin;%CD%\open-source\local\lib
```

I have been able to run kvssink on Windows with no trouble:
```
C:\Users\Administrator\Desktop\cpp\amazon-kinesis-video-streams-producer-sdk-cpp\build>gst-inspect-1.0 kvssink
Factory Details:
  Rank                     primary + 10 (266)
  Long-name                KVS Sink
  Klass                    Sink/Video/Network
  Description              GStreamer AWS KVS plugin
  Author                   AWS KVS <kinesis-video-support@amazon.com>

Plugin Details:
  Name                     kvssink
  Description              GStreamer AWS KVS plugin
  Filename                 C:\Users\Administrator\Desktop\cpp\amazon-kinesis-video-streams-producer-sdk-cpp\build\gstkvssink.dll
  Version                  1.0
  
```

Gstreamer pipeline:

```
gst-launch-1.0 -v videotestsrc is-live=true ! video/x-raw,framerate=30/1,width=720,height=480 ! clockoverlay time-format="%a %B %d, %Y %I:%M:%S %p" ! openh264enc ! h264parse ! kvssink stream-name=demo-stream
```

niyatim23, 2023-09-11:
Closing assuming answered

closed 2023-09-11 by niyatim23