awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #886: [QUESTION] kvssrc

clogwog, 2022-07-27:
Hello,

We've been using kvssink to it's fullest and are getting a nice collection of streams on kinesis video.

We now want to start using the video either on the server or on an application and I was wondering how people get the streams back from kinesis video. Is there a matching kvssrc to kvssink ? If not how are people actually consuming the streams ?

I noticed there is a way to get the media https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/API_reader_GetMediaForFragmentList.html but those are blobs of data.. how do people use that to start playing/streaming the video ?

Kind regards,
Tom

disa6302, 2022-07-27:
We do not have gstreamer kvssrc plugin. However, customers use the parser library for consumption: https://github.com/aws/amazon-kinesis-video-streams-parser-library

MushMal, 2022-07-27:
The real-time path is using GetMedia API https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/API_dataplane_GetMedia.html

The stream then can be parsed by the library that @disa6302 mentioned

clogwog, 2022-07-27:
The parser library seems to only support MKV media ? Is that correct ? We are streaming H264 is that still supported ?
And it is a Java only library ?

MushMal, 2022-07-27:
H264 is the encoding. MKV is the wrapper. The parser library allows to "unwrap" the MKV on-the-fly. What you end up with is H264 elementary stream which you would still need to decode in order to get to the decoded frame. You can use any MKV decoders for that matter - the parser library is Java only. There are a few available from open source, including the CPP reference implementation from MKV site.

clogwog, 2022-07-28:
@MushMal Ah... so even though I stream it as 

... ! x264enc ! video/x-h264,stream-format=avc,alignment=au,profile=baseline ! kvssink ... 

it gets sent to kinesis video with an MKV wrapper inside kvssink (or it gets added when requesting the GetMedia )?

hassanctech, 2022-07-28:
Our SDK packages it as MKV before sending it to the backend, from your point of view using `kvssink` yeah it happens in there.

clogwog, 2022-07-28:
thank you all !! very enlightening !

closed 2022-07-28 by clogwog