awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #887: [BUG]

Coldplayer1995, 2022-08-02:
**Logging**
gst-launch-1.0 v4l2src do-timestamp=TRUE device=/dev/video0 ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=30/1 ! omxh264enc control-rate=1 target-bitrate=1024000 periodicity-idr=30 inline-header=FALSE ! h264parse ! video/x-h264,stream-format=avc,alignment=au,width=640,height=480,framerate=30/1,profile=baseline ! kvssink stream-name="2230000001" iot-certificate="iot-certificate,endpoint=cl5w2mxpp69fr.credentials.iot.us-east-2.amazonaws.com,cert-path=/home/pi/device.pem,key-path=/home/pi/device.key,ca-path=/home/pi/cacert.pem,role-aliases=KvsCameraIoTRoleAlias, iot-thing-name=2230000001" aws-region="us-east-2"

**Describe the bug**
A clear and concise description of what the bug is.
ERROR - Value should be of "String" type for iot-thing-name

**To Reproduce**
Steps to reproduce the behavior:
Try to run above command with iot-thing-name as a number

Coldplayer1995, 2022-08-23:
@disa6302  if you try to create a device 111111 in IoT core that will be created without a problem, you should either case that INT into the string or let the numbers there.

disa6302, 2022-08-23:
Have you tried adding it in double quotes? From the line you pasted, it does not seem like your iot thing name is in double quotes

Coldplayer1995, 2022-08-23:
We tried singular, double nothing works because of that check which you have  G_VALUE_HOLDS_STRING this functions checks very well

disa6302, 2022-08-23:
Ok. I see what you mean. Let me run some checks and verify and create a PR. :) Thank you.

disa6302, 2022-09-19:
@Coldplayer1995 , 

I missed suggesting this, but can you try type casting the thing name with (string) in the command line? That should fix your issue. Like this:

gst-launch-1.0 v4l2src do-timestamp=TRUE device=/dev/video0 ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=30/1 ! omxh264enc control-rate=1 target-bitrate=1024000 periodicity-idr=30 inline-header=FALSE ! h264parse ! video/x-h264,stream-format=avc,alignment=au,width=640,height=480,framerate=30/1,profile=baseline ! kvssink stream-name="2230000001" iot-certificate="iot-certificate,endpoint=secret-endpointiot.us-east-2.amazonaws.com,cert-path=/home/pi/device.pem,key-path=/home/pi/device.key,ca-path=/home/pi/cacert.pem,role-aliases=KvsCameraIoTRoleAlias, iot-thing-name=_(string)_ 2230000001" aws-region="us-east-2"

disa6302, 2022-09-22:
@Coldplayer1995 , 

I tested the pipeline by typecasting the value with (string) and that works. I am closing this assuming answered. Please feel free to reach out if this does not help you.