awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #889: AWS Kinesis Video Stream handle 0

Redtigercod4, 2022-08-04:
I am trying to create a stream from a Raspberry Pi 4 with a camera module and have successfully got the connection between the IoT device and my Video Stream. Currently GStreamer using kvssink, is writing the data but the upload stream handle sits at 0.

Output Log:
```
DEBUG - postReadCallback(): Wrote 1117 bytes to Kinesis Video. Upload stream handle: 0
2022-08-03 14:34:18 [2311189568] DEBUG - postReadCallback(): Wrote 1084 bytes to Kinesis Video. Upload stream handle: 0
2022-08-03 14:34:18 [2311189568] DEBUG - postReadCallback(): Wrote 926 bytes to Kinesis Video. Upload stream handle: 0
DEBUG - postReadCallback(): Wrote 2286 bytes to Kinesis Video. Upload stream handle: 0
2022-08-03 14:34:18 [2311189568] DEBUG - postReadCallback(): Wrote 1570 bytes to Kinesis Video. Upload stream handle: 0
2022-08-03 14:34:18 [2311189568] DEBUG - postReadCallback(): Wrote 1045 bytes to Kinesis Video. Upload stream handle: 0
DEBUG - postReadCallback(): Wrote 1272 bytes to Kinesis Video. Upload stream handle: 0
2022-08-03 14:34:18 [2311189568] DEBUG - postReadCallback(): Wrote 3246 bytes to Kinesis Video. Upload stream handle: 0
DEBUG - postReadCallback(): Wrote 1292 bytes to Kinesis Video. Upload stream handle: 0
2022-08-03 14:34:18 [2311189568] DEBUG - postReadCallback(): Wrote 1653 bytes to Kinesis Video. Upload stream handle: 0
2022-08-03 14:34:18 [2311189568] DEBUG - postReadCallback(): Wrote 1549 bytes to Kinesis Video. Upload stream handle: 0
DEBUG - postReadCallback(): Wrote 3160 bytes to Kinesis Video. Upload stream handle: 0
DEBUG - postReadCallback(): Wrote 2039 bytes to Kinesis Video. Upload stream handle: 0
2022-08-03 14:34:18 [2311189568] DEBUG - postReadCallback(): Wrote 2520 bytes to Kinesis Video. Upload stream handle: 0
```

MushMal, 2022-08-04:
Upload handle is an obscure handle that's used by the producer SDK layers to indicate the session. If you keep it running and when the stream connection gets interrupted (due to connection error or on the session rotation) you will see the upload handle value change.

Redtigercod4, 2022-08-07:
Thank you, I have managed to solve the issue being down to my network not allowing the data to be streamed. Checked on a different network and it worked perfectly.

closed 2022-08-07 by Redtigercod4