awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #890: Hi @niyatim23, Is this issue fixed?

ajith-karkera, 2022-08-08:
Hi @niyatim23, Is this issue fixed?
I am still facing issue with 1.20 version when running the below pipeline: 

gst-launch-1.0 --gst-debug-level=3 rtspsrc location=rtsp://admin:Password@10.10.8.51:8554/recorded/4784640?start_time=2022-07-25T00:00:00Z ! rtph264depay ! h264parse ! decodebin ! kvssink stream-name=demo storage-size=512

_Originally posted by @ajith-karkera in https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/800#issuecomment-1206146333_

ajith-karkera, 2022-08-08:
I tried doing URL encoding of query params  in pipeline like below, still it fails.
gst-launch-1.0 --gst-debug-level=3  rtspsrc location=rtsp://admin:Password@10.10.8.51:8554/recorded/4784640%3Fstart_time=2022-08-07T01:00:00Z ! rtph264depay ! h264parse ! kvssink stream-name=demo storage-size=512
Use Windows high-resolution clock, precision: 1 ms
Setting pipeline to PAUSED ...
log4cplus:ERROR could not open file ./kvs_log_configuration
INFO - createKinesisVideoClient(): Creating Kinesis Video Client
2022-08-07 23:52:12 [7984] INFO - heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
2022-08-07 23:52:12 [7984] INFO - heapInitialize(): Creating AIV heap.
2022-08-07 23:52:12 [7984] INFO - heapInitialize(): Heap is initialized OK
2022-08-07 23:52:12 [7984] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
2022-08-07 23:52:12 [7984] DEBUG - getSecurityTokenHandler invoked
2022-08-07 23:52:12 [7984] DEBUG - Refreshing credentials. Force refreshing: 0 Now time is: 16599307325842789 Expiration: 0
2022-08-07 23:52:12 [7984] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
2022-08-07 23:52:12 [7984] INFO - createDeviceResultEvent(): Create device result event.
2022-08-07 23:52:12 [7984] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
2022-08-07 23:52:12 [7984] DEBUG - clientReadyHandler invoked
2022-08-07 23:52:12 [7984] INFO - try creating stream
2022-08-07 23:52:12 [7984] INFO - Creating Kinesis Video Stream Demo123
2022-08-07 23:52:12 [7984] INFO - createKinesisVideoStream(): Creating Kinesis Video Stream.
2022-08-07 23:52:12 [7984] DEBUG - logStreamInfo(): Kinesis Video Stream Info
2022-08-07 23:52:12 [7984] DEBUG - logStreamInfo():     Stream name: Demo123
2022-08-07 23:52:12 [7984] DEBUG - logStreamInfo():     Streaming type: STREAMING_TYPE_REALTIME
2022-08-07 23:52:12 [7984] DEBUG - logStreamInfo():     Content type: video/h264
2022-08-07 23:52:12 [7984] DEBUG - logStreamInfo():     Max latency (100ns): 20000000
2022-08-07 23:52:12 [7984] DEBUG - logStreamInfo():     Fragment duration (100ns): 20000000
2022-08-07 23:52:12 [7984] DEBUG - logStreamInfo():     Key frame fragmentation: Yes
2022-08-07 23:52:12 [7984] DEBUG - logStreamInfo():     Use frame timecode: No
2022-08-07 23:52:12 [7984] DEBUG - logStreamInfo():     Absolute frame timecode: Yes
2022-08-07 23:52:12 [7984] DEBUG - logStreamInfo():     Nal adaptation flags: 0
2022-08-07 23:52:12 [7984] DEBUG - logStreamInfo():     Average bandwith (bps): 4194304
2022-08-07 23:52:12 [7984] DEBUG - logStreamInfo():     Framerate: 25
2022-08-07 23:52:12 [7984] DEBUG - logStreamInfo():     Buffer duration (100ns): 1200000000
2022-08-07 23:52:12 [7984] DEBUG - logStreamInfo():     Replay duration (100ns): 400000000
2022-08-07 23:52:12 [7984] DEBUG - logStreamInfo():     Connection Staleness duration (100ns): 600000000
2022-08-07 23:52:12 [7984] DEBUG - logStreamInfo():     Store Pressure Policy: 1
2022-08-07 23:52:12 [7984] DEBUG - logStreamInfo():     View Overflow Policy: 1
2022-08-07 23:52:12 [7984] DEBUG - logStreamInfo():     Segment UUID: NULL
2022-08-07 23:52:12 [7984] DEBUG - logStreamInfo():     Frame ordering mode: 0
2022-08-07 23:52:12 [7984] DEBUG - logStreamInfo(): Track list
2022-08-07 23:52:12 [7984] DEBUG - logStreamInfo():     Track id: 1
2022-08-07 23:52:12 [7984] DEBUG - logStreamInfo():     Track name: kinesis_video
2022-08-07 23:52:12 [7984] DEBUG - logStreamInfo():     Codec id: V_MPEG4/ISO/AVC
2022-08-07 23:52:12 [7984] DEBUG - logStreamInfo():     Track type: TRACK_INFO_TYPE_VIDEO
2022-08-07 23:52:12 [7984] DEBUG - logStreamInfo():     Track cpd: NULL
2022-08-07 23:52:12 [7984] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
2022-08-07 23:52:12 [7984] INFO - writeHeaderCallback(): RequestId: 9d8420a3-90a7-4725-9c51-6d3a1027b9a7
2022-08-07 23:52:13 [2668] DEBUG - describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.659707389786E9,"DataRetentionInHours":48,"DeviceName":"Kinesis_Video_Device","IngestionConfiguration":null,"KmsKeyId":"arn:aws:kms:us-east-1:764601613376:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-east-1:764601613376:stream/Demo123/1659707389786","StreamName":"Demo123","Version":"CncwQsWmnBP7HLaKun7I"}}
2022-08-07 23:52:13 [2668] INFO - describeStreamResultEvent(): Describe stream result event.
2022-08-07 23:52:13 [2668] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000020
2022-08-07 23:52:13 [2668] INFO - writeHeaderCallback(): RequestId: 906d1155-c1ee-411b-8458-274fde76e9c7
2022-08-07 23:52:13 [4888] DEBUG - getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-808fdb44.kinesisvideo.us-east-1.amazonaws.com"}
2022-08-07 23:52:13 [4888] INFO - getStreamingEndpointResultEvent(): Get streaming endpoint result event.
2022-08-07 23:52:13 [4888] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000020, Next state: 0x0000000000000010
2022-08-07 23:52:13 [4888] DEBUG - getStreamingTokenHandler invoked
[... cut ...]
Progress: (request) Sending PLAY request
Progress: (request) Sending PLAY request
Progress: (request) Sent PLAY request
Redistribute latency...
0:00:01.702061000  1624 000002C845403660 WARN                     bin gstbin.c:2764:gst_bin_do_latency_func:<pipeline0> did not really configure latency of 0:00:00.000000000
0:00:27.507981000  1624 000002C84552B0C0 WARN                 rtspsrc gstrtspsrc.c:3639:on_timeout_common:<rtspsrc0> source ce39a2f3, stream 00000000 in session 0 timed out
0:00:00.0 / 99:99:99.

hassanctech, 2022-08-08:
Can you please try with the `rtspsrc` and `kvssink` pipeline we have in our docs:

`gst-launch-1.0 -v rtspsrc location=rtsp://YourCameraRtspUrl short-header=TRUE ! rtph264depay ! h264parse ! kvssink stream-name=YourStreamName storage-size=128`

If it doesn't work please post logs attached as a separate file.  Are you specifying any additional parameters in your `rtspsrc` ?  Make sure to set `short-header` to `TRUE` but if you're setting anything else please remove it and try.

ajith-karkera, 2022-08-09:
Hi @hassanctech , 

Thanks for the reply, I tried the pipline that you have shared.

One with the RTSP URL as it is:
gst-launch-1.0 -v rtspsrc location=rtsp://admin:Password@10.10.8.51:8554/recorded/4784640?start_time=2022-08-03T11:20:00Z ! rtph264depay ! h264parse ! kvssink stream-name=Demo storage-size=128

One with RTSP URL encoded:
gst-launch-1.0 -v rtspsrc location=rtsp://admin:Password@10.10.8.51:8554/recorded/4784640%3Fstart_time=2022-08-03T11:20:00Z ! rtph264depay ! h264parse ! kvssink stream-name=Demo storage-size=128

Please find attached logs for the same.

[logs_normal_url.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/9286863/logs_normal_url.txt)

[logs_encoded_url.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/9286862/logs_encoded_url.txt)

ajith-karkera, 2022-08-09:
Hi @hassanctech,

I have also tried the pipline that you have shared with the live RTSP URL as below, It works like charm, the issue is with the RTSP URL of recording which as query string.

gst-launch-1.0 -v rtspsrc location=rtsp://admin:Password@10.10.8.51:8554/4784640 ! rtph264depay ! h264parse ! kvssink stream-name=Demo storage-size=128

Please find attached logs for the same.

[logs_without_query_params.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/9287395/logs_without_query_params.txt)

ajith-karkera, 2022-08-10:
Hi @hassanctech , Any update on the issue?

ajith-karkera, 2022-08-10:
This issue is fixed,  

We need to add a / at the end of RTSP URL.

gst-launch-1.0 -v rtspsrc location=rtsp://admin:Password@10.10.8.51:8554/recorded/4784640%3Fstart_time=2022-08-03T11:20:00Z/  ! rtph264depay ! h264parse ! kvssink stream-name=Demo storage-size=128

closed 2022-08-10 by ajith-karkera