awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #891: Installation on macbok pro (cpu: M1 pro) fails with : C

j0928888843, 2022-08-09:
A one liner description about the use case and what you are trying to achieve

** Logging **
```
cmake -DBUILD_GSTREAMER_PLUGIN=TRUE ..
-- Kinesis Video Cpp Producer path is /Users/yunches/test/amazon-kinesis-video-streams-producer-sdk-cpp
-- dependencies install path is /Users/yunches/test/amazon-kinesis-video-streams-producer-sdk-cpp/open-source
-- Begin building dependencies.
-- autoconf already built
-- automake already built
-- log4cplus already built
-- Finished building dependencies.
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/yunches/test/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer
[ 11%] Performing update step for 'libkvscproducer-download'
[ 22%] No patch step for 'libkvscproducer-download'
[ 33%] No configure step for 'libkvscproducer-download'
[ 44%] No build step for 'libkvscproducer-download'
[ 55%] No install step for 'libkvscproducer-download'
[ 66%] No test step for 'libkvscproducer-download'
[ 77%] Completed 'libkvscproducer-download'
[100%] Built target libkvscproducer-download
-- Kinesis Video Producer path is /Users/yunches/test/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src
-- Begin building dependencies.
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/yunches/test/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/libopenssl
[ 11%] Creating directories for 'project_libopenssl'
[ 22%] Performing download step (git clone) for 'project_libopenssl'
Cloning into 'project_libopenssl'...
HEAD is now at e2e09d9fba Prepare for 1.1.1g release
Submodule 'boringssl' (https://boringssl.googlesource.com/boringssl) registered for path 'boringssl'
Submodule 'krb5' (https://github.com/krb5/krb5) registered for path 'krb5'
Submodule 'pyca.cryptography' (https://github.com/pyca/cryptography.git) registered for path 'pyca-cryptography'
Cloning into '/Users/yunches/test/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/libopenssl/build/src/project_libopenssl/boringssl'...
Cloning into '/Users/yunches/test/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/libopenssl/build/src/project_libopenssl/krb5'...
Cloning into '/Users/yunches/test/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/libopenssl/build/src/project_libopenssl/pyca-cryptography'...
Submodule path 'boringssl': checked out '2070f8ad9151dc8f3a73bffaa146b5e6937a583f'
Submodule path 'krb5': checked out '890ca2f401924cdcb88f493950b04700bbe52db3'
Submodule path 'pyca-cryptography': checked out '09403100de2f6f1cdd0d484dcb8e620f1c335c8f'
[ 33%] Performing update step for 'project_libopenssl'
[ 44%] No patch step for 'project_libopenssl'
[ 55%] Performing configure step for 'project_libopenssl'
Operating system: i686-apple-darwinDarwin Kernel Version 21.5.0: Tue Apr 26 21:08:37 PDT 2022; root:xnu-8020.121.3~4/RELEASE_ARM64_T6000
Configuring OpenSSL version 1.1.1g (0x1010107fL) for darwin-i386-cc
Using os-specific seed configuration
Creating configdata.pm
Creating Makefile

**********************************************************************
***                                                                ***
***   OpenSSL has been successfully configured                     ***
***                                                                ***
***   If you encounter a problem while building, please open an    ***
***   issue on GitHub <https://github.com/openssl/openssl/issues>  ***
***   and include the output from the following command:           ***
***                                                                ***
***       perl configdata.pm --dump                                ***
***                                                                ***
***   (If you are new to OpenSSL, you might want to consult the    ***
***   'Troubleshooting' section in the INSTALL file first)         ***
***                                                                ***
**********************************************************************
[ 66%] Performing build step for 'project_libopenssl'
ar: creating archive apps/libapps.a
ar: creating archive libcrypto.a
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: libcrypto.a(async_null.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: libcrypto.a(async_win.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: libcrypto.a(cms_cd.o) has no symbols
[... 653 lines cut ...]
======= CMakeError.log ======

Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: /Library/Developer/CommandLineTools/usr/bin/cc 
Build flags: 
Id flags:  

The output was:
1
ld: library not found for -lSystem
clang: error: linker command failed with exit code 1 (use -v to see invocation)

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: /Library/Developer/CommandLineTools/usr/bin/c++ 
Build flags: 
Id flags:  

The output was:
1
ld: library not found for -lc++
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```
platform: M1 pro macbook pro
The C compiler identification is AppleClang 13.1.6.13160021
The CXX compiler identification is AppleClang 13.1.6.13160021

closed 2022-08-18 by j0928888843

ThePhen, 2022-08-29:
I faced this exact problem recently. 

To get around it, I followed the [clues in the OSX build instructions](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/docs/macos.md#open-source-dependencies) and installed all of the needed dependencies via HomeBrew, the used the CMake flag to prevent the build from trying to assemble the dependencies itself.

Here's some history from my shell:
```sh
git clone https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp.git\
cd amazon-kinesis-video-streams-producer-sdk-cpp
mkdir build
cd build
KVS_GST_PLUGIN_PATH=$(pwd)
brew install pkg-config openssl cmake gstreamer \
  gst-plugins-base gst-plugins-good gst-plugins-bad gst-plugins-ugly \
  log4cplus gst-libav openjdk
export PATH="/opt/homebrew/opt/openjdk/bin:$PATH"
cmake .. -DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_JNI=FALSE -DBUILD_DEPENDENCIES=OFF
make
export GST_PLUGIN_PATH=/opt/homebrew/lib/gstreamer-1.0:${KVS_GST_PLUGIN_PATH}
```
After that, the build and these commands started working for me:
```sh
gst-device-monitor
gst-launch-1.0 videotestsrc ! videoconvert ! autovideosink
gst-launch-1.0 avfvideosrc device-index=0 ! video/x-raw, framerate=30/1, width=1280, height=720 ! queue ! autovideosink
```

GrantPerkins, 2023-06-12:
Some notes I have found: you also need to install GTest on a clean system: `brew install googletest`, and you need to specify OpenSSL location in the PKG_CONFIG_PATH. Thank you for the help@ThePhen!