awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #894: [QUESTION] Is it possible to cross compile sdk for android to build static gstreamer kvssink plugin ?

ChoudharyRamesh, 2022-08-13:
I am working in Qt project where i use kvssink gstreamer plugin to produce live stream . Now i want to compile this sdk for android to build static kvssink plugin for android .

To build for android i used these cmake properties
```
set(CMAKE_SYSTEM_NAME ANDROID)
set(CMAKE_SYSTEM_VERSION 24)
set(CMAKE_ANDROID_ARCH_ABI "armeabi-v7a")
set(ANDROID_PLATFORM android-24)
set(TARGET_ARCH ANDROID)
set(NDK_VERSION 22.1.7171670) 
set(ANDROID_NDK "${ANDROID_SDK_HOME}/ndk/${NDK_VERSION}")
set(CMAKE_ANDROID_NDK "${ANDROID_SDK_HOME}/ndk/${NDK_VERSION}")
set(CMAKE_TOOLCHAIN_FILE "${ANDROID_SDK_HOME}/ndk/${NDK_VERSION}/build/cmake/android.toolchain.cmake")

```
But not able to build target .

ChoudharyRamesh, 2022-08-15:
I managed to compile it for android but stream is not created.
Error :  curl perform failed for url https://kinesisvideo.us-east-2.amazonaws.com/describeStream with result SSL peer certificate or SSH remote key was not OK: SSL certificate problem: unable to get local issuer certificate

video stream log
```

D KinesisVideoClient: createKinesisVideoClient(): Creating Kinesis Video Client
D Heap    : heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
D Heap    : heapInitialize(): Creating AIV heap.
D Heap    : heapInitialize(): Heap is initialized OK
D KinesisVideoClient: createDeviceResultEvent(): Create device result event.
D KinesisVideoClient: createKinesisVideoStream(): Creating Kinesis Video Stream.
D Stream  : logStreamInfo(): SDK version:
V Stream  : logStreamInfo(): Kinesis Video Stream Info
V Stream  : logStreamInfo(): 	Stream name: Android-stream
V Stream  : logStreamInfo(): 	Streaming type: STREAMING_TYPE_REALTIME
V Stream  : logStreamInfo(): 	Content type: video/h264
V Stream  : logStreamInfo(): 	Max latency (100ns): 600000000
V Stream  : logStreamInfo(): 	Fragment duration (100ns): 20000000
V Stream  : logStreamInfo(): 	Key frame fragmentation: Yes
V Stream  : logStreamInfo(): 	Use frame timecode: Yes
V Stream  : logStreamInfo(): 	Absolute frame timecode: Yes
V Stream  : logStreamInfo(): 	Nal adaptation flags: 0
V Stream  : logStreamInfo(): 	Average bandwith (bps): 4194304
V Stream  : logStreamInfo(): 	Framerate: 25
V Stream  : logStreamInfo(): 	Buffer duration (100ns): 1200000000
V Stream  : logStreamInfo(): 	Replay duration (100ns): 400000000
V Stream  : logStreamInfo(): 	Connection Staleness duration (100ns): 600000000
V Stream  : logStreamInfo(): 	Store Pressure Policy: 1
V Stream  : logStreamInfo(): 	View Overflow Policy: 1
V Stream  : logStreamInfo(): 	Segment UUID: NULL
V Stream  : logStreamInfo(): 	Frame ordering mode: 0
V Stream  : logStreamInfo(): Track list
V Stream  : logStreamInfo(): 	Track id: 1
V Stream  : logStreamInfo(): 	Track name: kinesis_video
V Stream  : logStreamInfo(): 	Codec id: V_MPEG4/ISO/AVC
V Stream  : logStreamInfo(): 	Track type: TRACK_INFO_TYPE_VIDEO
V Stream  : logStreamInfo(): 	Track cpd: NULL
I Response: curlCompleteSync(): curl perform failed for url https://kinesisvideo.us-east-2.amazonaws.com/describeStream with result SSL peer certificate or SSH remote key was not OK: SSL certificate problem: unable to get local issuer certificate
I Response: curlCompleteSync(): HTTP Error 0 : Response: (null)
I Response: Request URL: https://kinesisvideo.us-east-2.amazonaws.com/describeStream
```

ChoudharyRamesh, 2022-08-23:
@disa6302 

What you think, i don't know about amazon have android sdk for kinesis , my requirement is to intregate into Qt c++ project , please take a look at Question i asked in first comment . btw thank you for your response.

disa6302, 2022-08-23:
Got it. It is likely this is missing in your directory: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/certs/cert.pem. How have you set the cert path?