awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #897: [QUESTION] Error: Operation too slow. Less than 30 bytes/sec transferred the last 30 seconds

Heena2411, 2022-08-26:
Hi,

I had cross compiled CPP SDK for qualcomm(LINUX) to send video from RTSP source to Kinesis Video Stream.

It was running without any problem initially. From 2-3days, only one frame is getting uploaded. After that getting below logs from SDK.
Can someone please help ?

[DEBUG] [26-08-2022 11:39:16:855.512 GMT] fragmentAckReceivedHandler invoked
[INFO ] [26-08-2022 11:39:16:855.591 GMT] Reporting fragment ack received. Ack timecode315964836864
[DEBUG] [26-08-2022 11:39:17:248.268 GMT] postReadCallback(): Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [26-08-2022 11:39:20:067.331 GMT] postReadCallback(): Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [26-08-2022 11:39:21:486.605 GMT] postReadCallback(): Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [26-08-2022 11:39:23:435.679 GMT] postReadCallback(): Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [26-08-2022 11:39:27:751.286 GMT] postReadCallback(): Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [26-08-2022 11:39:27:855.752 GMT] postReadCallback(): Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [26-08-2022 11:39:28:878.579 GMT] postReadCallback(): Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [26-08-2022 11:39:30:681.767 GMT] postReadCallback(): Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [26-08-2022 11:39:32:439.621 GMT] postReadCallback(): Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [26-08-2022 11:39:33:834.926 GMT] postReadCallback(): Wrote 4322 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [26-08-2022 11:39:34:768.901 GMT] postReadCallback(): Pausing CURL read for upload handle: 0
[DEBUG] [26-08-2022 11:39:36:139.456 GMT] postReadCallback(): Wrote 84 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [26-08-2022 11:39:36:761.074 GMT] postReadCallback(): Pausing CURL read for upload handle: 0
[DEBUG] [26-08-2022 11:39:36:761.478 GMT] postWriteCallback(): Curl post body write function for stream with handle: 350344982589390 and upload handle: 0 returned: {"EventType":"RECEIVED","FragmentTimecode":315964836864,"FragmentNumber":"91343852333181516572604734053777159613864855138"}

[DEBUG] [26-08-2022 11:39:36:761.702 GMT] fragmentAckReceivedHandler invoked
[INFO ] [26-08-2022 11:39:36:761.823 GMT] Reporting fragment ack received. Ack timecode315964836864
[DEBUG] [26-08-2022 11:39:36:762.087 GMT] postWriteCallback(): Curl post body write function for stream with handle: 350344982589390 and upload handle: 0 returned: {"EventType":"PERSISTED","FragmentTimecode":315964836864,"FragmentNumber":"91343852333181516572604734053777159613864855138"}

[DEBUG] [26-08-2022 11:39:36:772.644 GMT] fragmentAckReceivedHandler invoked
[INFO ] [26-08-2022 11:39:36:772.901 GMT] Reporting fragment ack received. Ack timecode315964836864
[DEBUG] [26-08-2022 11:39:37:394.499 GMT] postReadCallback(): Pausing CURL read for upload handle: 0
**[WARN ] [26-08-2022 11:40:13:409.457 GMT] curlCompleteSync(): curl perform failed for url https://s-703775db.kinesisvideo.ap-south-1.amazonaws.com/putMedia with result Error: Operation too slow. Less than 30 bytes/sec transferred the last 30 seconds**
[WARN ] [26-08-2022 11:40:13:409.833 GMT] curlCompleteSync(): HTTP Error 0 : Response: (null)
Request URL: https://s-703775db.kinesisvideo.ap-south-1.amazonaws.com/putMedia
Request Headers:

disa6302, 2022-08-29:
@Heena2411 ,

1. Can you verify your RTSP source is still sending frames?
2. Can you verify the network is stable?

disa6302, 2022-09-19:
Closing due to no response.

closed 2022-09-19 by disa6302