awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #898: ERROR: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstUDPSrc:udpsrc0: Internal data stream error.

Ashish1308, 2022-08-30:
I'm trying to stream a CCTV camera with RTSP URL using Amazon Kinesis Video Streams. I've tested the URL in VLC Player and it works fine there. Have also tested the same using ffplay.

The command that I'm using is this.

AWS_ACCESS_KEY_ID=myAccessKey AWS_SECRET_ACCESS_KEY=MySecretKey ./kvs_gstreamer_sample CCTVStream rtsp://myrtspUrl

But upon running this command I am getting this error in my terminal.

And I am failing to understand why exactly is this happening. Can anyone help me understand what am I doing wrong here?
Thanks for any help

[INFO ] [29-08-2022 19:42:42:126.828 GMT] Using aws credentials for Kinesis Video Streams
[INFO ] [29-08-2022 19:42:42:126.847 GMT] No session token was detected.
[INFO ] [29-08-2022 19:42:42:128.475 GMT] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ] [29-08-2022 19:42:42:128.573 GMT] heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ] [29-08-2022 19:42:42:128.593 GMT] heapInitialize(): Creating AIV heap.
[INFO ] [29-08-2022 19:42:42:128.660 GMT] heapInitialize(): Heap is initialized OK
[DEBUG] [29-08-2022 19:42:42:128.880 GMT] getSecurityTokenHandler invoked
[DEBUG] [29-08-2022 19:42:42:128.993 GMT] Refreshing credentials. Force refreshing: 0 Now time is: 1661802162128949048 Expiration: 0
[INFO ] [29-08-2022 19:42:42:129.030 GMT] New credentials expiration is 1661805762
[INFO ] [29-08-2022 19:42:42:129.095 GMT] createDeviceResultEvent(): Create device result event.
[DEBUG] [29-08-2022 19:42:42:129.137 GMT] clientReadyHandler invoked
[DEBUG] [29-08-2022 19:42:42:129.221 GMT] Client is ready
[INFO ] [29-08-2022 19:42:42:129.391 GMT] Creating Kinesis Video Stream CCTVStream
[INFO ] [29-08-2022 19:42:42:129.473 GMT] createKinesisVideoStream(): Creating Kinesis Video Stream.
[INFO ] [29-08-2022 19:42:42:129.508 GMT] logStreamInfo(): SDK version: 70f74f14cf27b09f71dc1889f36eb6e04cdd90a8
[DEBUG] [29-08-2022 19:42:42:129.533 GMT] logStreamInfo(): Kinesis Video Stream Info
[DEBUG] [29-08-2022 19:42:42:129.557 GMT] logStreamInfo():  Stream name: CCTVStream 
[DEBUG] [29-08-2022 19:42:42:129.581 GMT] logStreamInfo():  Streaming type: STREAMING_TYPE_REALTIME 
[DEBUG] [29-08-2022 19:42:42:129.605 GMT] logStreamInfo():  Content type: video/h264 
[DEBUG] [29-08-2022 19:42:42:129.631 GMT] logStreamInfo():  Max latency (100ns): 600000000
[DEBUG] [29-08-2022 19:42:42:129.657 GMT] logStreamInfo():  Fragment duration (100ns): 20000000
[DEBUG] [29-08-2022 19:42:42:129.682 GMT] logStreamInfo():  Key frame fragmentation: Yes
[DEBUG] [29-08-2022 19:42:42:129.706 GMT] logStreamInfo():  Use frame timecode: Yes
[DEBUG] [29-08-2022 19:42:42:129.730 GMT] logStreamInfo():  Absolute frame timecode: Yes
[DEBUG] [29-08-2022 19:42:42:129.755 GMT] logStreamInfo():  Nal adaptation flags: 0
[DEBUG] [29-08-2022 19:42:42:129.853 GMT] logStreamInfo():  Average bandwith (bps): 4194304
[DEBUG] [29-08-2022 19:42:42:129.880 GMT] logStreamInfo():  Framerate: 25
[DEBUG] [29-08-2022 19:42:42:129.903 GMT] logStreamInfo():  Buffer duration (100ns): 1200000000
[DEBUG] [29-08-2022 19:42:42:129.926 GMT] logStreamInfo():  Replay duration (100ns): 400000000
[DEBUG] [29-08-2022 19:42:42:129.950 GMT] logStreamInfo():  Connection Staleness duration (100ns): 600000000
[DEBUG] [29-08-2022 19:42:42:129.984 GMT] logStreamInfo():  Store Pressure Policy: 1
[DEBUG] [29-08-2022 19:42:42:130.007 GMT] logStreamInfo():  View Overflow Policy: 1
[DEBUG] [29-08-2022 19:42:42:130.030 GMT] logStreamInfo():  Segment UUID: NULL
[DEBUG] [29-08-2022 19:42:42:130.053 GMT] logStreamInfo():  Frame ordering mode: 0
[DEBUG] [29-08-2022 19:42:42:130.076 GMT] logStreamInfo(): Track list
[DEBUG] [29-08-2022 19:42:42:130.098 GMT] logStreamInfo():  Track id: 1
[DEBUG] [29-08-2022 19:42:42:130.120 GMT] logStreamInfo():  Track name: kinesis_video
[DEBUG] [29-08-2022 19:42:42:130.144 GMT] logStreamInfo():  Codec id: V_MPEG4/ISO/AVC
[DEBUG] [29-08-2022 19:42:42:130.167 GMT] logStreamInfo():  Track type: TRACK_INFO_TYPE_VIDEO
[DEBUG] [29-08-2022 19:42:42:130.190 GMT] logStreamInfo():  Track cpd: NULL
[INFO ] [29-08-2022 19:42:42:284.250 GMT] writeHeaderCallback(): RequestId: fdfda324-a980-4b67-aa36-5b7566c28c45
[DEBUG] [29-08-2022 19:42:42:284.421 GMT] describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.661188397273E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","IngestionConfiguration":null,"KmsKeyId":"arn:aws:kms:ap-south-1:610891839088:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:ap-south-1:610891839088:stream/CCTVStream/1661188397273","StreamName":"CCTVStream","Version":"KDDt86hr6jaUJ514hMDd"}}
[INFO ] [29-08-2022 19:42:42:286.022 GMT] describeStreamResultEvent(): Describe stream result event.
[INFO ] [29-08-2022 19:42:42:330.474 GMT] writeHeaderCallback(): RequestId: abb59a91-e0b1-48c6-b1f3-60b307ef9187
[DEBUG] [29-08-2022 19:42:42:330.612 GMT] getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-703775db.kinesisvideo.ap-south-1.amazonaws.com"}
[INFO ] [29-08-2022 19:42:42:332.244 GMT] getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG] [29-08-2022 19:42:42:332.326 GMT] getStreamingTokenHandler invoked
[DEBUG] [29-08-2022 19:42:42:332.362 GMT] Refreshing credentials. Force refreshing: 1 Now time is: 1661802162332355752 Expiration: 1661805762
[INFO ] [29-08-2022 19:42:42:332.387 GMT] New credentials expiration is 1661805762
[INFO ] [29-08-2022 19:42:42:332.433 GMT] getStreamingTokenResultEvent(): Get streaming token result event.
[DEBUG] [29-08-2022 19:42:42:332.476 GMT] streamReadyHandler invoked
[DEBUG] [29-08-2022 19:42:42:332.708 GMT] Stream is ready
[INFO ] [29-08-2022 19:42:42:352.499 GMT] Streaming from rtsp source
Error received from element source: Not found
Debugging information: gstrtspsrc.c(6536): gst_rtspsrc_send (): /GstPipeline:rtsp-kinesis-pipeline/GstRTSPSrc:source:
Not Found (404)
[INFO ] [29-08-2022 19:42:42:636.131 GMT] stopKinesisVideoStreamSync(): Synchronously stopping Kinesis Video Stream 000055e59cb8f5c0.
[DEBUG] [29-08-2022 19:42:42:636.241 GMT] streamClosedHandler invoked for upload handle: 18446744073709551615
[INFO ] [29-08-2022 19:42:42:636.285 GMT] Freeing Kinesis Video Stream CCTVStream
[INFO ] [29-08-2022 19:42:42:636.331 GMT] freeKinesisVideoStream(): Freeing Kinesis Video stream.
[DEBUG] [29-08-2022 19:42:42:636.358 GMT] curlApiCallbacksShutdownActiveRequests(): pActiveRequests hashtable is empty
[INFO ] [29-08-2022 19:42:43:037.946 GMT] Freeing Kinesis Video Stream CCTVStream
[INFO ] [29-08-2022 19:42:43:038.100 GMT] freeKinesisVideoClient(): Freeing Kinesis Video Client
[DEBUG] [29-08-2022 19:42:43:038.132 GMT] curlApiCallbacksShutdownActiveRequests(): pActiveRequests hashtable is empty
[DEBUG] [29-08-2022 19:42:43:439.995 GMT] freeKinesisVideoClientInternal(): Total allocated memory 0
[WARN ] [29-08-2022 19:42:43:440.118 GMT] curlApiCallbacksShutdown(): curlApiCallbacksShutdown called when already in progress of shutting down

MushMal, 2022-08-30:
A casual look at your log shows your upstream source is not found which seems to be an rtsp.

Error received from element source: Not found
Debugging information: gstrtspsrc.c(6536): gst_rtspsrc_send (): /GstPipeline:rtsp-kinesis-pipeline/GstRTSPSrc:source:
Not Found (404)

Try to get the gstreamer pipeline working without the kvs element and only then plug it in

Ashish1308, 2022-08-30:
I'm not very fluent with Gstreamer. I have checked that the source is working fine in VLC. And is also working fine with ffplay. 

Could you please help me get the gstreamer pipeline working without the kvs element?

Ashish1308, 2022-08-30:
Okay I tried running gstreamer using this command without kvs element:

`gst-launch-1.0 -v rtspsrc location="myRTSPURL" ! queue max-size-buffers=2 ! rtph264depay ! h264parse ! decodebin ! autovideosink sync=false` 

And this is the error it genenated. Might it help?
Thanks in advance.

```Setting pipeline to PAUSED ...

(gst-launch-1.0:30087): GStreamer-CRITICAL **: 12:47:05.063: gst_pad_query_caps: assertion 'GST_IS_PAD (pad)' failed

(gst-launch-1.0:30087): GStreamer-CRITICAL **: 12:47:05.063: gst_object_unref: assertion 'object != NULL' failed

(gst-launch-1.0:30087): GStreamer-CRITICAL **: 12:47:05.063: gst_caps_can_intersect: assertion 'GST_IS_CAPS (caps2)' failed

(gst-launch-1.0:30087): GStreamer-CRITICAL **: 12:47:05.063: gst_mini_object_unref: assertion 'mini_object != NULL' failed
Pipeline is live and does not need PREROLL ...
Progress: (open) Opening Stream
Progress: (connect) Connecting to myRTSPURL
Progress: (open) Retrieving server options
Progress: (open) Retrieving media info
Progress: (request) SETUP stream 0
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstRtpBin:manager: latency = 2000
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstRtpBin:manager: ntp-sync = false
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstRtpBin:manager: rfc7273-sync = false
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstRtpBin:manager: ntp-time-source = NTP time based on realtime clock
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstRtpBin:manager: drop-on-latency = false
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstRtpBin:manager: max-rtcp-rtp-time-diff = 1000
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstRtpBin:manager: max-ts-offset-adjustment = 0
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstRtpBin:manager: buffer-mode = Slave receiver to sender clock
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstUDPSrc:udpsrc0: timeout = 5000000000
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstUDPSrc:udpsrc0: caps = application/x-rtp, media=(string)video, payload=(int)98, clock-rate=(int)90000, encoding-name=(string)H265, profile-id=(string)1, sprop-sps=(string)QgEBAUAAAAMAAAMAAAMAAAMAmaADwIARB8uWu5Gwa5VB, sprop-pps=(string)"RAHAc8BMkA\=\=", sprop-vps=(string)QAEMAf//AUAAAAMAAAMAAAMAAAMAmawJ, a-packetization-supported=(string)DH, a-rtppayload-supported=(string)DH, a-framerate=(string)25.000000, a-recvonly=(string)"", ssrc=(uint)483022173
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstUDPSrc:udpsrc1: caps = application/x-rtcp
Progress: (open) Opened Stream
Setting pipeline to PLAYING ...
New clock: GstSystemClock
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstRtpBin:manager: buffer-mode = Slave receiver to sender clock
Progress: (request) Sending PLAY request
Progress: (request) Sending PLAY request
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstUDPSrc:udpsrc0: caps = application/x-rtp, media=(string)video, payload=(int)98, clock-rate=(int)90000, encoding-name=(string)H265, profile-id=(string)1, sprop-sps=(string)QgEBAUAAAAMAAAMAAAMAAAMAmaADwIARB8uWu5Gwa5VB, sprop-pps=(string)"RAHAc8BMkA\=\=", sprop-vps=(string)QAEMAf//AUAAAAMAAAMAAAMAAAMAmawJ, a-packetization-supported=(string)DH, a-rtppayload-supported=(string)DH, a-framerate=(string)25.000000, a-recvonly=(string)"", ssrc=(uint)483022173, clock-base=(uint)21853, seqnum-base=(uint)21853, npt-start=(guint64)0, play-speed=(double)1, play-scale=(double)1
Progress: (request) Sent PLAY request
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstUDPSrc:udpsrc1.GstPad:src: caps = application/x-rtcp
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstRtpBin:manager.GstGhostPad:recv_rtcp_sink_0.GstProxyPad:proxypad3: caps = application/x-rtcp
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstRtpBin:manager.GstGhostPad:recv_rtcp_sink_0.GstProxyPad:proxypad3: caps = application/x-rtcp
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstRtpBin:manager/GstRtpSession:rtpsession0.GstPad:sync_src: caps = application/x-rtcp
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstRtpBin:manager.GstGhostPad:recv_rtp_sink_0.GstProxyPad:proxypad2: caps = application/x-rtp, media=(string)video, payload=(int)98, clock-rate=(int)90000, encoding-name=(string)H265, profile-id=(string)1, sprop-sps=(string)QgEBAUAAAAMAAAMAAAMAAAMAmaADwIARB8uWu5Gwa5VB, sprop-pps=(string)"RAHAc8BMkA\=\=", sprop-vps=(string)QAEMAf//AUAAAAMAAAMAAAMAAAMAmawJ, a-packetization-supported=(string)DH, a-rtppayload-supported=(string)DH, a-framerate=(string)25.000000, a-recvonly=(string)"", ssrc=(uint)483022173, clock-base=(uint)21853, seqnum-base=(uint)21853, npt-start=(guint64)0, play-speed=(double)1, play-scale=(double)1
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstRtpBin:manager.GstGhostPad:recv_rtp_sink_0.GstProxyPad:proxypad2: caps = application/x-rtp, media=(string)video, payload=(int)98, clock-rate=(int)90000, encoding-name=(string)H265, profile-id=(string)1, sprop-sps=(string)QgEBAUAAAAMAAAMAAAMAAAMAmaADwIARB8uWu5Gwa5VB, sprop-pps=(string)"RAHAc8BMkA\=\=", sprop-vps=(string)QAEMAf//AUAAAAMAAAMAAAMAAAMAmawJ, a-packetization-supported=(string)DH, a-rtppayload-supported=(string)DH, a-framerate=(string)25.000000, a-recvonly=(string)"", ssrc=(uint)483022173, clock-base=(uint)21853, seqnum-base=(uint)21853, npt-start=(guint64)0, play-speed=(double)1, play-scale=(double)1
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstRtpBin:manager/GstRtpSession:rtpsession0.GstPad:recv_rtp_src: caps = application/x-rtp, media=(string)video, payload=(int)98, clock-rate=(int)90000, encoding-name=(string)H265, profile-id=(string)1, sprop-sps=(string)QgEBAUAAAAMAAAMAAAMAAAMAmaADwIAR [... cut ...] nt)483022173, clock-base=(uint)21853, seqnum-base=(uint)21853, npt-start=(guint64)0, play-speed=(double)1, play-scale=(double)1
ERROR: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstUDPSrc:udpsrc0: Internal data stream error.
Additional debug info:
gstbasesrc.c(3072): gst_base_src_loop (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstUDPSrc:udpsrc0:
streaming stopped, reason not-linked (-1)
Execution ended after 0:00:02.138872890
Setting pipeline to NULL ...
Freeing pipeline ...
```

jdelapla, 2022-08-31:
Generic GStreamer support is outside our scope, however a few things that I noticed:

1. your rtspsrc encoding-name is H265, but you're using a H264 depayer
2. queue with a max-buffer of 2 is only allowing 2 frames and is probably a bit small
3. decodebin is unnecessary, h264parse and maybe a capsfilter should be enough.

Ashish1308, 2022-09-01:
Thank you for your input. 

I tried using the H265 depayer, increasing the max-buffer and removing decodebin but to no success.

hassanctech, 2022-09-12:
Since you see this error without the `kvssink` it appears the issue is unrelated to KVS.  I would recommend you ask on the GStreamer forums how to get your pipeline working with just `audiovideosink` or `filesink`.  Once you have that piece working and if `kvssink` still does not work, please open a question here with the logs/errors.  But at this junction I do not see anything related to `kvssink` GStreamer plugin so I'm closing this ticket.

closed 2022-09-12 by hassanctech