awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #900: [QUESTION] kvssink logs errors, but does not pass them to the GStreamer Bus

charlieblevins, 2022-09-06:
I have a GStreamer pipeline that uses kvssink. It adds a handler for error messages, but the handler is never called even though kvssink logs errors. Why is this? Is there some other way of responding to kvssink errors?

An example log:
```
[ERROR] [06-09-2022 19:46:08:881.578 GMT] Reported stream error. Errored timecode: 955882190000 Status: 0x52000060
```

The error handler we have works for errors from other elements, but does not work for errors logged from kvssink. The error handler is added to the pipeline like so:
```python
        bus = self.pipeline.get_bus()
        bus.add_signal_watch()
        bus.connect('message::error', self.on_error, self)
```

I have also tried listening for errors on the bus from kvssink - this also does not get called on error. Any help appreciated - Thank you

disa6302, 2022-09-15:
@charlieblevins , 

We are looking into this and will update when we have a patch out. We are tracking this here: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/375
Thanks for reporting!

disa6302, 2022-09-19:
Closing as dup of the issue tagged.

closed 2022-09-19 by disa6302