awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #903: RTSP Stream: Not able to add audio on AWS kinesis-video-streams

Darktechie-programmer, 2022-09-12:
Hi,

I have used the following command for streaming..

gst-launch-1.0 -v rtspsrc location="rtsp://192.168.68.114:5554/playlist.m3u" short-header=TRUE ! rtph264depay ! h264parse ! kvssink access-key=<Access-key> secret-key=<secret-key> stream-name="ExampleStream" storage-size=512 

And my streaming is successfully displayed on the AWS kinesis-video-streams but i want to listen my audio too. but noting work for me, For audio purpose i have used this command

gst-launch-1.0 -v rtspsrc location="rtsp://192.168.68.114:5554/playlist.m3u" short-header=TRUE ! rtph264depay ! h264parse ! kvssink access-key=<Access-key> secret-key=<secret-key> stream-name="ExampleStream" storage-size=512 alsasrc device=hw:1,0 ! audioconvert ! avenc_aac ! queue ! sink.

WARNING: erroneous pipeline: No sink-element named "(null)" - omitting link
[image]
[image]

can enyone help so i can stream my audio and video.

hassanctech, 2022-09-12:
If you're receiving audio and video from RTSP you will need to split the pipeline and have depayer for h264 as well as for the audio (for example: https://gstreamer.freedesktop.org/documentation/rtp/rtpac3depay.html?gi-language=c), then you need to mux these together to a single stream and then send to kvssink.

Here's and example from someone else who tried to do this, if you follow this it should work for you!  If it does not please re-open: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/687#issuecomment-807859411

Darktechie-programmer, 2022-09-13:
Hi,

i have check all the solution but nothing worked for me, 

Live RTSP link:   rtsp://spicatv.dvrdns.org:555/Al_Ekhbariya_Stream

# Working without audio   # video live working
gst-launch-1.0 -v rtspsrc location="rtsp://spicatv.dvrdns.org:555/Al_Ekhbariya_Stream" short-header=TRUE ! rtph264depay ! h264parse ! kvssink access-key=<access-key> secret-key=<secret-key> stream-name="ExampleStream" storage-size=512 

----- For audio purose used commands:

# First command for audio  # not working

gst-launch-1.0 rtspsrc name=src location="rtsp://spicatv.dvrdns.org:555/Al_Ekhbariya_Stream" short-header=true ! rtph264depay ! h264parse ! kvssink name=sink access-key=<access-key> secret-key=<secret-key> stream-name="ExampleStream" storage-size=1024 aws-region="us-east-2" frame-timecodes=false src. ! rtppcmadepay ! alawdec ! audioresample ! audioconvert ! voaacenc ! aacparse ! queue ! sink.

[image]
[image]

# Second command for audio  # not working

gst-launch-1.0 rtspsrc name=src location="rtsp://spicatv.dvrdns.org:555/Al_Ekhbariya_Stream" short-header=true ! rtph264depay ! h264parse ! kvssink name=sink access-key=<access-key> secret-key=<secret-key> stream-name="ExampleStream" storage-size=1024 aws-region="us-east-2" frame-timecodes=false src. ! rtpmpadepay ! mpg123audiodec ! audioresample ! audioconvert ! voaacenc ! aacparse ! queue ! sink.

[image]
[image]

# third command for audio  # not working

gst-launch-1.0 rtspsrc name=src location="rtsp://spicatv.dvrdns.org:555/Al_Ekhbariya_Stream" short-header=true ! rtph264depay ! h264parse ! kvssink name=sink access-key=<access-key> secret-key=<secret-key> stream-name="ExampleStream" storage-size=1024 aws-region="us-east-2" frame-timecodes=false src. ! rtpaacdepay ! queue ! sink.

[image]

Darktechie-programmer, 2022-09-13:
Hi, 

Video:
  Codec: h264
  Resolution: 852 X 480
  Bitrate: 1083 kbps VBR

Audio:
  Codec: aac stereo
  Sample rate: 48000
  Bitrate: 67 kbps VBR

I have using these configuration for my rtsp but my audio is not coming on aws kinesis-video-stream. I don't know much about gst-launch-1.0 library, can you please share with me complete command. 

Thanks

javieralcantara, 2023-02-22:
Having the same issue here. Did anyone solve it?

jdelapla, 2023-02-23:
Hello, this is a pure gstreamer issue unrelated to the internals of kvssink. Your pads aren't connecting between your rtspsrc and rtpPCMAdepay elements.

Please add a `queue` element between your src and rtpXXXXdepay(s) when doing a tee like this.

`gst-launch-1.0 rtspsrc name=src location="rtsp://192.168.68.114:5554/playlist.m3u" short-header=true ! queue ! rtph264depay ! h264parse ! kvssink name=sink access-key="access-key" secret-key="secret-key" stream-name="ExampleStream" storage-size=1024 aws-region="us-east-2" frame-timecodes=false src. ! queue ! rtppcmadepay ! alawdec ! audioresample ! audioconvert ! voaacenc ! aacparse ! queue ! sink.`

Something like that should work.
Very similar to this issue someone had with a/v pipelines.
https://github.com/awslabs/amazon-kinesis-video-streams-webrtc-sdk-c/issues/1014

closed 2023-02-23 by jdelapla