awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #904: curl perform failed for url .....putMedia with result Timeout was reached: Operation too slow. Less than 30 bytes/sec transferred the last 30 seconds

FineWebworking, 2022-09-12:
Compiled kvs_gstreamer_sample on ubuntu 20.04 have 3 Axis RTSP streams, 2 working great 1 fails with Operation too slow error. 

[TRACE] [12-09-2022 19:15:58:628.833 GMT] putKinesisVideoFrame(): debug frame info pts: 16630101519261260, dts: 13700000, duration: 0, size: 285, trackId: 1, isKey 0
[TRACE] [12-09-2022 19:15:58:629.197 GMT] putKinesisVideoFrame(): debug frame info pts: 16630101519594491, dts: 13710000, duration: 0, size: 365, trackId: 1, isKey 0
[TRACE] [12-09-2022 19:15:58:655.281 GMT] putKinesisVideoFrame(): debug frame info pts: 16630101519761057, dts: 13720000, duration: 0, size: 789, trackId: 1, isKey 0
[TRACE] [12-09-2022 19:15:58:682.518 GMT] putKinesisVideoFrame(): debug frame info pts: 16630101519927623, dts: 13730000, duration: 0, size: 158, trackId: 1, isKey 0
[TRACE] [12-09-2022 19:15:58:708.903 GMT] putKinesisVideoFrame(): debug frame info pts: 16630101520094191, dts: 13740000, duration: 0, size: 181, trackId: 1, isKey 0
[TRACE] [12-09-2022 19:15:58:736.737 GMT] putKinesisVideoFrame(): debug frame info pts: 16630101520427425, dts: 13750000, duration: 0, size: 241, trackId: 1, isKey 0
[TRACE] [12-09-2022 19:15:58:737.045 GMT] putKinesisVideoFrame(): debug frame info pts: 16630101520593994, dts: 13760000, duration: 0, size: 196, trackId: 1, isKey 0
[TRACE] [12-09-2022 19:15:58:764.188 GMT] putKinesisVideoFrame(): debug frame info pts: 16630101520760564, dts: 13770000, duration: 0, size: 351, trackId: 1, isKey 0
[TRACE] [12-09-2022 19:15:58:791.177 GMT] putKinesisVideoFrame(): debug frame info pts: 16630101521093801, dts: 13780000, duration: 0, size: 247, trackId: 1, isKey 0
[TRACE] [12-09-2022 19:15:58:818.682 GMT] putKinesisVideoFrame(): debug frame info pts: 16630101521260372, dts: 13790000, duration: 0, size: 191, trackId: 1, isKey 0
[TRACE] [12-09-2022 19:15:58:846.293 GMT] putKinesisVideoFrame(): debug frame info pts: 16630101521426944, dts: 13800000, duration: 0, size: 253, trackId: 1, isKey 0
[TRACE] [12-09-2022 19:15:58:846.615 GMT] putKinesisVideoFrame(): debug frame info pts: 16630101521760184, dts: 13810000, duration: 0, size: 341, trackId: 1, isKey 0
[TRACE] [12-09-2022 19:15:58:875.026 GMT] putKinesisVideoFrame(): debug frame info pts: 16630101521926757, dts: 13820000, duration: 0, size: 744, trackId: 1, isKey 0
[WARN ] [12-09-2022 19:15:58:895.734 GMT] curlCompleteSync(): curl perform failed for url https://s-1e415f8b.kinesisvideo.us-east-1.amazonaws.com/putMedia with result Timeout was reached: Operation too slow. Less than 30 bytes/sec transferred the last 30 seconds
[WARN ] [12-09-2022 19:15:58:895.909 GMT] curlCompleteSync(): HTTP Error 0 : Response: (null)
Request URL: https://s-1e415f8b.kinesisvideo.us-east-1.amazonaws.com/putMedia
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AKIATOKZ5A7X4PHX67NL/20220912/us-east-1/kinesisvideo/aws4_request, SignedHeaders=connection;host;transfer-encoding;user-agent;x-amz-date;x-amzn-fragment-acknowledgment-required;x-amzn-fragment-timecode-type;x-amzn-producer-start-timestamp;x-amzn-stream-name, Signature=2f7b93fa4bc05e5a270a0cc997ddec09be8a8d1e4929ab4
[DEBUG] [12-09-2022 19:15:58:896.270 GMT] putStreamCurlHandler(): Network thread for Kinesis Video stream: camera_stone_cove_south with upload handle: 0 exited. http status: 0
[WARN ] [12-09-2022 19:15:58:896.516 GMT] putStreamCurlHandler(): Stream with streamHandle 94713047481072 uploadHandle 0 has exited without triggering end-of-stream. Service call result: 599
[INFO ] [12-09-2022 19:15:58:896.634 GMT] kinesisVideoStreamTerminated(): Stream 0x5624198cfaf0 terminated upload handle 0 with service call result 599.
[DEBUG] [12-09-2022 19:15:58:896.770 GMT] defaultStreamStateTransitionHook(): Stream state machine retry count: 0
[DEBUG] [12-09-2022 19:15:58:896.854 GMT] defaultStreamStateTransitionHook(): 
 KinesisVideoStream base result is [599]. Executing KVS retry handler of retry strategy type [1]
[TRACE] [12-09-2022 19:15:58:897.002 GMT] validateAndUpdateExponentialBackoffStatus(): Status changed from BACKOFF_NOT_STARTED to BACKOFF_IN_PROGRESS
[TRACE] [12-09-2022 19:15:58:897.100 GMT] getExponentialBackoffRetryStrategyWaitTime(): 
 Thread Id [139776657176128] Number of retries [1], Retry wait time [189] ms, Retry system time [16630101588970921]
[TRACE] [12-09-2022 19:15:58:897.249 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000100, Next state: 0x0000000000000200, Current local state retry count [0], Max local state retry count [0], State transition wait time [189] ms
[TRACE] [12-09-2022 19:15:58:900.875 GMT] putKinesisVideoFrame(): debug frame info pts: 16630101522093331, dts: 13830000, duration: 0, size: 133, trackId: 1, isKey 0
[DEBUG] [12-09-2022 19:15:58:900.966 GMT] defaultStreamStateTransitionHook(): Stream state machine retry count: 1
[DEBUG] [12-09-2022 19:15:58:901.028 GMT] defaultStreamStateTransitionHook(): 
 KinesisVideoStream base result is [599]. Executing KVS retry handler of retry strategy type [1]
[TRACE] [12-09-2022 19:15:58:901.129 GMT] validateAndUpdateExponentialBackoffStatus(): Current status is BACKOFF_IN_PROGRESS
[TRACE] [12-09-2022 19:15:58:901.182 GMT] getExponentialBackoffRetryStrategyWaitTime(): 
 Thread Id [139776905213504] Number of retries [2], Retry wait time [300] ms, Retry system time [16630101589011786]
[TRACE] [12-09-2022 19:15:58:901.270 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000200, Next state: 0x0000000000000040, Current local state retry count [0], Max local state retry count [5], State transition wait time [300] ms
[DEBUG] [12-09-2022 19:15:58:901.326 GMT] streamReadyHandler invoked
[DEBUG] [12-09-2022 19:15:58:901.381 GMT] defaultStreamStateTransitionHook(): Stream state machine retry count: 2
[DEBUG] [12-09-2022 19:15:58:901.442 GMT] defaultStreamStateTransitionHook(): 
 KinesisVideoStream base result is [599]. Executing KVS retry handler of retry strategy type [1]
[TRACE] [12-09-2022 19:15:58:901.599 GMT] validateAndUpdateExponentialBackoffStatus(): Current status is BACKOFF_IN_PROGRESS
[TRACE] [12-09-2022 19:15:58:901.688 GMT] getExponentialBackoffRetryStrategyWaitTime(): 
 Thread Id [139776905213504] Number of retries [3], Retry wait time [580] ms, Retry system time [16630101589016845]
[TRACE] [12-09-2022 19:15:58:901.852 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000040, Next state: 0x0000000000000080, Current local state retry count [0], Max local state retry count [0], State transition wait time [580] ms
[TRACE] [12-09-2022 19:15:58:901.951 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000040, Next state: 0x0000000000000040, Current local state retry count [5], Max local state retry count [0], State transition wait time [0] ms
[INFO ] [12-09-2022 19:15:58:902.083 GMT] putStreamResultEvent(): Put stream result event. New upload handle 1
[TRACE] [12-09-2022 19:15:58:902.155 GMT] notifyDataAvailable(): Note data received: duration(100ns): 0 bytes 0 for stream handle 1
[TRACE] [12-09-2022 19:15:58:912.312 GMT] streamDataAvailableHandler invoked for stream: camera_stone_cove_south and stream upload handle: 1
[TRACE] [12-09-2022 19:15:58:912.397 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000080, Next state: 0x0000000000000100, Current local state retry count [0], Max local state retry count [0], State transition wait time [0] ms
[TRACE] [12-09-2022 19:15:58:928.456 GMT] putKinesisVideoFrame(): debug frame info pts: 16630101522260314, dts: 13840000, duration: 0, size: 146, trackId: 1, isKey 0
[TRACE] [12-09-2022 19:15:58:928.665 GMT] putKinesisVideoFrame(): debug frame info pts: 16630101522593936, dts: 13850000, duration: 0, size: 294, trackId: 1, isKey 0
[TRACE] [12-09-2022 19:15:58:955.934 GMT] putKinesisVideoFrame(): debug frame info pts: 16630101522760889, dts: 13860000, duration: 0, size: 211, trackId: 1, isKey 0
[TRACE] [12-09-2022 19:15:58:983.211 GMT] putKinesisVideoFrame(): debug frame info pts: 16630101522927840, dts: 13870000, duration: 0, size: 300, trackId: 1, isKey 0
[TRACE] [12-09-2022 19:15:59:010.614 GMT] putKinesisVideoFrame(): debug frame info pts: 16630101523261455, dts: 13880000, duration: 0, size: 292, trackId: 1, isKey 0
[TRACE] [12-09-2022 19:15:59:038.985 GMT] putKinesisVideoFrame(): debug frame info pts: 16630101523428401, dts: 13890000, duration: 0, size: 275, trackId: 1, isKey 0
[TRACE] [12-09-2022 19:15:59:039.234 GMT] putKinesisVideoFrame(): debug frame info pts: 16630101523595345, dts: 13900000, duration: 0, size: 247, trackId: 1, isKey 0
[TRACE] [12-09-2022 19:15:59:065.512 GMT] putKinesisVideoFrame(): debug frame info pts: 16630101523762287, dts: 13910000, duration: 0, size: 265, trackId: 1, isKey 0

Same code is running fine on two rtsp streams but one is throwing this error.

Log is attached:  
[kvs.log](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/9551145/kvs.log)

disa6302, 2022-09-15:
@FineWebworking , 

I do not see any key frames in your logs. The SDK skips frames till the first key frame is encountered. Can you ensure the 3rd RTSP camera is sending out a key frame?

FineWebworking, 2022-09-16:
Below is a frame analysis of the feed using ffprobe.  As I understand the output I think it shows that keyframes are in the stream.

 ffprobe -v error -hide_banner -select_streams v:0 -show_frames  rtsp://72.87.114.60:551/axis-media/media.amp
[FRAME]
media_type=video
stream_index=0
key_frame=1
pkt_pts=18000
pkt_pts_time=0.200000
pkt_dts=18000
pkt_dts_time=0.200000
best_effort_timestamp=18000
best_effort_timestamp_time=0.200000
pkt_duration=3000
pkt_duration_time=0.033333
pkt_pos=N/A
pkt_size=528596
width=2304
height=1728
pix_fmt=yuvj420p
sample_aspect_ratio=N/A
pict_type=I
coded_picture_number=0
display_picture_number=0
interlaced_frame=0
top_field_first=0
repeat_pict=0
color_range=pc
color_space=bt709
color_primaries=bt709
color_transfer=bt709
chroma_location=center
[/FRAME]
[FRAME]
media_type=video
stream_index=0
key_frame=0
pkt_pts=3000
pkt_pts_time=0.033333
pkt_dts=3000
pkt_dts_time=0.033333
best_effort_timestamp=3000
best_effort_timestamp_time=0.033333
pkt_duration=3000
pkt_duration_time=0.033333
pkt_pos=N/A
pkt_size=2678
width=2304
height=1728
pix_fmt=yuvj420p
sample_aspect_ratio=N/A
pict_type=P
coded_picture_number=1
display_picture_number=0
interlaced_frame=0
top_field_first=0
repeat_pict=0
color_range=pc
color_space=bt709
color_primaries=bt709
color_transfer=bt709
chroma_location=center
[/FRAME]
[FRAME]
media_type=video
stream_index=0
key_frame=0
pkt_pts=6000
pkt_pts_time=0.066667
pkt_dts=6000
pkt_dts_time=0.066667
best_effort_timestamp=6000
best_effort_timestamp_time=0.066667
pkt_duration=3000
pkt_duration_time=0.033333
pkt_pos=N/A
pkt_size=4447
width=2304
height=1728
pix_fmt=yuvj420p
sample_aspect_ratio=N/A
pict_type=P
coded_picture_number=2
display_picture_number=0
interlaced_frame=0
top_field_first=0
repeat_pict=0
color_range=pc
color_space=bt709
color_primaries=bt709
color_transfer=bt709
chroma_location=center
[/FRAME]
[FRAME]
media_type=video
stream_index=0
key_frame=0
pkt_pts=9000
pkt_pts_time=0.100000
pkt_dts=9000
pkt_dts_time=0.100000
best_effort_timestamp=9000
best_effort_timestamp_time=0.100000
pkt_duration=3000
pkt_duration_time=0.033333
pkt_pos=N/A
pkt_size=5230
width=2304
height=1728
pix_fmt=yuvj420p
sample_aspect_ratio=N/A
pict_type=P
coded_picture_number=3
display_picture_number=0
interlaced_frame=0
top_field_first=0
repeat_pict=0
color_range=pc
color_space=bt709
color_primaries=bt709
color_transfer=bt709
chroma_location=center
[/FRAME]
[FRAME]
media_type=video
stream_index=0
key_frame=0
pkt_pts=12000
pkt_pts_time=0.133333
pkt_dts=12000
pkt_dts_time=0.133333
best_effort_timestamp=12000
best_effort_timestamp_time=0.133333
pkt_duration=3000
pkt_duration_time=0.033333
pkt_pos=N/A
pkt_size=5105
width=2304
height=1728
pix_fmt=yuvj420p
sample_aspect_ratio=N/A
pict_type=P
coded_picture_number=4
display_picture_number=0
interlaced_frame=0
top_field_first=0
repeat_pict=0
color_range=pc
color_space=bt709
color_primaries=bt709
color_transfer=bt709
chroma_location=center
[/FRAME]
[FRAME]
media_type=video
stream_index=0
key_frame=0
pkt_pts=15000
pkt_pts_time=0.166667
pkt_dts=15000
pkt_dts_time=0.166667
best_effort_timestamp=15000
best_effort_timestamp_time=0.166667
pkt_duration=3000
pkt_duration_time=0.033333
pkt_pos=N/A
pkt_size=6223
width=2304
height=1728
pix_fmt=yuvj420p
sample_aspect_ratio=N/A
pict_type=P
coded_picture_number=5
display_picture_number=0
interlaced_frame=0
top_field_first=0
repeat_pict=0
color_range=pc
color_space=bt709
color_primaries=bt709
color_transfer=bt709
chroma_location=center
[/FRAME]

disa6302, 2022-09-20:
@FineWebworking,
The logs do you not indicate any key frame though (isKey is 0 in the logs). It is likely the frame is not being set to key frame for your camera somehow. Would suggest debugging your source and adding some debug in the sample to understand where the disconnect is. In our sample, the flag is set here: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/70f74f14cf27b09f71dc1889f36eb6e04cdd90a8/samples/kvs_gstreamer_sample.cpp#L374

disa6302, 2022-09-25:
Closing due to no response. Feel free to reach out if you have any more questions.

closed 2022-09-25 by disa6302