awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #907: [QUESTION] curlApiCallbacksShutdown(): curlApiCallbacksShutdown called when already in progress of shutting down

DreyOlad, 2022-09-16:
hello, i'm trying to stream from a cctv using its ip and i'm getting this error 

** Logging **
[INFO ] [16-09-2022 17:12:23:364.724 UTC] Using region: eu-west-1
[INFO ] [16-09-2022 17:12:23:365.053 UTC] Using IoT credentials for Kinesis Video Streams
[DEBUG] [16-09-2022 17:12:23:367.174 UTC] Creating IoT auth callbacks.
[INFO ] [16-09-2022 17:12:24:261.058 UTC] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ] [16-09-2022 17:12:24:261.162 UTC] heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ] [16-09-2022 17:12:24:261.183 UTC] heapInitialize(): Creating AIV heap.
[INFO ] [16-09-2022 17:12:24:261.220 UTC] heapInitialize(): Heap is initialized OK
[INFO ] [16-09-2022 17:12:24:261.456 UTC] createDeviceResultEvent(): Create device result event.
[DEBUG] [16-09-2022 17:12:24:261.508 UTC] clientReadyHandler invoked
[DEBUG] [16-09-2022 17:12:24:261.542 UTC] Client is ready
[INFO ] [16-09-2022 17:12:24:261.611 UTC] Creating Kinesis Video Stream b_a_poc_thing
[INFO ] [16-09-2022 17:12:24:261.638 UTC] createKinesisVideoStream(): Creating Kinesis Video Stream.
[INFO ] [16-09-2022 17:12:24:261.661 UTC] logStreamInfo(): SDK version: 70f74f14cf27b09f71dc1889f36eb6e04cdd90a8
[DEBUG] [16-09-2022 17:12:24:261.676 UTC] logStreamInfo(): Kinesis Video Stream Info
[DEBUG] [16-09-2022 17:12:24:261.690 UTC] logStreamInfo():      Stream name: b_a_poc_thing 
[DEBUG] [16-09-2022 17:12:24:261.712 UTC] logStreamInfo():      Streaming type: STREAMING_TYPE_REALTIME 
[DEBUG] [16-09-2022 17:12:24:261.729 UTC] logStreamInfo():      Content type: video/h264 
[DEBUG] [16-09-2022 17:12:24:261.743 UTC] logStreamInfo():      Max latency (100ns): 600000000
[DEBUG] [16-09-2022 17:12:24:261.757 UTC] logStreamInfo():      Fragment duration (100ns): 20000000
[DEBUG] [16-09-2022 17:12:24:261.770 UTC] logStreamInfo():      Key frame fragmentation: Yes
[DEBUG] [16-09-2022 17:12:24:261.784 UTC] logStreamInfo():      Use frame timecode: Yes
[DEBUG] [16-09-2022 17:12:24:261.797 UTC] logStreamInfo():      Absolute frame timecode: Yes
[DEBUG] [16-09-2022 17:12:24:261.811 UTC] logStreamInfo():      Nal adaptation flags: 0
[DEBUG] [16-09-2022 17:12:24:261.824 UTC] logStreamInfo():      Average bandwith (bps): 4194304
[DEBUG] [16-09-2022 17:12:24:261.837 UTC] logStreamInfo():      Framerate: 25
[DEBUG] [16-09-2022 17:12:24:261.851 UTC] logStreamInfo():      Buffer duration (100ns): 1200000000
[DEBUG] [16-09-2022 17:12:24:261.864 UTC] logStreamInfo():      Replay duration (100ns): 400000000
[DEBUG] [16-09-2022 17:12:24:261.885 UTC] logStreamInfo():      Connection Staleness duration (100ns): 600000000
[DEBUG] [16-09-2022 17:12:24:261.900 UTC] logStreamInfo():      Store Pressure Policy: 1
[DEBUG] [16-09-2022 17:12:24:261.914 UTC] logStreamInfo():      View Overflow Policy: 1
[DEBUG] [16-09-2022 17:12:24:261.927 UTC] logStreamInfo():      Segment UUID: NULL
[DEBUG] [16-09-2022 17:12:24:261.952 UTC] logStreamInfo():      Frame ordering mode: 0
[DEBUG] [16-09-2022 17:12:24:261.969 UTC] logStreamInfo(): Track list
[DEBUG] [16-09-2022 17:12:24:261.983 UTC] logStreamInfo():      Track id: 1
[DEBUG] [16-09-2022 17:12:24:261.997 UTC] logStreamInfo():      Track name: kinesis_video
[DEBUG] [16-09-2022 17:12:24:262.010 UTC] logStreamInfo():      Codec id: V_MPEG4/ISO/AVC
[DEBUG] [16-09-2022 17:12:24:262.024 UTC] logStreamInfo():      Track type: TRACK_INFO_TYPE_VIDEO
[DEBUG] [16-09-2022 17:12:24:262.046 UTC] logStreamInfo():      Track cpd: NULL
[INFO ] [16-09-2022 17:12:24:774.300 UTC] writeHeaderCallback(): RequestId: b70b5a27-2272-4b5c-ad48-0d455a25b56e
[DEBUG] [16-09-2022 17:12:24:774.509 UTC] describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.663328119213E9,"DataRetentionInHours":8760,"DeviceName":null,"IngestionChannelARN":null,"IngestionConfiguration":null,"KmsKeyId":"arn:aws:kms:eu-west-1:551701396087:alias/aws/kinesisvideo","MediaType":null,"Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:eu-west-1:551701396087:stream/aws_lcc_poc_thing/1663328119213","StreamName":"b_a_poc_thing","Version":"G0CDDBJDf91UVN4H7TYJ"}}
[INFO ] [16-09-2022 17:12:24:777.988 UTC] describeStreamResultEvent(): Describe stream result event.
[WARN ] [16-09-2022 17:12:24:778.095 UTC] describeStreamResult(): Retention period returned from the DescribeStream call doesn't match the one specified in the StreamInfo
[WARN ] [16-09-2022 17:12:24:778.249 UTC] describeStreamResult(): Content type returned from the DescribeStream call doesn't match the one specified in the StreamInfo
[INFO ] [16-09-2022 17:12:25:310.512 UTC] writeHeaderCallback(): RequestId: 9f97e598-8d99-4c73-9720-04c26fc7994b
[DEBUG] [16-09-2022 17:12:25:310.662 UTC] getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-5ac677c0.kinesisvideo.eu-west-1.amazonaws.com"}
[INFO ] [16-09-2022 17:12:25:314.144 UTC] getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[INFO ] [16-09-2022 17:12:25:314.251 UTC] getStreamingTokenResultEvent(): Get streaming token result event.
[DEBUG] [16-09-2022 17:12:25:314.311 UTC] streamReadyHandler invoked
[DEBUG] [16-09-2022 17:12:25:314.510 UTC] Stream is ready
[INFO ] [16-09-2022 17:12:25:328.106 UTC] Streaming from rtsp source
Not all elements could be created.
[INFO ] [16-09-2022 17:12:25:331.994 UTC] stopKinesisVideoStreamSync(): Synchronously stopping Kinesis Video Stream 00007fa98a039210.
[DEBUG] [16-09-2022 17:12:25:332.066 UTC] streamClosedHandler invoked for upload handle: 18446744073709551615
[INFO ] [16-09-2022 17:12:25:332.101 UTC] Freeing Kinesis Video Stream b_a_poc_thing
[INFO ] [16-09-2022 17:12:25:332.125 UTC] freeKinesisVideoStream(): Freeing Kinesis Video stream.
[DEBUG] [16-09-2022 17:12:25:332.144 UTC] curlApiCallbacksShutdownActiveRequests(): pActiveRequests hashtable is empty
[INFO ] [16-09-2022 17:12:25:737.708 UTC] Freeing Kinesis Video Stream b_a_poc_thing
[INFO ] [16-09-2022 17:12:25:744.547 UTC] freeKinesisVideoClient(): Freeing Kinesis Video Client
[DEBUG] [16-09-2022 17:12:25:744.615 UTC] curlApiCallbacksShutdownActiveRequests(): pActiveRequests hashtable is empty
[DEBUG] [16-09-2022 17:12:26:153.639 UTC] freeKinesisVideoClientInternal(): Total allocated memory 0
[WARN ] [16-09-2022 17:12:26:153.770 UTC] curlApiCallbacksShutdown(): curlApiCallbacksShutdown called when already in progress of shutting down
-----------------------------------------------------
and GST_DEBUG=4 is giving this

[INFO ] [16-09-2022 17:51:34:541.373 UTC] Using region: eu-west-1
[INFO ] [16-09-2022 17:51:34:542.047 UTC] Using IoT credentials for Kinesis Video Streams
[DEBUG] [16-09-2022 17:51:34:545.907 UTC] Creating IoT auth callbacks.
[... 58 lines cut ...]
0:00:00.026571000 77954 0x7fdfb9007200 INFO            GST_REGISTRY gstregistry.c:1862:ensure_current_registry: registry reading and updating done, result = 1
0:00:00.026579000 77954 0x7fdfb9007200 INFO                GST_INIT gst.c:834:init_post: GLib runtime version: 2.72.3
0:00:00.026586000 77954 0x7fdfb9007200 INFO                GST_INIT gst.c:836:init_post: GLib headers version: 2.72.2
0:00:00.026597000 77954 0x7fdfb9007200 INFO                GST_INIT gst.c:837:init_post: initialized GStreamer successfully
[INFO ] [16-09-2022 17:51:37:129.895 UTC] Streaming from rtsp source
0:00:00.026744000 77954 0x7fdfb9007200 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:489:gst_element_factory_create_with_properties: creating element "pipeline"
0:00:00.029762000 77954 0x7fdfb9007200 INFO      GST_PLUGIN_LOADING gstplugin.c:987:_priv_gst_plugin_load_file_for_registry: plugin "/usr/local/opt/gstreamer/lib/gstreamer-1.0/libgstcoreelements.dylib" loaded
0:00:00.029877000 77954 0x7fdfb9007200 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:489:gst_element_factory_create_with_properties: creating element "capsfilter"
0:00:00.029938000 77954 0x7fdfb9007200 INFO        GST_ELEMENT_PADS gstelement.c:760:gst_element_add_pad:<GstBaseTransform@0x7fdfb9816170> adding pad 'sink'
0:00:00.029957000 77954 0x7fdfb9007200 INFO        GST_ELEMENT_PADS gstelement.c:760:gst_element_add_pad:<GstBaseTransform@0x7fdfb9816170> adding pad 'src'
0:00:00.029974000 77954 0x7fdfb9007200 WARN     GST_ELEMENT_FACTORY gstelementfactory.c:754:gst_element_factory_make_valist: no such element factory "appsink"!
[... cut ...]
[INFO ] [16-09-2022 17:51:37:540.520 UTC] Freeing Kinesis Video Stream b_a_poc_thing
[INFO ] [16-09-2022 17:51:37:540.792 UTC] freeKinesisVideoClient(): Freeing Kinesis Video Client
[DEBUG] [16-09-2022 17:51:37:540.913 UTC] curlApiCallbacksShutdownActiveRequests(): pActiveRequests hashtable is empty
[DEBUG] [16-09-2022 17:51:37:946.323 UTC] freeKinesisVideoClientInternal(): Total allocated memory 0
[WARN ] [16-09-2022 17:51:37:946.504 UTC] curlApiCallbacksShutdown(): curlApiCallbacksShutdown called when already in progress of shutting down

MushMal, 2022-09-16:
GST_DEBUG=4 doesn't seem to show what element failed. My guess would be a caps negotiation which is not really related to KVS sink. I would recommend establishing the pipeline without KVS (use fakesink or similar), once it's working, plug the kvs sink and use gst_debug = 5 to get more information which element fails in the pipeline.

DreyOlad, 2022-09-16:
thank you so much for the help

> GST_DEBUG=4 doesn't seem to show what element failed. My guess would be a caps negotiation which is not really related to KVS sink. I would recommend establishing the pipeline without KVS (use [...]

@MushMal  so how do i implement this in my case

MushMal, 2022-09-19:
Sorry for the delay. Reading back, I am not entirely sure whether you are using kvs sink plugin or the gst sample application. In either case, it would be useful to use gst-launch to ensure the pipeline gets constructed and run with the fakesink. Once you have it working you can then plug the kvs sink plugin to the pipeline as a sink. For the sample application you might want to modify the pipeline construction string in the body of the sample source.

There are some pipeline samples in the documentation in this repo and there are also discussions in the past issues but issues regarding the gstreamer are best to be asked to gstreamer forums.

disa6302, 2022-09-19:
@DreyOlad , 

As pointed out by @MushMal , this might not be a kvs specific issue. I see these:

no such element factory "appsink"!
no such element factory "rtph264depay"!

I think your GStreamer package is not installed appropriately on the system. My suggestion would be to try out the sample pipelines based on your platform:
1. Linux: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/docs/linux.md
2. MacOS: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/docs/macos.md
3. Windows: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/docs/windows.md

Note: if MacOS is the platform and the OS is Monterey, you might have to add the path to the gstreamer package in the GST_PLUGIN_PATH along with the kvssink path. Something like this:
export GST_PLUGIN_PATH=<path>/gstreamer-1.0

closed 2022-09-19 by disa6302