awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #908: Unable to stream in AWS kinesis video from Ubuntu install on Virtual Box having Camera

EHVKumar, 2022-09-17:
**Using cmd:**
gst-launch-1.0 v4l2src do-timestamp=TRUE device=/dev/video0 ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=30/1 ! x264enc  bframes=0 key-int-max=45 bitrate=500 ! video/x-h264,stream-format=avc,alignment=au,profile=baseline ! kvssink stream-name="TestVideoStream2" storage-size=512 access-key="" secret-key="" aws-region="us-east-1"

**Error**:
[DEBUG] [17-09-2022 12:11:02:262.457 GMT] streamReadyHandler invoked
Stream is ready
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Internal data stream error.
Additional debug info:
../libs/gst/base/gstbasesrc.c(3127): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
streaming stopped, reason not-negotiated (-4)
Execution ended after 0:00:00.000589469
Setting pipeline to NULL ...
[INFO ] [17-09-2022 12:11:02:263.764 GMT] stopKinesisVideoStreamSync(): Synchronously stopping Kinesis Video Stream 000055db47763600.
[DEBUG] [17-09-2022 12:11:02:263.852 GMT] streamClosedHandler invoked for upload handle: 18446744073709551615
[DEBUG] [17-09-2022 12:11:02:263.883 GMT] Reported streamClosed callback for stream handle 94400285128192. Upload handle 18446744073709551615
[INFO ] [17-09-2022 12:11:02:263.897 GMT] Sending eos
Freeing pipeline ...

[image]
[image]

niyatim23, 2022-09-18:
Hi @EHVKumar, looks like an issue with the system's camera. Can you please ensure that GStreamer has access to the device's camera before running the pipeline?

EHVKumar, 2022-09-19:
I ran gst-device-monitor-1.0 in terminal and got this output.
[device_details.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/9596869/device_details.txt)

disa6302, 2022-09-19:
@EHVKumar , 

This does not seem to be a KVS specific issue. I would suggest to reach out to GStreamer forums to understand how you need to fix your source pipeline. A few pointers:

1. Start with fakesink. Ensure your source pipeline is connected appropriately.
2. Try other sinks - autovideosink for example. 
3. See if one of the sample pipelines here works as well: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/docs/linux.md. See if removing do-timestamps helps. 

I am closing this issue, but do feel free to reopen if after investigation on your end, it does not work with kvssink.

closed 2022-09-19 by disa6302