awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #911: [QUESTION] Offline upload of file, throttle bandwidth

clogwog, 2022-09-26:
Is it possible to upload a file with kvssink streaming-type=offline but limit how fast it tries to upload it ?

I'm uploading a 1 minute file using 

`gst-launch-1.0 filesrc location=/data/myvideo.mp4 ! qtdemux name=demux demux.video_0 !  h264parse config-interval=-1  ! video/x-h264,stream-format=avc,alignment=au ! kvs.video_0  kvssink name=kvs streaming-type=offline file-start-time=xxxxxx storage-size=512 iot-certificate="......" iot-thing-name="mydevice" stream-name="mystream"`

and it looks like it is eating so much of my bandwidth that nothing else is happening while the upload finishes (after about 25 seconds) 

[avg-bandwidth-bps](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/src/gstreamer/gstkvssink.cpp#L508) looks like it can be set, but i can't find the documentation for it.

MushMal, 2022-09-26:
The library doesn't have any provisions currently. You can modify https://github.com/awslabs/amazon-kinesis-video-streams-producer-c/blob/a8091d1a140fbdcb260ba3c8302aa2e4712d2d91/src/source/Response.c#L147 to add CURLOPT_MAX_SEND_SPEED_LARGE to the options 

https://curl.se/libcurl/c/[CURLOPT_MAX_SEND_SPEED_LARGE](https://curl.se/libcurl/c/CURLOPT_MAX_SEND_SPEED_LARGE.html).html

clogwog, 2022-09-26:
@MushMal thank you, that is very interesting, i'm trying to find a path from kvssink all the way into the depths of the videostreams producer library so it can be passed in. lost my way around createStreamCurl

MushMal, 2022-09-27:
The higher-level logic creates an API callback provider based on CURL with a call to createCurlApiCallbacks. This sets the actual callbacks in https://github.com/awslabs/amazon-kinesis-video-streams-producer-c/blob/a8091d1a140fbdcb260ba3c8302aa2e4712d2d91/src/source/CurlApiCallbacks.c#L58-L65 which would then be called by the state automata.

The one for streaming is called putStreamCurl which is defined here: https://github.com/awslabs/amazon-kinesis-video-streams-producer-c/blob/a8091d1a140fbdcb260ba3c8302aa2e4712d2d91/src/source/CurlApiCallbacks.c#L1993

It spins up a handler thread that services the streaming itself in https://github.com/awslabs/amazon-kinesis-video-streams-producer-c/blob/a8091d1a140fbdcb260ba3c8302aa2e4712d2d91/src/source/CurlApiCallbacks.c#L2105

A proper solution might be to add plumbing for rate limitation from KVS sink all the way to StreamInfo.StreamCaps, rev the structure for back compat and handle it properly in the CURL callback provider.

For your case though, to avoid all the churn, I would simply recommend calling curl_easy_setopt with CURLOPT_MAX_SENDSPEED_LARGE in the putStreamCurl somewhere here: https://github.com/awslabs/amazon-kinesis-video-streams-producer-c/blob/a8091d1a140fbdcb260ba3c8302aa2e4712d2d91/src/source/CurlApiCallbacks.c#L2044

clogwog, 2022-09-27:
@MushMal thank you for that. I guess the complexity is the exact reason why this hasn't been done already. Even if i implement the MAX_SENDSPEED_LARGE i would still need to know the total bandwidth and take a percentage of that, set it during the upload and hope the bandwidth doesn't change (we are on 4G so we can't assume this).

I may have a look at traffic shaping instead. 
Thank you for your solution though, it may be a good fallback.

closed 2022-09-27 by clogwog

MushMal, 2022-09-27:
@clogwog I would encourage cutting an issue with a feature to get this implemented. Of course, I would recommend thinking about what other parameters would be interesting to be passed in. Honestly, while this is a little bit of a churn but it's rather a straightforward task to add and I can also help.