awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #912: [QUESTION] how can i identify traffic that goes to kinesis video

clogwog, 2022-09-28:
I'm trying to do some traffic shaping so an offline upload of a file doesn't flood my network connection.
But to create these tc rules i need to know either the destination ip address and/or the ports that are used for uploading.

Is there a default port that gets used ? I noticed the kinesis video end-point is a dns, and apparently it can resolve to multiple ip addresses.. so i'm just wondering what would be the best way to identify kvssink -> kinesis video traffic ..

clogwog, 2022-09-28:
i did a network capture and it looks like it is a tcp connection to port **443** ? is that right ?

MushMal, 2022-09-28:
KVS is using HTTP (1.1) with TLS to communicate with AWS and to upload the stream so the ports would be 443. Another issue is the actual IP ports which for streaming can and will be a range and not a preset IP or DNS entry. Before KVS can stream, it should get the endpoint to stream to and that can vary. Perhaps AWS folks can help with the IP ranges I am not sure.

clogwog, 2022-09-28:
thank you, i've started with re:post, but may escalate it to the support centre if needed.

closed 2022-09-28 by clogwog