awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #916: [BUG] Bus error (core dumped)

asenousy, 2022-10-17:
**Logging**
```
2022-10-14 08:41:11 [3988780144] Stream is ready
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Redistribute latency...
INFO - kinesisVideoStreamFormatChanged(): Stream format changed.
2022-10-14 08:41:25 [3988780144] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000040, Next state: 0x0000000000000080
2022-10-14 08:41:25 [3988780144] INFO - putStreamResultEvent(): Put stream result event. New upload handle 0
2022-10-14 08:41:25 [3988780144] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000080, Next state: 0x0000000000000100
2022-10-14 08:41:25 [3988780144] DEBUG - Kinesis Video client and stream metrics
	>> Overall storage byte size: 134217728
	>> Available storage byte size: 134209120
	>> Allocated storage byte size: 8608
	>> Total view allocation byte size: 144080
	>> Total streams frame rate (fps): 25
	>> Total streams transfer rate (bps): 33554432 (32768 Kbps)
	>> Current view duration (ms): 0
	>> Overall view duration (ms): 0
	>> Current view byte size: 8568
	>> Overall view byte size: 8568
	>> Current frame rate (fps): 25
	>> Current transfer rate (bps): 33554432 (32768 Kbps)

Bus error (core dumped)
```

**Describe the bug**
Running the Gstreamer docker image to stream from USB Camera on a raspberry pi 4 (64-bit OS) as instructed [here](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/examples-gstreamer-plugin.html), It fails and throws this error "Bus error (core dumped)"

FYI i don't get this issue on a 32-bit OS

**To Reproduce**
Follow steps  [here](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/examples-gstreamer-plugin.html) to run docker image on raspberry pi 4 on any 64bit OS

**Platform (please complete the following information):**
 - OS: Raspberry Pi OS Lite (64-bit)
 - Processor: ARM v8
 - Version Debian Bullseye 2022-09-22

or

 - OS: Ubuntu Server 20.04.5 LTS (64-bit)
 - Processor: ARM v8
 - Version 20.04.5 LTS (64-bit)

disa6302, 2022-10-24:
@asenousy , 

Hmm. Not sure what is happening. Could you add a stack trace that will help the team understand the issue?

disa6302, 2022-11-20:
Closing due to inactivity

closed 2022-11-20 by disa6302