awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #921: [BUG] ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Internal data stream error.

Karthiktmv, 2022-10-26:
Logging
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Internal data stream error.
Additional debug info:
../libs/gst/base/gstbasesrc.c(3127): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
streaming stopped, reason not-negotiated (-4)

Describe the bug
I have Lenovo Legion Y540 laptop & I am trying to set up AWS Kinesis Video stream with gstreamer in my oracle virtual box with ubuntu 22.04 OS.
I have followed all the steps and the whole setup is complete without any problem.

I want to stream video from my laptop's integrated camera to AWS KVS via this cpp sdk.
I can use my camera properly inside my ubuntu in virtualbox with "cheese ."

I used this command from https://github.com/awsdocs/AWS-Kinesis-Video-Documentation/blob/master/doc_source/examples-gstreamer-plugin.md:-

Example 2: Encode and Stream Video from a USB Camera on Ubuntu:-

$ gst-launch-1.0 v4l2src do-timestamp=TRUE device=/dev/video0 ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=30/1 ! x264enc bframes=0 key-int-max=45 bitrate=500 ! video/x-h264,stream-format=avc,alignment=au,profile=baseline ! kvssink stream-name="YourStreamName" storage-size=512 access-key="YourAccessKey" secret-key="YourSecretKey" aws-region="YourAWSRegion"

After using v4l2-ctl --list-devices, my devices also show in output, but i am getting this error:-

[image]
[image]

Here is an output of gst-device-monitor-1.0:-

[image]

Here is the output of "gst-inspect-1.0 kvssink" if that will help:-

[image]

I am new to this whole KVS and I am not that technically that aware about how the cpp producer sdk works, I just want to get it up and working so that I know it's working on my system. Can you kindly help me with the code as well to start the streaming succesfully.

Please help me in resolving this issue.

OS:- Windows 11 with Oracle VirtualBox 6.1.40 with Ubuntu 22.04

Note:- Also any contributors I would like to request you all to not close this issue after giving a solution, please discuss it with me and let me try to solve the issue with your suggestion/code correction. Once the issue is solved I would appreciate closing the issue then.

jdelapla, 2022-10-31:
If you remove kvssink from the pipeline and instead use a filesink does it work?
`st-launch-1.0 v4l2src do-timestamp=TRUE device=/dev/video0 ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=30/1 ! x264enc bframes=0 key-int-max=45 bitrate=500 ! video/x-h264,stream-format=avc,alignment=au,profile=baseline ! filesink location=test.h264`

This is likely unrelated to anything KVS, and more an issue with using VirtualBox and a gstreamer element that is trying to access a peripheral. 

If the filesink works please let us know and we can do a deeper dive into why KVSSINK would be affected.

niyatim23, 2022-11-14:
Closing due to no response. Feel free to reopen if there are any further questions

closed 2022-11-14 by niyatim23