awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #925: Injecting my own timestamp and getting it from the consumer [QUESTION]

S0UL4, 2022-12-02:
Hey Folks!

My use case is the following ( sending frames from my PC to Cloud via KVS ) and getting the Frames in real time on an EC2 instance with a consumer.
Can i inject my PC timestamp at which i sent the frame and extracting that specific timestamp by the consumer ? 
PS: the producer used is the C++ one and the Consumer is the Python Lib.

I would be very grateful for your help.
Thank you

MushMal, 2022-12-02:
@S0UL4 using the producer SDK you could generate your own timestamps in the Frame structure. Not sure from the description of your scenario whether you need to use the presentation timestsamp or the decoding timestamp but in likely case, it's presentation timestamp then it's defined here: https://github.com/awslabs/amazon-kinesis-video-streams-pic/blob/afc15aff555090424d6b92ef316116e85004a479/src/mkvgen/include/com/amazonaws/kinesis/video/mkvgen/Include.h#L304

You need to ensure that those timestamps are monotonically increasing. In the C producer SDK the sample uses the synthetic frames with generated timestamps. Most unit tests do the same too.

S0UL4, 2022-12-02:
@MushMal 
Thank u for your reply mate!
Indeed, i tried to add my own timestamps in the frame structure or modify the presentation timestsamp or the decoding timestamp.. But the problem that i couldn't find them in the received payload by the consumer to extract them..!

MushMal, 2022-12-02:
I assume you are using GetMedia API to consume the stream correct? The stream is packaged in MKV format and the cluster timestamp is generated from the I-frames. Make sure you set the absolute fragment timestamp to TRUE in the stream caps https://github.com/awslabs/amazon-kinesis-video-streams-pic/blob/master/src/client/include/com/amazonaws/kinesis/video/client/Include.h#L1000 which I believe is the default. 

Frames themselves don't really have timestamps. It's a higher-level packaging that does. In this case, MKV. For more information about the timestamps refer to: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/docs/timestamps.md

S0UL4, 2022-12-02:
Ah i see! the idea that i'm trying to do is to send frame by frame to the EC2 instance with timestamp so i can then measure the Network latency that that frame took to be parsed by the Python Consumer ( GetMedia API ).. I did changed the producer to send 1 frame by fragment and i wanted to inject my timestamp with the metadata of the frame to then calculate the Network Latency from my python consumer after receiving the frame.. Am i in the right Path??
Thank u very much for your reply and time @MushMal

MushMal, 2022-12-02:
Yeah, that should work just fine. In fact, there should be a ready sample doing exactly that: https://github.com/aws-samples/amazon-kinesis-video-streams-demos/tree/master/canary

@disa6302 might be able to help with the canary but the idea is exactly that - measuring the latency by embedding the timestamp in the frame and extracting/comparing it on the consumer side.

S0UL4, 2022-12-02:
Thank you @MushMal  i really appreciate your help.
Do u think is there any documentation that could be useful to run the Canary Sample?
Any idea @disa6302  please?
Thanks in advance guys!

hassanctech, 2022-12-05:
This is provided as a sample so there is no further documentation than what is already present in the readme here: https://github.com/aws-samples/amazon-kinesis-video-streams-demos/tree/master/canary/producer-c

I believe the original question regarding how to test with your timestamp has been answered, so I will close this ticket.  If you have a new question please open a new ticket.

closed 2022-12-05 by hassanctech