awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #926: Failed to insert event metadata [QUESTION]

S0UL4, 2022-12-05:
Hey folks,
A problem occures during adding EventMedadata..
below some logs:
putKinesisVideoEventMetadata(): operation returned status code: 0x5200008c 
Failed to insert event metadata with: 1375731852

hassanctech, 2022-12-05:
This error code is defined here:

https://github.com/awslabs/amazon-kinesis-video-streams-pic/blob/dc4bc6cb87cf97cd3b1dc3e8027e0c75cdfa13b3/src/client/include/com/amazonaws/kinesis/video/client/Include.h#L189

Hello, this indicates that you're attempting to send metadata before the stream has started.  You must first start streaming before you can insert event metadata. If that does not resolve your issue please re-open the ticket.

shritib, 2023-01-12:
Hi @niyatim23, I'm also getting same error when trying to run KVS binary. I've used IAM role with Admin access. It has all permissions for all AWS services. Following is error I'm seeing:

[DEBUG] [12-01-2023 05:58:52:785.076 GMT] Stream is ready
[INFO ] [12-01-2023 05:58:52:952.503 GMT] Streaming from rtsp source
New pad found: recv_rtp_src_0_1056831275_96
[INFO ] [12-01-2023 05:58:53:571.830 GMT] Pad link successful
[INFO ] [12-01-2023 05:58:55:621.943 GMT] kinesisVideoStreamFormatChanged(): Stream format changed.
[INFO ] [12-01-2023 05:58:55:632.393 GMT] putStreamResultEvent(): Put stream result event. New upload handle 0
[INFO ] [12-01-2023 05:58:55:893.888 GMT] writeHeaderCallback(): RequestId: e249035a-3dce-d5a9-be70-a085b37476ed
[DEBUG] [12-01-2023 05:58:56:517.905 GMT] postReadCallback(): Pausing CURL read for upload handle: 0
[ERROR] [12-01-2023 05:58:57:550.856 GMT] putKinesisVideoEventMetadata(): operation returned status code: 0x5200008c
[ERROR] [12-01-2023 05:58:57:551.566 GMT] Failed to insert event metadata with: 1375731852

jdelapla, 2023-01-19:
Hi, the error your receiving is from calling `putKinesisVideoEventMetaData` before the stream has started. Start the stream, then call the API and it will function as intended.

closed 2023-01-19 by jdelapla

sirknightj, 2023-06-06:
If you were previously running `kvs_gstreamer_sample`, the issue has been addressed in #1019.