awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #927: RTT Latency

S0UL4, 2022-12-12:
Hello Developers!

I'm sending a fragment of 7 frames to  KVS and using a consumer on an EC2 instance to extract the 7th frame then send back the message to my PC to calculate the RTT ( Round Trip Time).
I found that when i try to send 1 frame/fragment, the latency goes crazy and won't be suitable for my use case.
Otherwise if i increase the number of frames/fragment i got a good latency around 100 ms ! here is a screenshot below:
[image]

My Question is : why when i try to send only 1 frame/fragment i got a huge RTT Latency ? because at the end i need to send only 1 frame..

MushMal, 2022-12-12:
It's not immediately clear what could be the cause and how the e2e is measured. One thing you could check is whether any piece of your e2e is based on ACKs - Received and Persisted ACKs are sent after the entire fragment is assembled and the backed doesn't know when the fragment ends unless it detects a new fragment start. 

You could try to send an EoFR after your frame and see if that fixes it

niyatim23, 2023-01-09:
Closing since the issue has gone stale. Feel free to re-open / create another issue