awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #928: [QUESTION]

S0UL4, 2022-12-13:
Hello guys,

when i try to stream with kvs_gstreamer_sample through RTSP url (rtsp://192.168.1.134:8554/mystream) i got this error below:

[image]

Any idea please?

S0UL4, 2022-12-13:
The problem comes from UDP Buffer i think !
[image]

Should i increase the UDP BUFFER SIZE ?

ryanfogarty, 2023-01-04:
I get the same error when streaming from file source but not live source. Increasing buffer size did not resolve for me however.

disa6302, 2023-01-10:
@S0UL4 , 

What version of gstreamer does your system have? Can you share the OS version details as well?

disa6302, 2023-01-16:
Resolving due to no response. A few pointers:

1. Test your pipeline using gst-launch-1.0 with fakesink to make sure your source piece of the pipeline works. 
2. If you have VPN enabled, disable it. I have seen RTSP linking errors occur when using VPN.

closed 2023-01-16 by disa6302