awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #929: [BUG] Parallel producers to the same stream create jumbled video

bkovacs-orca, 2022-12-15:
**Logging**
No errors in logs, so I don't think it is relevant here.

**Describe the bug**
I'd like to concurrently upload multiple on-demand clips for historical video data to the same stream. I made sure the producer timestamps are set correctly for each fragment. On the viewer side I use LIVE_REPLAY mode when calling the [GetHLSStreamingSessionURL](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/API_reader_GetHLSStreamingSessionURL.html) API, since I'd like to start viewing the video clip even if it's not fully uploaded yet. The `getHLSMediaPlaylist` call returns a playlist that has mixed timestamps between the different historical video clips which causes the player to show a stream that jumps between fragments between the different upload sessions.

**To Reproduce**
Steps to reproduce the behavior:
1. Start streaming a historical video clip by creating one Gstreamer pipeline that ends with kvssink. Use OFFLINE mode for kvssink to upload frames, making sure the producer timestamp is set correctly by setting the PTS of the Gstreamer buffers correctly.
2. Start another stream in parallel the same way as above in a separate Gstreamer pipeline.
3. Make `GetHLSStreamingSessionURL` API call with `LIVE_REPLAY` mode, start/end timestamp of the first video clip and `PRODUCER_TIMESTAMP`. Using the following parameters and using the python boto API:
```
video_client = boto_session.client("kinesisvideo")
data_endpoint: str = video_client.get_data_endpoint(
    StreamName=stream_name,
    APIName="GET_HLS_STREAMING_SESSION_URL",
)["DataEndpoint"]
archive_client = boto_session.client(
    "kinesis-video-archived-media", endpoint_url=data_endpoint
)
archive_client.get_hls_streaming_session_url(
    StreamName=stream_name,
    PlaybackMode="LIVE_REPLAY",
    HLSFragmentSelector={
        "FragmentSelectorType": "PRODUCER_TIMESTAMP",
        "TimestampRange": {
            "StartTimestamp": start_time
            "EndTimestamp": end_time,
        }
    },
    ContainerFormat="FRAGMENTED_MP4",
    DiscontinuityMode="ALWAYS",
    DisplayFragmentTimestamp="ALWAYS",
    MaxMediaPlaylistFragmentResults=600,
    Expires=43200);
```
4. Use VideoJS to play the HLS URL we got
5. Observe jumbled stream between the two different streaming sessions.

**Expected behavior**
We expect to see continuous streaming of the first video clip without seeing any frames from the second video clip upload session.

**SDK version number**
We are using the last commit on the master branch, release 3.3.1 (commit 70f74f14cf).

**Open source building**
N/A

**Screenshots**
N/A

**Platform (please complete the following information):**
 - OS: Linux, Ubuntu 
 - Processor: 11th Gen Intel(R) Core(TM) i9-11900K @ 3.50GHz
 - Version: 20.04

**Additional context**
N/A

niyatim23, 2023-01-09:
Hi @bkovacs-orca, KVS Streams are not meant for this use case. Only a single source should upload to a single stream at one point in time. You will have to use multiple streams or make sure while uploading to the same stream you use the correct timestamps. 

Closing assuming answered

bkovacs-orca, 2023-01-11:
@niyatim23 Thanks for the answer! We are uploading to the same KVS stream concurrently in multiple sessions and we are using the correct timestamps for each session. These sessions upload video clips that are not overlapping. In this case, I assume KVS Streams should work properly, so I think we should keep this bug open. What do you think?

jdelapla, 2023-01-19:
Currently KVSSink overwrites the producer timestamp to the current time, so it will not work as intended for your use case. We are currently testing a a variety of features in the **add-signals-kvssink** branch, one of which is a property **use-original-pts** which is what you would need for this use case to work. Feel free to take the necessary commits if you need this immediately, or wait until our next release sometime mid February.

closed 2023-01-19 by jdelapla

bkovacs-orca, 2023-02-01:
@jdelapla Thanks for the answer! `Currently KVSSink overwrites the producer timestamp to the current time` I think this is true in LIVE mode, but we use OFFLINE mode when using KVSSink. We make sure the producer timestamps are set correctly as the PTS of the GstBuffers. And based on reading the code of KVSSink, I saw that in OFFLINE mode it will use the PTS of the GstBuffer, unless I'm missing something.

bkovacs-orca, 2023-02-01:
> Sure, in that case, please share verbose logs from both uploads demonstrating the issue with timestamps

@niyatim23 On the uploader side, I think the timestamps are correct, but somewhere on the server side something goes wrong and because of that when we watch an HLS stream in the browser that we got from `GetHLSStreamingSessionURL` with `LIVE_REPLAY` mode, it is mixed between the two uploads.

I think the issue is on the server side because if we call `GetHLSStreamingSessionURL` with `ON_DEMAND` mode, we don't see a mixed stream. `ON_DEMAND` is not sufficient for our use case, because we might be uploading the clip while we are already watching it and `ON_DEMAND` will not return an HLS link that can provide a dynamically growing "live" playlist, so it will just give a playlist of fragments that were already uploaded from the requested clip.

Knowing all this, do you think it would still be useful to share logs from both uploads on the uploader side?