awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #930: [BUG] kvs_gstreamer_sample.cpp.o: undefined reference to symbol 'strtoi32'

thomas-roos, 2022-12-21:
mentioned here: https://github.com/aws4embeddedlinux/meta-aws/issues/1701

aarch64-poky-linux/11.3.0/ld: CMakeFiles/kvs_gstreamer_sample.dir/samples/kvs_gstreamer_sample.cpp.o: undefined reference to symbol 'strtoi32'

[error_kvs_gstreamer_sample.log](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/10277065/error_kvs_gstreamer_sample.log)

thomas-roos, 2023-01-04:
the bug is not gone - as mentioned here: https://github.com/aws4embeddedlinux/meta-aws/issues/1701#issuecomment-1361176108
I was just disabling this ONE test - the others just build fine.
Did dive deep a few hours and found the root of the problem.
Firstly, I build **shared** (as any Linux user).
When building static (pic only works) - the problem is magically gone.
Therefor I was looking at the produced libs (where this strtoi32 symbol is in)
And found that when building static this is in the libcproducer.so
When building shared, this is not - it's in a shared lib, but this lib is not linked against the kvs_gstreamer_sample.cpp
So I all what's finally necessary is editing the CMakeLists.txt to link the sample to kvspic as well (strtoi32 is in there)
-  target_link_libraries(kvs_gstreamer_sample ${GST_APP_LIBRARIES} KinesisVideoProducer)
+  target_link_libraries(kvs_gstreamer_sample ${GST_APP_LIBRARIES} KinesisVideoProducer kvspic)

sirknightj, 2023-02-13:
Hi @thomas-roos, unfortunately I was unable to replicate the issue you ran into.

I just built latest/master on an Ubuntu 22.04.1 LTS machine using the following arguments:
```
cmake .. -DBUILD_DEPENDENCIES=OFF -DBUILD_GSTREAMER_PLUGIN=ON
make
```

I was able to run the sample applications just fine, after setting GST_PLUGIN_PATH to the `build` directory:
```
./kvs_gstreamer_sample test-stream ~/test.mp4
./kvs_gstreamer_file_uploader_sample test-stream-2 ~/test.mp4 0
```

Here's the output of `nm -D` on the libcproducer:
```
$ nm -D dependency/libkvscproducer/kvscproducer-src/libcproducer.so | grep "strtoi32"
000000000006020f T strtoi32
```

@thomas-roos Could you please share the CMake command used to build the SDK?

thomas-roos, 2023-02-14:
Hi sirknightj,
as said the error occurs **ONLY** when builiding **SHARED**.
There for you have to set 
`-DBUILD_STATIC=OFF`
see other used options here: https://github.com/aws4embeddedlinux/meta-aws/blob/edfc1aa8aadbbdb095cedbec705f4b2f45ced440/recipes-sdk/amazon-kvs-producer-sdk/amazon-kvs-producer-sdk-cpp_3.3.1.bb#L53
did you build shared?

niyatim23, 2023-02-14:
Hi @thomas-roos, what do mean by "SHARED" here? We have a working linux static build [here](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/develop/.github/workflows/ci.yml#L377-L391). That's why, we do not think there is any issue with the SDKs builds. Please follow the steps here to fix your build using the latest release of the SDK and let us know if the issue persists

thomas-roos, 2023-02-14:
Hi @niyatim23
I mean SHARED linking - [shared libraries](https://stackoverflow.com/questions/2649334/difference-between-static-and-shared-libraries) as it's the Linux default.
When you build it to use shared libraries, it needs a CMake define `-DBUILD_STATIC=OFF`
I use version 3.3.1 (should be the latest) and this is the fix: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/pull/932 
And yes, still persists - may try to define `-DBUILD_STATIC=OFF` in your example and then try my fix?

niyatim23, 2023-02-14:
Hi @thomas-roos, we also have a `-DBUILD_STATIC=OFF` build [here](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/develop/.github/workflows/ci.yml#L240-L274) (it is off by default). It works fine without making any changes to CMakeLists. Can you try retracing those steps and let us know if it works for you?

thomas-roos, 2023-02-15:
Hi, 
think I found the understanding problem - you tell me building shared is default for this project - cool, I would do the same.
But here: https://github.com/awslabs/amazon-kinesis-video-streams-pic/blob/c8325887faa3a4a296c4367b281c778be69875b6/CMakeLists.txt#L22 static is default - when you build it shared you will have my problem.
I do build all libs separate and set ` -DBUILD_DEPENDENCIES=OFF ` and `-DBUILD_STATIC=OFF` (you never know what is default;)
Agree?

rpcme, 2023-04-05:
@benwkim what is the timeline for folding the fix into mainline and resolving this issue? We are waiting on the fix in order to resolve an issue in the meta-aws project.

jdelapla, 2023-04-06:
@rpcme We're currently testing a release candidate that includes this change, we hope to release before the end of the month.

thomas-roos, 2023-07-17:
Which month will this release candidate released?

sirknightj, 2023-08-17:
@thomas-roos we are aiming for before the end of this month

closed 2023-08-25 by disa6302

sirknightj, 2023-08-25:
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/releases/tag/v3.4.0

commit e609e6d2 (2023-04-11): Update CMakeLists.txt (#997)

* Update CMakeLists.txt

when building shared the kvs_gstreamer_sample.cpp example fails LINKING
fix #930

* Additional CI Build

---------