awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #931: Increasing the quality of the frames[QUESTION]

S0UL4, 2023-01-03:
Hey folks,

Is there any way to increase the quality of the frames that are being sent to the kvs ? should i change the DEFAULT_CODEC_ID variable to use H246 Encoder?

Thanks in advance!

hassanctech, 2023-01-03:
Hi.  This SDK provides a way to send video data to KVS.  You are responsible for providing the media source (whether it be from a local camera device, a set of files, does not matter).  If your video is coming from a camera then consult the camera API documentation to see how you can adjust the quality of the frames.

H264 is already supported you'll see in that [string](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/2b343c90bd9c8785d9b2b9ba321f053ebf175b5e/samples/kvs_gstreamer_audio_video_sample.cpp#L55) the last part is AVC, which is another name for H264.

S0UL4, 2023-01-04:
Ah okay Thank you, i already increased the bitrateInKBPS to insure a better quality..
Question solved!

closed 2023-01-04 by S0UL4