awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #935: could not link rtph264depay0 to videoconvert0

jorrego, 2023-01-09:
I have problems when run this command on same cameras: 

`gst-launch-1.0 rtspsrc location="rtsp://route-of-camera" short-header=TRUE ! rtph264depay ! h264parse ! video/x-h264, format=avc,alignment=au ! kvssink stream-name="NAME" storage-size=64 access-key="ACCESSKEY" secret-key="SECRETKEY" fragment-duration="10000" aws-region="us-east-1" framerate="15"`

When the terminal showed me the message error **MAX_FRAGMENT_DURATION_REACHED** I tried the solution from #147 and add his recomendation. But whe I run the new command the terminal show me **WARNING: erroneous pipeline: could not link rtph264depay0 to videoconvert0**:

`gst-launch-1.0 rtspsrc location="rtsp://route-of-camera" short-header=TRUE ! rtph264depay! videoconvert ! x264enc bframes=0 key-int-max=10 ! h264parse ! video/x-h264, format=avc,alignment=au ! kvssink stream-name="NAME" storage-size=64 access-key="ACCESSKEY" secret-key="SECRETKEY" fragment-duration="10000" aws-region="us-east-1" framerate="15"`

Have I possibly made a mistake while adding the parameters?

disa6302, 2023-01-10:
@jorrego , 

Can you modify the first command to set `key-int-max` to 10 and remove using fragment-duration?

` erroneous pipeline: could not link rtph264depay0 to videoconvert0`: This means linking rtph264depay to videoconvert is not accepted by gstreamer libraries.

jorrego, 2023-01-10:
> @jorrego ,
> 
> Can you modify the first command to set `key-int-max` to 10 and remove using fragment-duration?
> 
> ` erroneous pipeline: could not link rtph264depay0 to videoconvert0`: This means linking rtph264depay to videoconvert is not accepted by gstreamer libraries.

@disa6302 , i did it but it didn't work. It show me **WARNING: erroneous pipeline: no property "key-int-max" in element "kvssink0"**.I ran this command: 

`gst-launch-1.0 rtspsrc location="rtsp://route-of-camera" short-header=TRUE ! rtph264depay ! h264parse ! video/x-h264, format=avc,alignment=au ! kvssink stream-name="NAME" storage-size=64 access-key="ACCESSKEY" secret-key="SECRETKEY" aws-region="us-east-1" framerate="15" key-int-max=10`

disa6302, 2023-01-10:
Key-int-max is a x264enc property. It is not a kvssink property.

jorrego, 2023-01-11:
Which parameters should I use or remove? Based on what you mentioned, I modified my command adding the x264enc, looking like this:

`gst-launch-1.0 rtspsrc location="rtsp://route-of-camera" short-header=TRUE ! videoconvert ! x264enc bframes=0 key-int-max=10 ! kvssink stream-name="NAME" storage-size=64 access-key="ACCESSKEY" secret-key="SECRETKEY" aws-region="us-east-1" framerate="15"`

The console responded to me as follows:

```
Setting pipeline to PAUSED ...
log4cplus:ERROR could not open file ../kvs_log_configuration
INFO - createKinesisVideoClient(): Creating Kinesis Video Client
2023-01-11 10:32:39 [140024012063808] INFO - heapInitialize(): Initializing native heap with limit size 67108864, spill ratio 0% and flags 0x00000001
2023-01-11 10:32:39 [140024012063808] INFO - heapInitialize(): Creating AIV heap.
2023-01-11 10:32:39 [140024012063808] INFO - heapInitialize(): Heap is initialized OK
2023-01-11 10:32:39 [140024012063808] DEBUG - getSecurityTokenHandler invoked
2023-01-11 10:32:39 [140024012063808] DEBUG - Refreshing credentials. Force refreshing: 0 Now time is: 1673451159786557800 Expiration: 0
2023-01-11 10:32:39 [140024012063808] INFO - createDeviceResultEvent(): Create device result event.
2023-01-11 10:32:39 [140024012063808] DEBUG - clientReadyHandler invoked
2023-01-11 10:32:39 [140024012063808] INFO - try creating stream
2023-01-11 10:32:39 [140024012063808] INFO - Creating Kinesis Video Stream NAME
2023-01-11 10:32:39 [140024012063808] INFO - createKinesisVideoStream(): Creating Kinesis Video Stream.
2023-01-11 10:32:39 [140024012063808] INFO - logStreamInfo(): SDK version: 70f74f14cf27b09f71dc1889f36eb6e04cdd90a8
2023-01-11 10:32:39 [140024012063808] DEBUG - logStreamInfo(): Kinesis Video Stream Info
2023-01-11 10:32:39 [140024012063808] DEBUG - logStreamInfo(): Stream name: NAME 
2023-01-11 10:32:39 [140024012063808] DEBUG - logStreamInfo(): Streaming type: STREAMING_TYPE_REALTIME 
2023-01-11 10:32:39 [140024012063808] DEBUG - logStreamInfo(): Content type: video/h264 
2023-01-11 10:32:39 [140024012063808] DEBUG - logStreamInfo(): Max latency (100ns): 600000000
2023-01-11 10:32:39 [140024012063808] DEBUG - logStreamInfo(): Fragment duration (100ns): 20000000
2023-01-11 10:32:39 [140024012063808] DEBUG - logStreamInfo(): Key frame fragmentation: Yes
2023-01-11 10:32:39 [140024012063808] DEBUG - logStreamInfo(): Use frame timecode: Yes
2023-01-11 10:32:39 [140024012063808] DEBUG - logStreamInfo(): Absolute frame timecode: Yes
2023-01-11 10:32:39 [140024012063808] DEBUG - logStreamInfo(): Nal adaptation flags: 0
2023-01-11 10:32:39 [140024012063808] DEBUG - logStreamInfo(): Average bandwith (bps): 4194304
2023-01-11 10:32:39 [140024012063808] DEBUG - logStreamInfo(): Framerate: 15
2023-01-11 10:32:39 [140024012063808] DEBUG - logStreamInfo(): Buffer duration (100ns): 1200000000
2023-01-11 10:32:39 [140024012063808] DEBUG - logStreamInfo(): Replay duration (100ns): 400000000
2023-01-11 10:32:39 [140024012063808] DEBUG - logStreamInfo(): Connection Staleness duration (100ns): 600000000
2023-01-11 10:32:39 [140024012063808] DEBUG - logStreamInfo(): Store Pressure Policy: 1
2023-01-11 10:32:39 [140024012063808] DEBUG - logStreamInfo(): View Overflow Policy: 1
2023-01-11 10:32:39 [140024012063808] DEBUG - logStreamInfo(): Segment UUID: NULL
2023-01-11 10:32:39 [140024012063808] DEBUG - logStreamInfo(): Frame ordering mode: 0
2023-01-11 10:32:39 [140024012063808] DEBUG - logStreamInfo(): Track list
2023-01-11 10:32:39 [140024012063808] DEBUG - logStreamInfo(): Track id: 1
2023-01-11 10:32:39 [140024012063808] DEBUG - logStreamInfo(): Track name: kinesis_video
2023-01-11 10:32:39 [140024012063808] DEBUG - logStreamInfo(): Codec id: V_MPEG4/ISO/AVC
2023-01-11 10:32:39 [140024012063808] DEBUG - logStreamInfo(): Track type: TRACK_INFO_TYPE_VIDEO
2023-01-11 10:32:39 [140024012063808] DEBUG - logStreamInfo(): Track cpd: NULL
2023-01-11 10:32:39 [140024012063808] INFO - writeHeaderCallback(): RequestId: cfd450b0-0ac2-41fd-954d-ed984ff5aa66
2023-01-11 10:32:40 [140023917553408] DEBUG - describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.66632840935E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","IngestionConfiguration":null,"KmsKeyId":"arn:aws:kms:us-east-1:000000000000:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-east-1:000000000000:stream/NAME/1666328409350","StreamName":"NAME","Version":"b4nyaGcv8imiUxOK7ySh"}}
2023-01-11 10:32:40 [140023917553408] INFO - describeStreamResultEvent(): Describe stream result event.
2023-01-11 10:32:40 [140023917553408] INFO - writeHeaderCallback(): RequestId: 8a3d7383-a7c0-4c8e-908c-290b7fb6aa06
2023-01-11 10:32:40 [140023909160704] DEBUG - getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-d8eb3b21.kinesisvideo.us-east-1.amazonaws.com"}
2023-01-11 10:32:40 [140023909160704] INFO - getStreamingEndpointResultEvent(): Get streaming endpoint result event.
2023-01-11 10:32:40 [140023909160704] DEBUG - getStreamingTokenHandler invoked
2023-01-11 10:32:40 [140023909160704] DEBUG - Refreshing credentials. Force refreshing: 1 Now time is: 1673451160715493400 Expiration: 18446744073709551615
2023-01-11 10:32:40 [140023909160704] INFO - getStreamingTokenResultEvent(): Get streaming token result event.
2023-01-11 10:32:40 [140023909160704] DEBUG - streamReadyHandler invoked
[... cut ...]
2023-01-11 10:32:48 [140024012063808] DEBUG - curlApiCallbacksShutdownActiveRequests(): pActiveRequests hashtable is empty
2023-01-11 10:32:48 [140024012063808] DEBUG - freeKinesisVideoClientInternal(): Total allocated memory 0
2023-01-11 10:32:49 [140024012063808] WARN - curlApiCallbacksShutdown(): curlApiCallbacksShutdown called when already in progress of shutting down
2023-01-11 10:32:49 [140024012063808] # 
```

The indicated error is: **ERROR: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0: Internal data stream error.**

disa6302, 2023-01-11:
@jorrego , 

Please follow our sample pipelines which are available here:
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/tree/v3.3.1/docs

Not sure which platform you are using, but we instructions in both linux and MacOS. 

Also please attach complete verbose logs when you try the sample pipeline. You can get the VERBOSE logs by modifying this line to set to TRACE.
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/70f74f14cf27b09f71dc1889f36eb6e04cdd90a8/kvs_log_configuration#L1

jorrego, 2023-01-11:
The compilation is in TRACE and run on Ubuntu container with an Intelbras brand camera using a public IP.

disa6302, 2023-01-16:
Start with the pipeline samples. As such we wouldnt know the specifics of the IP cameras being used and it is likely there are some additional attributes that might be required. 

One thing you can do to understand if your pipeline is correct is to generate a pipeline using fakesink instead of KVS sink. Ensure you get the source of the pipeline correct before trying with kvssink. Based on the logs you attached, I do not see a MAX_FRAGMENT_DURATION error. It looks like there is a problem reading from the RTSP source itself.

jorrego, 2023-01-18:
I will test with the samples that you indicate. I have seen that the issue #183 mentions an error with the same brand of camera (Intelbras) with kvssink.

jdelapla, 2023-01-19:
It's not a camera compatibility issue, it's a gstreamer pipeline issue. If the pipeline is failing to construct then kvssink hasn't even been deployed in the application yet.
Additionally the issue with #183 was they were missing h264parse in their pipeline.
In your earlier pipeline you were including h264parse before kvssink, please add it back and try again.

jorrego, 2023-01-23:
I have run the connect to RTSP camera command on two cameras of the same make and model, one that is the one I have the problem with and the other one where the stream is successful.

Command:
`gst-launch-1.0 rtspsrc location="rtsp://CAMERA" short-header=TRUE ! rtph264depay ! h264parse ! video/x-h264, format=avc,alignment=au ! kvssink stream-name="NAME" storage-size=64 access-key="AKEY" secret-key="SKEY" fragment-duration="10000" aws-region="us-east-1" framerate="15"
`

The results of each camera are in each file that I attach.

[camera CONNECTION ERROR.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/10483183/camera.CONNECTION.ERROR.txt)
[camera CONNECTION OK.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/10483184/camera.CONNECTION.OK.txt)

Using VLC player I can see the RTSP stream from both cameras without any problems.

hassanctech, 2023-01-26:
Thank you for sharing that @jorrego, there appear to be two issues:

1.  I see lots of these in your log:
```
2023-01-23 19:42:14 [140670672467712] DEBUG - Dropping frame with flag %u1024
```

2. I see this log:
```
2023-01-23 19:42:18 [140670672467712] DEBUG - postWriteCallback(): Curl post body write function for stream with handle: jojo and upload handle: 0 returned: {"EventType":"ERROR","FragmentTimecode":1674502918659,"FragmentNumber":"91343852333181442296202376934447363596220606745","ErrorCode":"MAX_FRAGMENT_DURATION_REACHED","ErrorId":4002}
```

For (1) it looks like you have frames which the source has marked as HEADER, see here for details: https://gstreamer.freedesktop.org/documentation/gstreamer/gstbuffer.html?gi-language=c#GstBufferFlags

But in short we have logic in our `kvssink` that *if* it's a header flag *and* the timestamps (PTS/DTS) are INVALID then we drop the frame. 

```
(GST_BUFFER_FLAG_IS_SET(buf, GST_BUFFER_FLAG_HEADER) && (!GST_BUFFER_PTS_IS_VALID(buf) || !GST_BUFFER_DTS_IS_VALID(buf)));
```

In any case dropping this could be benign but it could also be a problem if this information is necessary to decode. In which case the issue is with the timestamps on your source.  I would recommend you first try to fix the issue with (2) and see if this is actually a problem.  If it is then please report back and we can debug what is wrong with your timestamps.

For (2), so we enforce a maximum fragment size for 10s (time between key frames), meaning with the default option where we create a new fragment at key frame, your rtsp source is not adhering to that requirement.  Can you check the camera vendor and see if there is a way you can force a keyframe for example every 2s?  Just any number less than 10s should work OK then you will not see this issue. 

For example on Axis cameras they allow you to pass in a parameter like `videokeyframeinterval` in the rtsp url to set the key frame interval, though by default it is fixed and would not cause this issue.  You have to find out from your source how to adjust this.  https://www.axis.com/vapix-library/subjects/t10175981/section/t10051110/display?section=t10051110-t10003638

hassanctech, 2023-01-30:
@jorrego Any updates here?  Were you able to try my suggestions?

jorrego, 2023-02-01:
We have another problem with another brand of camera. The error that shows us is the following:

`gst-launch-1.0 rtspsrc location="rtsp://camera" short-header=TRUE ! rtph264depay ! h264parse ! video/x-h264, format=avc,alignment=au ! kvssink stream-name="jojo" storage-size=64 access-key="ACCESSKEY" secret-key="SECRETKEY" fragment-duration="10000" aws-region="us-east-1" framerate="15"
Setting pipeline to PAUSED ...
log4cplus:ERROR could not open file ./kvs_log_configuration
INFO - createKinesisVideoClient(): Creating Kinesis Video Client
2023-02-01 21:49:24 [139648666998592] INFO - heapInitialize(): Initializing native heap with limit size 67108864, spill ratio 0% and flags 0x00000001
2023-02-01 21:49:24 [139648666998592] INFO - heapInitialize(): Creating AIV heap.
2023-02-01 21:49:24 [139648666998592] INFO - heapInitialize(): Heap is initialized OK
2023-02-01 21:49:24 [139648666998592] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
2023-02-01 21:49:24 [139648666998592] DEBUG - getSecurityTokenHandler invoked
2023-02-01 21:49:24 [139648666998592] DEBUG - Refreshing credentials. Force refreshing: 0 Now time is: 1675288164744378718 Expiration: 0
2023-02-01 21:49:24 [139648666998592] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
2023-02-01 21:49:24 [139648666998592] INFO - createDeviceResultEvent(): Create device result event.
2023-02-01 21:49:24 [139648666998592] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
2023-02-01 21:49:24 [139648666998592] DEBUG - clientReadyHandler invoked
2023-02-01 21:49:24 [139648666998592] INFO - try creating stream
2023-02-01 21:49:24 [139648666998592] INFO - Creating Kinesis Video Stream jojo
2023-02-01 21:49:24 [139648666998592] INFO - createKinesisVideoStream(): Creating Kinesis Video Stream.
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo(): Kinesis Video Stream Info
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Stream name: jojo
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Streaming type: STREAMING_TYPE_REALTIME
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Content type: video/h264
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Max latency (100ns): 600000000
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Fragment duration (100ns): 100000000
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Key frame fragmentation: Yes
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Use frame timecode: Yes
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Absolute frame timecode: Yes
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Nal adaptation flags: 0
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Average bandwith (bps): 4194304
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Framerate: 15
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Buffer duration (100ns): 1200000000
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Replay duration (100ns): 400000000
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Connection Staleness duration (100ns): 600000000
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Store Pressure Policy: 1
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  View Overflow Policy: 1
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Segment UUID: NULL
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Frame ordering mode: 0
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo(): Track list
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Track id: 1
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Track name: kinesis_video
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Codec id: V_MPEG4/ISO/AVC
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Track type: TRACK_INFO_TYPE_VIDEO
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Track cpd: NULL
2023-02-01 21:49:24 [139648666998592] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
2023-02-01 21:49:24 [139648666998592] INFO - writeHeaderCallback(): RequestId: c7142f5a-dcdd-49d3-8000-3e616329f620
2023-02-01 21:49:24 [139648486536960] DEBUG - describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.674165442087E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","IngestionConfiguration":null,"KmsKeyId":"arn:aws:kms:us-east-1:750035152062:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-east-1:750035152062:stream/jojo/1674165442087","StreamName":"jojo","Version":"lWs45pLZtgnGV59WWcJ9"}}
2023-02-01 21:49:24 [139648486536960] INFO - describeStreamResultEvent(): Describe stream result event.
2023-02-01 21:49:24 [139648486536960] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000020
2023-02-01 21:49:24 [139648486536960] INFO - writeHeaderCallback(): RequestId: d08a7d62-c56b-4b41-81fc-7063d69abda1
[... cut ...]
2023-02-01 21:51:22 [139648191325952] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000080, Next state: 0x0000000000000100
2023-02-01 21:51:22 [139648191325952] DEBUG - postReadCallback(): Pausing CURL read for upload handle: 4
2023-02-01 21:51:22 [139648212305664] INFO - writeHeaderCallback(): RequestId: e72c4c39-a542-5246-bb78-ad2335791dcc
2023-02-01 21:51:22 [139648212305664] DEBUG - Dropping frame with flag %u1024
2023-02-01 21:51:23 [139648191325952] DEBUG - Dropping frame with flag %u1024
2023-02-01 21:51:28 [139648191325952] DEBUG - Dropping frame with flag %u1024`

jorrego, 2023-02-01:
We have another problem with another brand of camera. The error that shows us is the following:
```
gst-launch-1.0 rtspsrc location="rtsp://camera" short-header=TRUE ! rtph264depay ! h264parse ! video/x-h264, format=avc,alignment=au ! kvssink stream-name="jojo" storage-size=64 access-key="ACCESSKEY" secret-key="SECRETKEY" fragment-duration="10000" aws-region="us-east-1" framerate="15"
Setting pipeline to PAUSED ...
log4cplus:ERROR could not open file ./kvs_log_configuration
INFO - createKinesisVideoClient(): Creating Kinesis Video Client
2023-02-01 21:49:24 [139648666998592] INFO - heapInitialize(): Initializing native heap with limit size 67108864, spill ratio 0% and flags 0x00000001
2023-02-01 21:49:24 [139648666998592] INFO - heapInitialize(): Creating AIV heap.
2023-02-01 21:49:24 [139648666998592] INFO - heapInitialize(): Heap is initialized OK
2023-02-01 21:49:24 [139648666998592] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
2023-02-01 21:49:24 [139648666998592] DEBUG - getSecurityTokenHandler invoked
2023-02-01 21:49:24 [139648666998592] DEBUG - Refreshing credentials. Force refreshing: 0 Now time is: 1675288164744378718 Expiration: 0
2023-02-01 21:49:24 [139648666998592] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
2023-02-01 21:49:24 [139648666998592] INFO - createDeviceResultEvent(): Create device result event.
2023-02-01 21:49:24 [139648666998592] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
2023-02-01 21:49:24 [139648666998592] DEBUG - clientReadyHandler invoked
2023-02-01 21:49:24 [139648666998592] INFO - try creating stream
2023-02-01 21:49:24 [139648666998592] INFO - Creating Kinesis Video Stream jojo
2023-02-01 21:49:24 [139648666998592] INFO - createKinesisVideoStream(): Creating Kinesis Video Stream.
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo(): Kinesis Video Stream Info
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Stream name: jojo
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Streaming type: STREAMING_TYPE_REALTIME
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Content type: video/h264
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Max latency (100ns): 600000000
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Fragment duration (100ns): 100000000
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Key frame fragmentation: Yes
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Use frame timecode: Yes
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Absolute frame timecode: Yes
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Nal adaptation flags: 0
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Average bandwith (bps): 4194304
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Framerate: 15
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Buffer duration (100ns): 1200000000
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Replay duration (100ns): 400000000
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Connection Staleness duration (100ns): 600000000
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Store Pressure Policy: 1
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  View Overflow Policy: 1
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Segment UUID: NULL
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Frame ordering mode: 0
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo(): Track list
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Track id: 1
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Track name: kinesis_video
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Codec id: V_MPEG4/ISO/AVC
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Track type: TRACK_INFO_TYPE_VIDEO
2023-02-01 21:49:24 [139648666998592] DEBUG - logStreamInfo():  Track cpd: NULL
2023-02-01 21:49:24 [139648666998592] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
2023-02-01 21:49:24 [139648666998592] INFO - writeHeaderCallback(): RequestId: c7142f5a-dcdd-49d3-8000-3e616329f620
2023-02-01 21:49:24 [139648486536960] DEBUG - describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.674165442087E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","IngestionConfiguration":null,"KmsKeyId":"arn:aws:kms:us-east-1:750035152062:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-east-1:750035152062:stream/jojo/1674165442087","StreamName":"jojo","Version":"lWs45pLZtgnGV59WWcJ9"}}
2023-02-01 21:49:24 [139648486536960] INFO - describeStreamResultEvent(): Describe stream result event.
2023-02-01 21:49:24 [139648486536960] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000020
2023-02-01 21:49:24 [139648486536960] INFO - writeHeaderCallback(): RequestId: d08a7d62-c56b-4b41-81fc-7063d69abda1
[... cut ...]
2023-02-01 21:51:22 [139648191325952] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000080, Next state: 0x0000000000000100
2023-02-01 21:51:22 [139648191325952] DEBUG - postReadCallback(): Pausing CURL read for upload handle: 4
2023-02-01 21:51:22 [139648212305664] INFO - writeHeaderCallback(): RequestId: e72c4c39-a542-5246-bb78-ad2335791dcc
2023-02-01 21:51:22 [139648212305664] DEBUG - Dropping frame with flag %u1024
2023-02-01 21:51:23 [139648191325952] DEBUG - Dropping frame with flag %u1024
2023-02-01 21:51:28 [139648191325952] DEBUG - Dropping frame with flag %u1024
```

jorrego, 2023-02-01:
> @jorrego Any updates here? Were you able to try my suggestions?

We cannot find where to manipulate the keyframe in these cameras, so we are looking at resetting them since not all of the Intelbras brand has this problem. As soon as we can prove it, I'll let you know what happened.

sirknightj, 2023-02-07:
Hey @jorrego your latest logs [here](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/935#issuecomment-1412792408) have the same items pointed out by @hassanctech in [this](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/935#issuecomment-1405235754) earlier comment.

Keyframe interval is the same thing as I-Frame interval. There should be an option to change it. Could you try @hassanctech's suggestion to try lowering the interval? Since you appear to be using a [framerate of 15](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/935#issuecomment-1400899351), a keyframe interval of 30 would fit the 2-second interval suggestion.

disa6302, 2023-02-20:
Closing due to no response. @jorrego , if you have more details or have any more questions, feel free to reopen or create a new issue.

closed 2023-02-20 by disa6302