awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #936: FILE UPLOAD: Pad link failed Error received from element demux: Internal data stream error.

ryanfogarty, 2023-01-14:
Trying to upload a .mp4 or .mkv file to kinesis and get this error every time `Pad link failed
Error received from element demux: Internal data stream error.`

** Logging **
The error occurs in the `pad_added_cb` callback method
at `link_ret = gst_pad_link(pad, target_sink);`
Upon some investigation,  pad and target_sink have different capabilities but I can't figure out how to get them to be in sync.
Can anyone else successfully upload a file? Live streaming from my RasPi camera works fine.

disa6302, 2023-01-16:
@ryanfogarty , 

Which sample are you using?

jdelapla, 2023-01-19:
Hey @ryanfogarty , I assume you're using kvssink, can you share the gstreamer pipeline you're using? Or what sample you're using?

ryanfogarty, 2023-01-26:
using kvs_gstreamer_sample.cpp sample

sirknightj, 2023-02-08:
Hi @ryanfogarty,

I was unable to reproduce. I ran the following commands, and both were able to upload the video correctly into kvs.
```
./kvs_gstreamer_sample test-stream ~/test.mp4
./kvs_gstreamer_file_uploader_sample test-stream-2 ~/test.mp4 0
```

Could you verify that your mkv/mp4 files are [encoded in h264](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/docs/raspberry-pi.md#running-the-gstreamer-sample-application-to-upload-a-video-file)?

Also, could you share which OS/architecture you're using, the command used, and also share some logs with GST_DEBUG=3 or 5?

niyatim23, 2023-02-14:
Closing due to no response. Feel free to reopen / open a new issue if the problem persists

closed 2023-02-14 by niyatim23