awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #937: [QUESTION]

DunkinDonat, 2023-01-23:
Gstreamer - kvssink plugin - example fails:  DEBUG - postReadCallback(): Wrote 8217 bytes to Kinesis Video. Upload stream handle: 0
 
I'm trying to stream a videotestsrc to kinesis video streams like this:
`gst-launch-1.0 videotestsrc ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=30/1 ! x264enc ! h264parse! video/x-h264,stream-format=avc,alignment=au,width=640,height=480,framerate=30/1,profile=baseline ! kvssink stream-name="" access-key="" secret-key=""
`
I get this output:

`
Setting pipeline to PAUSED ...
log4cplus:ERROR could not open file ../kvs_log_configuration
INFO - createKinesisVideoClient(): Creating Kinesis Video Client
2023-01-23 16:21:17 [1995788032] INFO - heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
2023-01-23 16:21:17 [1995788032] INFO - heapInitialize(): Creating AIV heap.
2023-01-23 16:21:17 [1995788032] INFO - heapInitialize(): Heap is initialized OK
2023-01-23 16:21:17 [1995788032] DEBUG - getSecurityTokenHandler invoked
2023-01-23 16:21:17 [1995788032] DEBUG - Refreshing credentials. Force refreshing: 0 Now time is: 1674487277767060917 Expiration: 0
2023-01-23 16:21:17 [1995788032] INFO - createDeviceResultEvent(): Create device result event.
2023-01-23 16:21:17 [1995788032] DEBUG - clientReadyHandler invoked
2023-01-23 16:21:17 [1995788032] INFO - try creating stream
2023-01-23 16:21:17 [1995788032] INFO - Creating Kinesis Video Stream video_stream_name_doni
2023-01-23 16:21:17 [1995788032] INFO - createKinesisVideoStream(): Creating Kinesis Video Stream.
2023-01-23 16:21:17 [1995788032] INFO - logStreamInfo(): SDK version: 8867047d8d8f56645009ed73788a57f0a6107772
2023-01-23 16:21:17 [1995788032] DEBUG - logStreamInfo(): Kinesis Video Stream Info
2023-01-23 16:21:17 [1995788032] DEBUG - logStreamInfo(): 	Stream name: video_stream_name_doni
2023-01-23 16:21:17 [1995788032] DEBUG - logStreamInfo(): 	Streaming type: STREAMING_TYPE_REALTIME
2023-01-23 16:21:17 [1995788032] DEBUG - logStreamInfo(): 	Content type: video/h264
2023-01-23 16:21:17 [1995788032] DEBUG - logStreamInfo(): 	Max latency (100ns): 600000000
2023-01-23 16:21:17 [1995788032] DEBUG - logStreamInfo(): 	Fragment duration (100ns): 20000000
2023-01-23 16:21:17 [1995788032] DEBUG - logStreamInfo(): 	Key frame fragmentation: Yes
2023-01-23 16:21:17 [1995788032] DEBUG - logStreamInfo(): 	Use frame timecode: Yes
2023-01-23 16:21:17 [1995788032] DEBUG - logStreamInfo(): 	Absolute frame timecode: Yes
2023-01-23 16:21:17 [1995788032] DEBUG - logStreamInfo(): 	Nal adaptation flags: 0
2023-01-23 16:21:17 [1995788032] DEBUG - logStreamInfo(): 	Average bandwith (bps): 4194304
2023-01-23 16:21:17 [1995788032] DEBUG - logStreamInfo(): 	Framerate: 25
2023-01-23 16:21:17 [1995788032] DEBUG - logStreamInfo(): 	Buffer duration (100ns): 1200000000
2023-01-23 16:21:17 [1995788032] DEBUG - logStreamInfo(): 	Replay duration (100ns): 400000000
2023-01-23 16:21:17 [1995788032] DEBUG - logStreamInfo(): 	Connection Staleness duration (100ns): 600000000
2023-01-23 16:21:17 [1995788032] DEBUG - logStreamInfo(): 	Store Pressure Policy: 1
2023-01-23 16:21:17 [1995788032] DEBUG - logStreamInfo(): 	View Overflow Policy: 1
2023-01-23 16:21:17 [1995788032] DEBUG - logStreamInfo(): 	Segment UUID: NULL
2023-01-23 16:21:17 [1995788032] DEBUG - logStreamInfo(): 	Frame ordering mode: 0
2023-01-23 16:21:17 [1995788032] DEBUG - logStreamInfo(): Track list
2023-01-23 16:21:17 [1995788032] DEBUG - logStreamInfo(): 	Track id: 1
2023-01-23 16:21:17 [1995788032] DEBUG - logStreamInfo(): 	Track name: kinesis_video
2023-01-23 16:21:17 [1995788032] DEBUG - logStreamInfo(): 	Codec id: V_MPEG4/ISO/AVC
2023-01-23 16:21:17 [1995788032] DEBUG - logStreamInfo(): 	Track type: TRACK_INFO_TYPE_VIDEO
2023-01-23 16:21:17 [1995788032] DEBUG - logStreamInfo(): 	Track cpd: NULL
2023-01-23 16:21:17 [1995788032] INFO - writeHeaderCallback(): RequestId: 8f317d80-b36d-437a-9d44-b451b3737619
2023-01-23 16:21:18 [1815413824] DEBUG - describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.674167064804E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","IngestionConfiguration":null,"KmsKeyId":"arn:aws:kms:us-west-2:149414536000:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-west-2:149414536000:stream/video_stream_name_doni/1674167064804","StreamName":"video_stream_name_doni","Version":"nEj3ocz154rjI04P3wVs"}}
2023-01-23 16:21:18 [1815413824] INFO - describeStreamResultEvent(): Describe stream result event.
2023-01-23 16:21:18 [1815413824] INFO - writeHeaderCallback(): RequestId: d33fe7b7-abeb-40af-893d-8ac932782d74
2023-01-23 16:21:19 [1805644864] DEBUG - getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-56832325.kinesisvideo.us-west-2.amazonaws.com"}
2023-01-23 16:21:19 [1805644864] INFO - getStreamingEndpointResultEvent(): Get streaming endpoint result event.
2023-01-23 16:21:19 [1805644864] DEBUG - getStreamingTokenHandler invoked
2023-01-23 16:21:19 [1805644864] DEBUG - Refreshing credentials. Force refreshing: 1 Now time is: 1674487279670999731 Expiration: 18446744073709551615
2023-01-23 16:21:19 [1805644864] INFO - getStreamingTokenResultEvent(): Get streaming token result event.
2023-01-23 16:21:19 [1805644864] DEBUG - streamReadyHandler invoked
2023-01-23 16:21:19 [1805644864] Stream is ready
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Redistribute latency...
INFO - kinesisVideoStreamFormatChanged(): Stream format changed.
2023-01-23 16:21:20 [1815413824] INFO - putStreamResultEvent(): Put stream result event. New upload handle 0
2023-01-23 16:21:20 [1815413824] DEBUG - Kinesis Video client and stream metrics
	>> Overall storage byte size: 134217728
	>> Available storage byte size: 134199959
	>> Allocated storage byte size: 17769
	>> Total view allocation byte size: 144080
	>> Total streams elementary frame rate (fps): 25
[... 12 lines cut ...]
2023-01-23 16:21:21 [1640100928] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2023-01-23 16:21:21 [1640100928] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2023-01-23 16:21:22 [1640100928] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2023-01-23 16:21:22 [1640100928] DEBUG - postReadCallback(): Wrote 5673 bytes to Kinesis Video. Upload stream handle: 0
DEBUG - postReadCallback(): Wrote 8840 bytes to Kinesis Video. Upload stream handle: 0
2023-01-23 16:21:22 [1640100928] DEBUG - postReadCallback(): Wrote 8215 bytes to Kinesis Video. Upload stream handle: 0
2023-01-23 16:21:22 [1640100928] DEBUG - postReadCallback(): Wrote 8643 bytes to Kinesis Video. Upload stream handle: 0
DEBUG - postReadCallback(): Wrote 8215 bytes to Kinesis Video. Upload stream handle: 0
2023-01-23 16:21:22 [1640100928] DEBUG - postReadCallback(): Wrote 8760 bytes to Kinesis Video. Upload stream handle: 0
DEBUG - postReadCallback(): Wrote 8194 bytes to Kinesis Video. Upload stream handle: 0
2023-01-23 16:21:22 [1640100928] DEBUG - postReadCallback(): Wrote 8618 bytes to Kinesis Video. Upload stream handle: 0
DEBUG - postReadCallback(): Wrote 8415 bytes to Kinesis Video. Upload stream handle: 0
2023-01-23 16:21:22 [1640100928] DEBUG - postReadCallback(): Wrote 7823 bytes to Kinesis Video. Upload stream handle: 0
2023-01-23 16:21:22 [1640100928] DEBUG - postReadCallback(): Wrote 8239 bytes to Kinesis Video. Upload stream handle: 0
DEBUG - postReadCallback(): Wrote 8226 bytes to Kinesis Video. Upload stream handle: 0
2023-01-23 16:21:22 [1640100928] DEBUG - postReadCallback(): Wrote 8215 bytes to Kinesis Video. Upload stream handle: 0
DEBUG - postReadCallback(): Wrote 8232 bytes to Kinesis Video. Upload stream handle: 0
2023-01-23 16:21:22 [1640100928] DEBUG - postReadCallback(): Wrote 8216 bytes to Kinesis Video. Upload stream handle: 0
DEBUG - postReadCallback(): Wrote 8633 bytes to Kinesis Video. Upload stream handle: 0
2023-01-23 16:21:23 [1640100928] DEBUG - postReadCallback(): Wrote 8263 bytes to Kinesis Video. Upload stream handle: 0
`

What do I do wrong?

hassanctech, 2023-01-26:
Can you please update the title, right now it says [Question] but there are no details.

Looking at your logs everything seems to be working fine :).  Have you tried to open up the aws console and view the video, and/or inspect the IncomingBytes metric on the stream to see the bytes are coming in?  From the logging it looks like your stream name is `video_stream_name_doni ` in region `us-west-2`.

DunkinDonat, 2023-01-30:
Hey hassanctech,

thank you for your fast response.

If i go to in the AWS console to Kinesis Video Streams page and select my stream (video_stream_name_doni), then there is in every monitoring chart the writing: "no data available". I didnt find "IncomingBytes" but there is a "PutMedia Incoming Data" this is also saying "no data available".

The CLI output says: 
`DEBUG - postReadCallback(): Wrote 8276 bytes to Kinesis Video. Upload stream handle: 0
2023-01-30 14:18:35 [1641178176] DEBUG - Kinesis Video client and stream metrics
	>> Overall storage byte size: 134217728
	>> Available storage byte size: 133570701
	>> Allocated storage byte size: 647027
	>> Total view allocation byte size: 144080
	>> Total streams elementary frame rate (fps): 30
	>> Total streams transfer rate (bps): 1551288 (1514 Kbps)
	>> Current view duration (ms): 0
	>> Overall view duration (ms): 2466
	>> Current view byte size: 11543
	>> Overall view byte size: 644027
	>> Current elementary frame rate (fps): 30
	>> Current transfer rate (bps): 1551288 (1514 Kbps)`

What does `Upload stream handle: 0` mean?

DunkinDonat, 2023-01-30:
Hey hassantech,

you are right. I'm streaming into the wrong region. I need to stream to `eu-central-1 `. But now I need to figure out how to configure that.

DunkinDonat, 2023-01-30:
Is there a documentation for **all parameters** the plugin (kvssink) can have?

hassanctech, 2023-01-30:
It is not specific to `kvssink` but the standard way to get details about a GStreamer plugin (including all the parameters) is to simply run `gst-inspect-1.0 kvssink`.  Make sure your `GST_PLUGIN_PATH` is set to where it can find `libgstkvssink.so`.

I will also just state we do have official documentation for this:  https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/examples-gstreamer-plugin-parameters.html

Regardless if there is a discrepancy between that documentation and what the inspect command returns, the latter is the source of truth since it queries the plugin directly.

Also it looks like the issue is resolved so I will close this ticket.  If you need more help please open a new ticket.

closed 2023-01-30 by hassanctech