awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #939: Segmetation fault on alpine docker when runing gst-launch-1.0

adirkola, 2023-01-25:
**Logging**
gst-launch-1.0 v4l2src do-timestamp=TRUE device="/dev/video0" ! videoconvert \^C                                                                                                          
/usr/lib/build #    gst-launch-1.0 v4l2src do-timestamp=TRUE device="/dev/video0" ! videoconvert \
>         ! video/x-raw,format=I420,width=640,height=480,framerate=30/1 \
>         ! x264enc tune=zerolatency bframes=0 key-int-max=45 bitrate=500 \
>         ! video/x-h264,stream-format=avc,alignment=au,profile=baseline \
>         ! kvssink stream-name="test-stream" storage-size=512 aws-region="us-east-1" 
Setting pipeline to PAUSED ...
log4cplus:ERROR could not open file ../kvs_log_configuration
INFO - createKinesisVideoClient(): Creating Kinesis Video Client
2023-01-24 16:04:50 [547533081672] INFO - heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
2023-01-24 16:04:50 [547533081672] INFO - heapInitialize(): Creating AIV heap.
2023-01-24 16:04:50 [547533081672] INFO - heapInitialize(): Heap is initialized OK
2023-01-24 16:04:50 [547533081672] DEBUG - getSecurityTokenHandler invoked
2023-01-24 16:04:50 [547533081672] DEBUG - Refreshing credentials. Force refreshing: 0 Now time is: 1674576290643185837 Expiration: 0
2023-01-24 16:04:50 [547533081672] INFO - createDeviceResultEvent(): Create device result event.
2023-01-24 16:04:50 [547533081672] DEBUG - clientReadyHandler invoked
2023-01-24 16:04:50 [547533081672] INFO - try creating stream
2023-01-24 16:04:50 [547533081672] INFO - Creating Kinesis Video Stream test-stream
2023-01-24 16:04:50 [547533081672] INFO - createKinesisVideoStream(): Creating Kinesis Video Stream.
2023-01-24 16:04:50 [547533081672] INFO - logStreamInfo(): SDK version: 8867047d8d8f56645009ed73788a57f0a6107772
2023-01-24 16:04:50 [547533081672] DEBUG - logStreamInfo(): Kinesis Video Stream Info
2023-01-24 16:04:50 [547533081672] DEBUG - logStreamInfo():     Stream name: test-stream 
2023-01-24 16:04:50 [547533081672] DEBUG - logStreamInfo():     Streaming type: STREAMING_TYPE_REALTIME 
2023-01-24 16:04:50 [547533081672] DEBUG - logStreamInfo():     Content type: video/h264 
2023-01-24 16:04:50 [547533081672] DEBUG - logStreamInfo():     Max latency (100ns): 600000000
2023-01-24 16:04:50 [547533081672] DEBUG - logStreamInfo():     Fragment duration (100ns): 20000000
2023-01-24 16:04:50 [547533081672] DEBUG - logStreamInfo():     Key frame fragmentation: Yes
2023-01-24 16:04:50 [547533081672] DEBUG - logStreamInfo():     Use frame timecode: Yes
2023-01-24 16:04:50 [547533081672] DEBUG - logStreamInfo():     Absolute frame timecode: Yes
2023-01-24 16:04:50 [547533081672] DEBUG - logStreamInfo():     Nal adaptation flags: 0
2023-01-24 16:04:50 [547533081672] DEBUG - logStreamInfo():     Average bandwith (bps): 4194304
2023-01-24 16:04:50 [547533081672] DEBUG - logStreamInfo():     Framerate: 25
2023-01-24 16:04:50 [547533081672] DEBUG - logStreamInfo():     Buffer duration (100ns): 1200000000
2023-01-24 16:04:50 [547533081672] DEBUG - logStreamInfo():     Replay duration (100ns): 400000000
2023-01-24 16:04:50 [547533081672] DEBUG - logStreamInfo():     Connection Staleness duration (100ns): 600000000
2023-01-24 16:04:50 [547533081672] DEBUG - logStreamInfo():     Store Pressure Policy: 1
2023-01-24 16:04:50 [547533081672] DEBUG - logStreamInfo():     View Overflow Policy: 1
2023-01-24 16:04:50 [547533081672] DEBUG - logStreamInfo():     Segment UUID: NULL
2023-01-24 16:04:50 [547533081672] DEBUG - logStreamInfo():     Frame ordering mode: 0
2023-01-24 16:04:50 [547533081672] DEBUG - logStreamInfo(): Track list
2023-01-24 16:04:50 [547533081672] DEBUG - logStreamInfo():     Track id: 1
2023-01-24 16:04:50 [547533081672] DEBUG - logStreamInfo():     Track name: kinesis_video
2023-01-24 16:04:50 [547533081672] DEBUG - logStreamInfo():     Codec id: V_MPEG4/ISO/AVC
2023-01-24 16:04:50 [547533081672] DEBUG - logStreamInfo():     Track type: TRACK_INFO_TYPE_VIDEO
2023-01-24 16:04:50 [547533081672] DEBUG - logStreamInfo():     Track cpd: NULL
Segmentation fault (core dumped)

**Describe the bug**
1. compile from source the repo on arm64v8/alpine:3.16 docker with kvssink plugin. after compilation I did the relevant exports.
2. checked with gst-inspect that kvssink plugin is exist. added with export the secret key, access key, session token.
run gst-launch-1.0 as written at the logs above but the program is crashing 
is there any why to run the program properly on alpine docker i have problem with space on device that's why i am working with alpine 
thanks

**SDK version number**
master branch v3

MushMal, 2023-02-07:
Not much to go with but the address seems corrupted. Does your device allow unaligned memory access? Are you able to run the sample application with clean compilation without any issues (no gstreamer)? It could also be caused by stale files/mismatched headers.

adirkola, 2023-02-08:
thanks for the answer @MushMal 
1. Not much to go with but the address seems corrupted
tell me what you need from me I will provide it.
2. Does your device allow unaligned memory access?
yes, I am using artik710 and artik710s.
3. Are you able to run the sample application with clean compilation without any issues (no gstreamer)?
all the samples in the repo are with gstreamer am I wrong? and I want to add that when I run this on ubuntu docker it works as expected. 
4. It could also be caused by stale files/mismatched headers.
i compile the code with dependencies and with the open-source third-party libs that this project is providing so i am not sure about that. have to say that i didnt run it on clean alpine docker i used a docker image that i had and add this repo and compile it on that so maybe it is important to provide you this information if i am succeed to make this work on clean alpine docker

disa6302, 2023-02-20:
@adirkola , 

Could you confirm the curl version on your docker container? The SDK requires curl version 7.68, I am wondering if your docker has a higher version that could be causing the issues.

I suspect if it is related to this: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/844

hassanctech, 2023-03-06:
Also as you stated you did not run on a clean docker, can you make sure during runtime your process is loading the correct openssl/curl that you built using the build dependencies and not a system version so we can rule out (4) from above?  `lsof -p <pid>` should should you that, that should be available on alpine linux.

closed 2023-03-10 by niyatim23