awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #941: kvssink - WARNING: erroneous pipeline: No sink-element named "(null)" - omitting link

DunkinDonat, 2023-01-30:
Hey, 

I'm trying to stream audio and video to kinesis like described in this example:
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/docs/linux.md#running-the-gst-launch-10-command-to-start-streaming-both-audio-and-video-in-raspberry-pi-and-ubuntu

**I'm **able** to stream audio only:**
`gst-launch-1.0 -m -v -e  audiotestsrc ! audioconvert ! queue ! avenc_aac ! kvssink stream-name="" access-key="" secret-key="" aws-region=""
`

I'm **able** to stream video only:
gst-launch-1.0 videotestsrc ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=30/1 ! x264enc ! h264parse! video/x-h264,stream-format=avc,alignment=au,width=640,height=480,framerate=30/1,profile=baseline ! kvssink stream-name="" access-key="" secret-key="" aws-region=""

Both together **fails**:
`gst-launch-1.0 videotestsrc ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=30/1 ! x264enc ! h264parse! video/x-h264,stream-format=avc,alignment=au,width=640,height=480,framerate=30/1,profile=baseline ! kvssink stream-name="" access-key="" secret-key="" aws-region="" audiotestsrc ! audioconvert ! queue ! avenc_aac ! sink.
`

** Logging **
`WARNING: erroneous pipeline: No sink-element named "(null)" - omitting link`

Any ideas?

hassanctech, 2023-01-30:
Here is a sample pipeline we have with audio in our docs:

```
gst-launch-1.0 -v v4l2src device=/dev/video0 ! h264parse ! video/x-h264,stream-format=avc,alignment=au ! kvssink name=sink stream-name="my-stream-name" access-key="YourAccessKey" secret-key="YourSecretKey" alsasrc device=hw:1,0 ! audioconvert ! avenc_aac ! queue ! sink.
```

Looks like your error is the ordering of `queue` and `avenc_aac`.  You need to do the audio encode before the `queue` element.

closed 2023-01-30 by hassanctech

DunkinDonat, 2023-01-30:
Found my fault:

you need to add `name=sink` to define sink