awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #942: [QUESTION] Video continuously pauses when streaming from deepstream python app

razekmaiden, 2023-02-02:
Hello everyone. I'm trying to stream video to kinesis from an application using Nvidia Deepstream (based on Gstreamer) on a Jetson Xavier AGX. This application currently reads a video, processes it (inference) and at the end of the pipeline there is kvssink that performs the transmission.

I've tried setting the iframeinterval parameters of nvv4l2h264enc, along with the fragment-duration and framerate parameters of kvssink. However, when viewing the video in the AWS player, it continually pauses, apparently waiting for the arrival of new fragments ([Video](https://www.youtube.com/watch?v=szFAqRPeXGg)).

I am not an expert in Gstreamer although little by little I have managed to learn. Until now I have mainly tried to vary the aforementioned parameters, without having better results.

I appreciate your time. Regards!!

** Logging **
Add relevent SDK logging.  IMPORTANT NOTE:  Please make sure to NOT share AWS access credentials under any circumstance!  Please make sure they are not in the logs.

> [INFO ] [02-02-2023 19:11:09:663.780 GMT] createKinesisVideoClient(): Creating Kinesis Video Client
> [INFO ] [02-02-2023 19:11:09:663.995 GMT] heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
> [INFO ] [02-02-2023 19:11:09:664.032 GMT] heapInitialize(): Creating AIV heap.
> [INFO ] [02-02-2023 19:11:09:664.119 GMT] heapInitialize(): Heap is initialized OK
> [DEBUG] [02-02-2023 19:11:09:664.876 GMT] getSecurityTokenHandler invoked
> [DEBUG] [02-02-2023 19:11:09:664.959 GMT] Refreshing credentials. Force refreshing: 0 Now time is: 1675365069664951113 Expiration: 0
> [INFO ] [02-02-2023 19:11:09:665.005 GMT] createDeviceResultEvent(): Create device result event.
> [DEBUG] [02-02-2023 19:11:09:665.043 GMT] clientReadyHandler invoked
> [INFO ] [02-02-2023 19:11:09:665.209 GMT] try creating stream
> [INFO ] [02-02-2023 19:11:09:665.273 GMT] Creating Kinesis Video Stream RLBokqksQo7p1g3yujU0_FsJPrfvuGifvcpaHovkw_2
> [INFO ] [02-02-2023 19:11:09:665.309 GMT] createKinesisVideoStream(): Creating Kinesis Video Stream.
> [INFO ] [02-02-2023 19:11:09:665.340 GMT] logStreamInfo(): SDK version: 70f74f14cf27b09f71dc1889f36eb6e04cdd90a8
> [DEBUG] [02-02-2023 19:11:09:665.367 GMT] logStreamInfo(): Kinesis Video Stream Info
> [DEBUG] [02-02-2023 19:11:09:665.393 GMT] logStreamInfo(): 	Stream name: RLBokqksQo7p1g3yujU0_FsJPrfvuGifvcpaHovkw_2 
> [DEBUG] [02-02-2023 19:11:09:665.418 GMT] logStreamInfo(): 	Streaming type: STREAMING_TYPE_REALTIME 
> [DEBUG] [02-02-2023 19:11:09:665.478 GMT] logStreamInfo(): 	Content type: video/h264 
> [DEBUG] [02-02-2023 19:11:09:665.506 GMT] logStreamInfo(): 	Max latency (100ns): 600000000
> [DEBUG] [02-02-2023 19:11:09:665.560 GMT] logStreamInfo(): 	Fragment duration (100ns): 15000000
> [DEBUG] [02-02-2023 19:11:09:665.587 GMT] logStreamInfo(): 	Key frame fragmentation: Yes
> [DEBUG] [02-02-2023 19:11:09:665.630 GMT] logStreamInfo(): 	Use frame timecode: Yes
> [DEBUG] [02-02-2023 19:11:09:665.654 GMT] logStreamInfo(): 	Absolute frame timecode: Yes
> [DEBUG] [02-02-2023 19:11:09:665.678 GMT] logStreamInfo(): 	Nal adaptation flags: 0
> [DEBUG] [02-02-2023 19:11:09:665.702 GMT] logStreamInfo(): 	Average bandwith (bps): 4194304
> [DEBUG] [02-02-2023 19:11:09:665.725 GMT] logStreamInfo(): 	Framerate: 24
> [DEBUG] [02-02-2023 19:11:09:665.750 GMT] logStreamInfo(): 	Buffer duration (100ns): 20000000
> [DEBUG] [02-02-2023 19:11:09:665.774 GMT] logStreamInfo(): 	Replay duration (100ns): 20000000
> [DEBUG] [02-02-2023 19:11:09:665.798 GMT] logStreamInfo(): 	Connection Staleness duration (100ns): 600000000
> [DEBUG] [02-02-2023 19:11:09:665.822 GMT] logStreamInfo(): 	Store Pressure Policy: 1
> [DEBUG] [02-02-2023 19:11:09:665.847 GMT] logStreamInfo(): 	View Overflow Policy: 1
> [DEBUG] [02-02-2023 19:11:09:665.873 GMT] logStreamInfo(): 	Segment UUID: NULL
> [DEBUG] [02-02-2023 19:11:09:665.901 GMT] logStreamInfo(): 	Frame ordering mode: 0
> [DEBUG] [02-02-2023 19:11:09:665.924 GMT] logStreamInfo(): Track list
> [DEBUG] [02-02-2023 19:11:09:665.971 GMT] logStreamInfo(): 	Track id: 1
> [DEBUG] [02-02-2023 19:11:09:665.996 GMT] logStreamInfo(): 	Track name: kinesis_video
> [DEBUG] [02-02-2023 19:11:09:666.035 GMT] logStreamInfo(): 	Codec id: V_MPEG4/ISO/AVC
> [DEBUG] [02-02-2023 19:11:09:666.060 GMT] logStreamInfo(): 	Track type: TRACK_INFO_TYPE_VIDEO
> [DEBUG] [02-02-2023 19:11:09:666.109 GMT] logStreamInfo(): 	Track cpd: NULL
> [INFO ] [02-02-2023 19:11:10:285.398 GMT] writeHeaderCallback(): RequestId: 8ca06d31-1f29-4cfa-9546-9c5ce4328acb
> [DEBUG] [02-02-2023 19:11:10:285.619 GMT] describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.675274432424E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","IngestionConfiguration":null,"KmsKeyId":"arn:aws:kms:us-east-1:532644800621:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-east-1:532644800621:stream/RLBokqksQo7p1g3yujU0_FsJPrfvuGifvcpaHovkw_2/1675274432424","StreamName":"RLBokqksQo7p1g3yujU0_FsJPrfvuGifvcpaHovkw_2","Version":"vkTeIiR
> [INFO ] [02-02-2023 19:11:10:287.653 GMT] describeStreamResultEvent(): Describe stream result event.
> [INFO ] [02-02-2023 19:11:10:842.739 GMT] writeHeaderCallback(): RequestId: e4f39920-d608-41ee-b1af-c90f38aaa9c6
> [DEBUG] [02-02-2023 19:11:10:842.962 GMT] getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-356988f0.kinesisvideo.us-east-1.amazonaws.com"}
> [INFO ] [02-02-2023 19:11:10:845.002 GMT] getStreamingEndpointResultEvent(): Get streaming endpoint result event.
> [DEBUG] [02-02-2023 19:11:10:845.111 GMT] getStreamingTokenHandler invoked
> [DEBUG] [02-02-2023 19:11:10:845.162 GMT] Refreshing credentials. Force refreshing: 1 Now time is: 1675365070845155180 Expiration: 18446744073709551615
> [INFO ] [02-02-2023 19:11:10:845.225 GMT] getStreamingTokenResultEvent(): Get streaming token result event.
> [DEBUG] [02-02-2023 19:11:10:845.274 GMT] streamReadyHandler invoked
> Stream is ready
> Opening in BLOCKING MODE 
> gstnvtracker: Loading low-level lib at /opt/nvidia/deepstream/deepstream/lib/libnvds_nvmultiobjecttracker.so
> gstnvtracker: Batch processing is ON
> gstnvtracker: Past frame output is OFF
> [NvMultiObjectTracker] Initialized
> WARNING: [TRT]: Using an engine plan file across different models of devices is not recommended and is likely to affect performance or even cause errors.
> INFO: [Implicit Engine Info]: layers num: 3
> 0   INPUT  kFLOAT input_1         3x544x960       
> 1   OUTPUT kFLOAT output_bbox/BiasAdd 4x34x60         
> 2   OUTPUT kFLOAT output_cov/Sigmoid 1x34x60         
> 
[... 168 lines cut ...]
> [DEBUG] [02-02-2023 19:11:28:121.869 GMT] postReadCallback(): Wrote 16264 bytes to Kinesis Video. Upload stream handle: 0
> [DEBUG] [02-02-2023 19:11:28:262.676 GMT] postReadCallback(): Wrote 13621 bytes to Kinesis Video. Upload stream handle: 0
> [DEBUG] [02-02-2023 19:11:28:403.395 GMT] postReadCallback(): Wrote 8709 bytes to Kinesis Video. Upload stream handle: 0
> [DEBUG] [02-02-2023 19:11:28:544.099 GMT] postReadCallback(): Wrote 9605 bytes to Kinesis Video. Upload stream handle: 0
> [DEBUG] [02-02-2023 19:11:28:604.841 GMT] Kinesis Video client and stream metrics
> 	>> Overall storage byte size: 536870912
> 	>> Available storage byte size: 535841654
> 	>> Allocated storage byte size: 1029258
> 	>> Total view allocation byte size: 2384
> 	>> Total streams elementary frame rate (fps): 24
> 	>> Total streams transfer rate (bps): 3464752 (3383 Kbps)
> 	>> Current view duration (ms): 0
> 	>> Overall view duration (ms): 1442
> 	>> Current view byte size: 290123
> 	>> Overall view byte size: 1027482
> 	>> Current elementary frame rate (fps): 24.975
> 	>> Current transfer rate (bps): 3464752 (3383 Kbps)
> [DEBUG] [02-02-2023 19:11:28:684.833 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
> [DEBUG] [02-02-2023 19:11:28:685.336 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
> [DEBUG] [02-02-2023 19:11:28:685.818 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
> [DEBUG] [02-02-2023 19:11:28:686.172 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
> [DEBUG] [02-02-2023 19:11:28:686.488 GMT] postReadCallback(): Wrote 28027 bytes to Kinesis Video. Upload stream handle: 0
[... cut ...] rote 12175 bytes to Kinesis Video. Upload stream handle: 0
> [DEBUG] [02-02-2023 19:11:28:887.673 GMT] postReadCallback(): Wrote 9541 bytes to Kinesis Video. Upload stream handle: 0
> [DEBUG] [02-02-2023 19:11:28:888.064 GMT] postWriteCallback(): Curl post body write function for stream with handle: RLBokqksQo7p1g3yujU0_FsJPrfvuGifvcpaHovkw_2 and upload handle: 0 returned: {"EventType":"RECEIVED","FragmentTimecode":1675365077828,"FragmentNumber":"91343852333181541341309040079383833462330220184"}

sirknightj, 2023-02-07:
Hi @razekmaiden,

I am seeing the following line in your logs:
```
WARNING: [TRT]: Using an engine plan file across different models of devices is not recommended and is likely to affect performance or even cause errors.
```

Could you try running the sample application and check to see if there are any errors with the stream?

niyatim23, 2023-02-14:
Closing due to no response. Feel free to reopen / open a new issue if the problem persists

closed 2023-02-14 by niyatim23