awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #956: No Stream in Kinesis Video

Alex-Wenner-FHR, 2023-02-17:
I am using the dockerfile example for the rtsp stream. Things appear to be streaming from within the container, but I am seeing nothing in Kinsesis Video except a log that says:
`No media found
Verify that you are streaming to us-east-1 and the fragment timecodes are correct, or download the Kinesis Video Streams Producer SDK to set up your device.`

## Dockerfile
Here is a sample of my run Dockerfile Entrypoint:
```Dockerfile
ENTRYPOINT gst-launch-1.0 rtspsrc location=${RTSP_STREAM_URL} ! rtph264depay ! h264parse ! kvssink stream-name=${STREAM_NAME} storage-size=512 framerate=30
```
The rest of my Dockerfile looks like the example [Here]
(https://github.com/aws-samples/amazon-kinesis-video-streams-demos/blob/master/producer-cpp/docker-rtsp/Dockerfile)

## Logging
Here are sample logs of my container boot.
```python
docker run -it latest
Setting pipeline to PAUSED ...
[INFO ] [17-02-2023 14:37:43:033.064 GMT] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ] [17-02-2023 14:37:43:033.193 GMT] heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
[INFO ] [17-02-2023 14:37:43:033.214 GMT] heapInitialize(): Creating AIV heap.
[INFO ] [17-02-2023 14:37:43:033.244 GMT] heapInitialize(): Heap is initialized OK
[DEBUG] [17-02-2023 14:37:43:033.564 GMT] getSecurityTokenHandler invoked
[DEBUG] [17-02-2023 14:37:43:033.643 GMT] Refreshing credentials. Force refreshing: 0 Now time is: 1676644663033638676 Expiration: 0
[INFO ] [17-02-2023 14:37:43:033.671 GMT] createDeviceResultEvent(): Create device result event.
[DEBUG] [17-02-2023 14:37:43:033.692 GMT] clientReadyHandler invoked
[INFO ] [17-02-2023 14:37:43:033.783 GMT] try creating stream
[INFO ] [17-02-2023 14:37:43:033.904 GMT] Creating Kinesis Video Stream Axis_Testing_RTSP
[INFO ] [17-02-2023 14:37:43:033.974 GMT] createKinesisVideoStream(): Creating Kinesis Video Stream.
[INFO ] [17-02-2023 14:37:43:034.071 GMT] logStreamInfo(): SDK version: 8867047d8d8f56645009ed73788a57f0a6107772
[DEBUG] [17-02-2023 14:37:43:034.182 GMT] logStreamInfo(): Kinesis Video Stream Info
[DEBUG] [17-02-2023 14:37:43:034.287 GMT] logStreamInfo():      Stream name: Axis_Testing_RTSP 
[DEBUG] [17-02-2023 14:37:43:034.330 GMT] logStreamInfo():      Streaming type: STREAMING_TYPE_REALTIME 
[DEBUG] [17-02-2023 14:37:43:034.417 GMT] logStreamInfo():      Content type: video/h264 
[DEBUG] [17-02-2023 14:37:43:034.454 GMT] logStreamInfo():      Max latency (100ns): 600000000
[DEBUG] [17-02-2023 14:37:43:034.520 GMT] logStreamInfo():      Fragment duration (100ns): 20000000
[DEBUG] [17-02-2023 14:37:43:034.594 GMT] logStreamInfo():      Key frame fragmentation: Yes
[DEBUG] [17-02-2023 14:37:43:034.698 GMT] logStreamInfo():      Use frame timecode: Yes
[DEBUG] [17-02-2023 14:37:43:034.732 GMT] logStreamInfo():      Absolute frame timecode: Yes
[DEBUG] [17-02-2023 14:37:43:034.833 GMT] logStreamInfo():      Nal adaptation flags: 0
[DEBUG] [17-02-2023 14:37:43:034.849 GMT] logStreamInfo():      Average bandwith (bps): 4194304
[DEBUG] [17-02-2023 14:37:43:034.861 GMT] logStreamInfo():      Framerate: 30
[DEBUG] [17-02-2023 14:37:43:034.873 GMT] logStreamInfo():      Buffer duration (100ns): 1200000000
[DEBUG] [17-02-2023 14:37:43:034.889 GMT] logStreamInfo():      Replay duration (100ns): 400000000
[DEBUG] [17-02-2023 14:37:43:034.915 GMT] logStreamInfo():      Connection Staleness duration (100ns): 600000000
[DEBUG] [17-02-2023 14:37:43:034.937 GMT] logStreamInfo():      Store Pressure Policy: 1
[DEBUG] [17-02-2023 14:37:43:034.960 GMT] logStreamInfo():      View Overflow Policy: 1
[DEBUG] [17-02-2023 14:37:43:035.051 GMT] logStreamInfo():      Segment UUID: NULL
[DEBUG] [17-02-2023 14:37:43:035.067 GMT] logStreamInfo():      Frame ordering mode: 0
[DEBUG] [17-02-2023 14:37:43:035.077 GMT] logStreamInfo(): Track list
[DEBUG] [17-02-2023 14:37:43:035.085 GMT] logStreamInfo():      Track id: 1
[DEBUG] [17-02-2023 14:37:43:035.092 GMT] logStreamInfo():      Track name: kinesis_video
[DEBUG] [17-02-2023 14:37:43:035.099 GMT] logStreamInfo():      Codec id: V_MPEG4/ISO/AVC
[DEBUG] [17-02-2023 14:37:43:035.106 GMT] logStreamInfo():      Track type: TRACK_INFO_TYPE_VIDEO
[DEBUG] [17-02-2023 14:37:43:035.114 GMT] logStreamInfo():      Track cpd: NULL
[INFO ] [17-02-2023 14:37:43:495.086 GMT] writeHeaderCallback(): RequestId: 5c9acfee-ffc4-4675-8e12-82a5e8ff55f1
[DEBUG] [17-02-2023 14:37:43:495.341 GMT] describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.676563940598E9,"DataRetentionInHours":24,"DeviceName":null,"IngestionConfiguration":null,"KmsKeyId":"arn:aws:kms:us-east-1:<acct_ID>:alias/aws/kinesisvideo","MediaType":null,"Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-east-1:<acct_ID>:stream/Axis_Testing_RTSP/1676563940598","StreamName":"Axis_Testing_RTSP","Version":"<Version_ID>"}}
[INFO ] [17-02-2023 14:37:43:496.872 GMT] describeStreamResultEvent(): Describe stream result event.
[WARN ] [17-02-2023 14:37:43:496.962 GMT] describeStreamResult(): Retention period returned from the DescribeStream call doesn't match the one specified in the StreamInfo
[WARN ] [17-02-2023 14:37:43:496.992 GMT] describeStreamResult(): Content type returned from the DescribeStream call doesn't match the one specified in the StreamInfo
[INFO ] [17-02-2023 14:37:43:908.088 GMT] writeHeaderCallback(): RequestId: 3ddd8955-cd74-4c18-b272-ef793656ab85
[DEBUG] [17-02-2023 14:37:43:908.488 GMT] getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-f10ab6ea.kinesisvideo.us-east-1.amazonaws.com"}
[INFO ] [17-02-2023 14:37:43:910.555 GMT] getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG] [17-02-2023 14:37:43:910.826 GMT] getStreamingTokenHandler invoked
[DEBUG] [17-02-2023 14:37:43:910.852 GMT] Refreshing credentials. Force refreshing: 1 Now time is: 1676644663910849369 Expiration: 18446744073709551615
[INFO ] [17-02-2023 14:37:43:911.060 GMT] getStreamingTokenResultEvent(): Get streaming token result event.
[DEBUG] [17-02-2023 14:37:43:911.222 GMT] streamReadyHandler invoked
Stream is ready
Pipeline is live and does not need PREROLL ...
Progress: (open) Opening Stream
Progress: (connect) Connecting to rtsp://<my_camera_IP>/axis-media/media.amp?videocodec=h264
[... 108 lines cut ...]
[DEBUG] [17-02-2023 14:38:13:651.767 GMT] getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-f10ab6ea.kinesisvideo.us-east-1.amazonaws.com"}
[INFO ] [17-02-2023 14:38:13:653.094 GMT] getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG] [17-02-2023 14:38:13:653.200 GMT] getStreamingTokenHandler invoked
[DEBUG] [17-02-2023 14:38:13:653.226 GMT] Refreshing credentials. Force refreshing: 1 Now time is: 1676644693653222826 Expiration: 18446744073709551615
[INFO ] [17-02-2023 14:38:13:653.309 GMT] getStreamingTokenResultEvent(): Get streaming token result event.
[DEBUG] [17-02-2023 14:38:13:653.370 GMT] streamReadyHandler invoked
[INFO ] [17-02-2023 14:38:13:653.722 GMT] putStreamResultEvent(): Put stream result event. New upload handle 1
[INFO ] [17-02-2023 14:38:14:130.622 GMT] writeHeaderCallback(): RequestId: d39c51c9-ceae-3cbb-8ff5-c25f046e1429
[DEBUG] [17-02-2023 14:38:14:761.332 GMT] postReadCallback(): Pausing CURL read for upload handle: 1
```

## Cloudwatch Details
- PutMedia Active Connections show active connections as expected.
- PutMedia Success shows successful Put events.
- HLS and MPEG-DASH Requests shows GetDASHStreamingSessionURL.Requests
- HLS and MPEG-DASH Success shows _0 successful events_
- Everything else has no data _which seems fishy_

## Steps Taken
- I have confirmed my AWS Credentials are being picked up as expected
- I have confirmed my Region is set correctly. 
- I do notice the error `MAX_FRAGMENT_DURATION_REACHED` and have attempted to mitigate but not super sure how to do so successfully.
- Confirmed that my camera supports rtsp streaming.

Any help would be much appreciated. I am very stuck!

Alex-Wenner-FHR, 2023-02-17:
@niyatim23 thanks for the swift response! 

How would I go about checking the "up-to 10 seconds GoP"? I am now super experienced when it comes to video streaming. I think I can double check the H264 but not sure about the latter comment!

niyatim23, 2023-02-17:
@Alex-Wenner-FHR, GoP is the distance between two consecutive key frames, it can be measured in frames or in the amount of time between key frames. You can also check your key frame interval if you are using a gstreamer pipeline. 
If you have a sample of the video that you are sending, you can use `ffprobe` to figure out the GoP with the following:
`ffprobe -i example.mkv -show_frames -of flat |grep I `

Alex-Wenner-FHR, 2023-02-17:
@niyatim23 got it. So I turned off Dynamic GOP on our hardware. It seems to still hit the `MAX_FRAGMENT_DURATION_REACHED` error and seems to restart itself. However when it comes back online it looks like it resumes writing bytes again which is not what I had seen previously... maybe that is progress? I will keep investigating this. I would like to keep this issue open for now as a line of communication if that is okay!

Alex-Wenner-FHR, 2023-02-17:
@niyatim23 thanks for the help, that seems to have worked. It appears that the bitrate is low... is there a good way to improve that? I am at 1.9mbps... any thoughts?

Alex-Wenner-FHR, 2023-02-17:
Closing, this seems resolved. Thanks for the help!

closed 2023-02-17 by Alex-Wenner-FHR