awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #978: [BUG] Need to run cmake twice to build

sirknightj, 2023-02-24:
**Logging**
```
$ cmake ..
-- The C compiler identification is AppleClang 13.1.6.13160021
-- The CXX compiler identification is AppleClang 13.1.6.13160021
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Kinesis Video Cpp Producer path is /Users/Me/Downloads/test/amazon-kinesis-video-streams-producer-sdk-cpp
-- dependencies install path is /Users/Me/Downloads/test/amazon-kinesis-video-streams-producer-sdk-cpp/open-source
-- Begin building dependencies.
-- autoconf already built
-- automake already built
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/Me/Downloads/test/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/liblog4cplus
[ 11%] Creating directories for 'project_log4cplus'
[ 22%] Performing download step (git clone) for 'project_log4cplus'
Cloning into 'project_log4cplus'...
HEAD is now at 6ae6110a Update Automake to 1.16.1.
[ 33%] Performing update step for 'project_log4cplus'
[ 44%] No patch step for 'project_log4cplus'
[ 55%] Performing configure step for 'project_log4cplus'
checking build system type... arm-apple-darwin21.6.0
checking host system type... arm-apple-darwin21.6.0
checking target system type... arm-apple-darwin21.6.0
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... ./install-sh -c -d
checking for gawk... no
checking for mawk... no
checking for nawk... no
checking for awk... awk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... yes
checking whether make supports the include directive... yes (GNU style)
checking for gcc... /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
checking whether the C compiler works... no
configure: error: in `/Users/Me/Downloads/test/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/liblog4cplus/build/src/project_log4cplus':
configure: error: C compiler cannot create executables
See `config.log' for more details
make[2]: *** [build/src/project_log4cplus-stamp/project_log4cplus-configure] Error 77
make[1]: *** [CMakeFiles/project_log4cplus.dir/all] Error 2
make: *** [all] Error 2
CMake Error at CMake/Utilities.cmake:99 (message):
  CMake step for liblog4cplus failed: 2
Call Stack (most recent call first):
  CMakeLists.txt:85 (build_dependency)
```

**Describe the bug**
Running `cmake ..` does not build the first time, results in a cmake error.

**To Reproduce**
Steps to reproduce the behavior:
1. Clone the repository
2. Checkout `develop`
3. `mkdir build && cd build`
4. `cmake ..`

**Expected behavior**
Only need to run it once. Running `cmake ..` a second time builds successfully.

**SDK version number**
Latest `develop` branch (4abb7cd5e78098e1e5c7e217f77b61e9911e272a)

**Platform (please complete the following information):**
 - OS: Mac, M1

**Additional context**
* Can build it fine without needing to run `cmake ..` twice on Ubuntu 22.04.1 LTS

sirknightj, 2023-02-24:
Another stacktrace, if needed:

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/discussions/968#discussioncomment-5099512

niyatim23, 2023-02-27:
Need to setup paths similar tothis for M1:
https://github.com/awslabs/amazon-kinesis-video-streams-webrtc-sdk-c/blob/master/.github/workflows/ci.yml#L31

sirknightj, 2023-03-01:
Unfortunately even after setting all of those, still need to run it twice.

niyatim23, 2023-03-01:
I think it is not correctly set then. Those paths may be different for M1. They work for intel and that's why we don't have to run it twice in that case

robert-hanuschke, 2023-08-03:
for me, building twice on M1 Pro did not work. Credits to @timetxt in [#1030](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/1030#issuecomment-1656523459) though:

run this : `cmake .. -DBUILD_GSTREAMER_PLUGIN=TRUE -DBUILD_DEPENDENCIES=OFF`
run this: `cmake .. -DBUILD_GSTREAMER_PLUGIN=TRUE`

that did the trick for me, make could successfully be executed afterwards.

niyatim23, 2023-10-24:
This is no longer an issue. Please refer to the CI build for steps

closed 2023-10-24 by niyatim23

disa6302, 2023-10-24:
In the develop branch.