awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #981: curl timeout issues depending on camera

stormon-force, 2023-03-01:
Hi, we're having some issues with sending up an RTSP stream to KVS. 
* We have a few cameras on site: One camera (model AXIS M3206-LVE) works and the other (model AXIS P1448-LE) does not. Both of them are updated to latest LTM firmware.
* Both are on the same network and under the same firewall rules, the RTSP link is the same. 
* I am using the example ```kvs_sample_app```, with no modifications. 
* Streaming to a KVS instance on us-west-2. 
* This application setup works with a public RTSP test stream (```rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mp4```).
* The RTSP stream works on VLC no matter what camera.

Version: v3.3.1
Commit: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/commit/8867047d8d8f56645009ed73788a57f0a6107772 

Below is a log that shows error we're experiencing when the RTSP link is pointed to the non working camera. Also attached is another log is an example of the same stream URL but with a different camera, that works. Any ideas? I'm not sure what to look for. Considering one camera works and the other doesn't leads me to believe it's a problem with some camera setting, but I'm not sure what to fiddle with. But based on some other issue pages, it may also have something to do with not producing frames for putMedia to execute? Will the basic example work here or will we need to modify it?

Log excerpt here. The error section will continue to repeat indefinetly, until token expires.
```
[INFO ] [01-03-2023 00:21:03:643.863 GMT] Using region: us-west-2
[INFO ] [01-03-2023 00:21:03:643.903 GMT] Using aws credentials for Kinesis Video Streams
[INFO ] [01-03-2023 00:21:03:643.911 GMT] Session token detected.
[INFO ] [01-03-2023 00:21:03:645.027 GMT] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ] [01-03-2023 00:21:03:645.081 GMT] heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ] [01-03-2023 00:21:03:645.091 GMT] heapInitialize(): Creating AIV heap.
[INFO ] [01-03-2023 00:21:03:645.109 GMT] heapInitialize(): Heap is initialized OK
[DEBUG] [01-03-2023 00:21:03:645.260 GMT] getSecurityTokenHandler invoked
[DEBUG] [01-03-2023 00:21:03:645.294 GMT] Refreshing credentials. Force refreshing: 0 Now time is: 1677630063645278575 Expiration: 0
[INFO ] [01-03-2023 00:21:03:645.308 GMT] New credentials expiration is 1677633663
[INFO ] [01-03-2023 00:21:03:645.329 GMT] createDeviceResultEvent(): Create device result event.
[DEBUG] [01-03-2023 00:21:03:645.342 GMT] clientReadyHandler invoked
[DEBUG] [01-03-2023 00:21:03:645.370 GMT] Client is ready
[INFO ] [01-03-2023 00:21:03:645.426 GMT] Creating Kinesis Video Stream StormonTest
[INFO ] [01-03-2023 00:21:03:645.452 GMT] createKinesisVideoStream(): Creating Kinesis Video Stream.
[INFO ] [01-03-2023 00:21:03:645.461 GMT] logStreamInfo(): SDK version: 8867047d8d8f56645009ed73788a57f0a6107772
[DEBUG] [01-03-2023 00:21:03:645.477 GMT] logStreamInfo(): Kinesis Video Stream Info
[DEBUG] [01-03-2023 00:21:03:645.483 GMT] logStreamInfo():      Stream name: StormonTest
[DEBUG] [01-03-2023 00:21:03:645.488 GMT] logStreamInfo():      Streaming type: STREAMING_TYPE_REALTIME
[DEBUG] [01-03-2023 00:21:03:645.494 GMT] logStreamInfo():      Content type: video/h264
[DEBUG] [01-03-2023 00:21:03:645.499 GMT] logStreamInfo():      Max latency (100ns): 600000000
[DEBUG] [01-03-2023 00:21:03:645.505 GMT] logStreamInfo():      Fragment duration (100ns): 20000000
[DEBUG] [01-03-2023 00:21:03:645.510 GMT] logStreamInfo():      Key frame fragmentation: Yes
[DEBUG] [01-03-2023 00:21:03:645.515 GMT] logStreamInfo():      Use frame timecode: Yes
[DEBUG] [01-03-2023 00:21:03:645.520 GMT] logStreamInfo():      Absolute frame timecode: Yes
[DEBUG] [01-03-2023 00:21:03:645.525 GMT] logStreamInfo():      Nal adaptation flags: 0
[DEBUG] [01-03-2023 00:21:03:645.531 GMT] logStreamInfo():      Average bandwith (bps): 4194304
[DEBUG] [01-03-2023 00:21:03:645.536 GMT] logStreamInfo():      Framerate: 25
[DEBUG] [01-03-2023 00:21:03:645.541 GMT] logStreamInfo():      Buffer duration (100ns): 1200000000
[DEBUG] [01-03-2023 00:21:03:645.546 GMT] logStreamInfo():      Replay duration (100ns): 400000000
[DEBUG] [01-03-2023 00:21:03:645.552 GMT] logStreamInfo():      Connection Staleness duration (100ns): 600000000
[DEBUG] [01-03-2023 00:21:03:645.557 GMT] logStreamInfo():      Store Pressure Policy: 1
[DEBUG] [01-03-2023 00:21:03:645.562 GMT] logStreamInfo():      View Overflow Policy: 1
[DEBUG] [01-03-2023 00:21:03:645.567 GMT] logStreamInfo():      Segment UUID: NULL
[DEBUG] [01-03-2023 00:21:03:645.572 GMT] logStreamInfo():      Frame ordering mode: 0
[DEBUG] [01-03-2023 00:21:03:645.578 GMT] logStreamInfo(): Track list
[DEBUG] [01-03-2023 00:21:03:645.583 GMT] logStreamInfo():      Track id: 1
[DEBUG] [01-03-2023 00:21:03:645.588 GMT] logStreamInfo():      Track name: kinesis_video
[DEBUG] [01-03-2023 00:21:03:645.593 GMT] logStreamInfo():      Codec id: V_MPEG4/ISO/AVC
[DEBUG] [01-03-2023 00:21:03:645.598 GMT] logStreamInfo():      Track type: TRACK_INFO_TYPE_VIDEO
[DEBUG] [01-03-2023 00:21:03:645.603 GMT] logStreamInfo():      Track cpd: NULL
[INFO ] [01-03-2023 00:21:03:971.626 GMT] writeHeaderCallback(): RequestId: e3964ea7-515e-44d0-bd35-d1e512013200
[DEBUG] [01-03-2023 00:21:03:971.705 GMT] describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.675190736586E9,"DataRetentionInHours":2,"DeviceName":null,"IngestionConfiguration":null,"KmsKeyId":"arn:aws:kms:us-west-2:600340721340:alias/aws/kinesisvideo","MediaType":null,"Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-west-2:600340721340:stream/StormonTest/1675190736586","StreamName":"StormonTest","Version":"gUDmvDqhXD2UvN2LfaKg"}}
[INFO ] [01-03-2023 00:21:03:972.581 GMT] describeStreamResultEvent(): Describe stream result event.
[WARN ] [01-03-2023 00:21:03:972.606 GMT] describeStreamResult(): Content type returned from the DescribeStream call doesn't match the one specified in the StreamInfo
[INFO ] [01-03-2023 00:21:04:268.965 GMT] writeHeaderCallback(): RequestId: 79edeaf2-5879-4f21-84eb-935b8f1d4c43
[DEBUG] [01-03-2023 00:21:04:269.035 GMT] getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-8e923eff.kinesisvideo.us-west-2.amazonaws.com"}
[INFO ] [01-03-2023 00:21:04:269.873 GMT] getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG] [01-03-2023 00:21:04:269.905 GMT] getStreamingTokenHandler invoked
[DEBUG] [01-03-2023 00:21:04:269.920 GMT] Refreshing credentials. Force refreshing: 1 Now time is: 1677630064269916059 Expiration: 1677633663
[INFO ] [01-03-2023 00:21:04:269.930 GMT] New credentials expiration is 1677633664
[INFO ] [01-03-2023 00:21:04:270.137 GMT] getStreamingTokenResultEvent(): Get streaming token result event.
[DEBUG] [01-03-2023 00:21:04:270.168 GMT] streamReadyHandler invoked
[DEBUG] [01-03-2023 00:21:04:270.253 GMT] Stream is ready
[INFO ] [01-03-2023 00:21:04:279.121 GMT] Streaming from rtsp source
[... 212 lines cut ...]
[DEBUG] [28-02-2023 19:50:23:047.590 GMT] postReadCallback(): Wrote 1025 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [28-02-2023 19:50:23:047.839 GMT] postWriteCallback(): Curl post body write function for stream with handle: StormonTest and upload handle: 0 returned: {"EventType":"BUFFERING","FragmentTimecode":1677613820819,"FragmentNumber":"91343852333182006806763811382970836773498674005"}

[DEBUG] [28-02-2023 19:50:23:047.961 GMT] fragmentAckReceivedHandler invoked
[DEBUG] [28-02-2023 19:50:23:048.023 GMT] Reporting fragment ack received. Ack timecode 1677613820819
[DEBUG] [28-02-2023 19:50:23:068.197 GMT] postReadCallback(): Wrote 828 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [28-02-2023 19:50:23:068.427 GMT] postWriteCallback(): Curl post body write function for stream with handle: StormonTest and upload handle: 0 returned: {"EventType":"PERSISTED","FragmentTimecode":1677613819540,"FragmentNumber":"91343852333182006801812051225829315333083406205"}

[DEBUG] [28-02-2023 19:50:23:068.563 GMT] fragmentAckReceivedHandler invoked
[DEBUG] [28-02-2023 19:50:23:068.623 GMT] Reporting fragment ack received. Ack timecode 1677613819540
[DEBUG] [28-02-2023 19:50:23:088.822 GMT] postReadCallback(): Wrote 862 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [28-02-2023 19:50:23:149.325 GMT] postReadCallback(): Wrote 822 bytes to Kinesis Video. Upload stream handle: 0
[... cut ...]
[DEBUG] [28-02-2023 19:50:23:553.162 GMT] postReadCallback(): Wrote 709 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [28-02-2023 19:50:23:573.542 GMT] postReadCallback(): Wrote 710 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [28-02-2023 19:50:23:634.056 GMT] postReadCallback(): Wrote 624 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [28-02-2023 19:50:23:694.552 GMT] postReadCallback(): Wrote 1298 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [28-02-2023 19:50:23:755.060 GMT] postReadCallback(): Wrote 592 bytes to Kinesis Video. Upload stream handle: 0
```

MushMal, 2023-03-01:
There is little information available in the logs. Here is an interesting part:

[INFO ] [01-03-2023 00:21:06:498.228 GMT] writeHeaderCallback(): RequestId: cd3b8359-e463-eeea-9143-1febc07e745c
[DEBUG] [01-03-2023 00:21:07:118.992 GMT] postReadCallback(): Pausing CURL read for upload handle: 0
[WARN ] [01-03-2023 00:21:36:499.120 GMT] curlCompleteSync(): curl perform failed for url https://s-8e923eff.kinesisvideo.us-west-2.amazonaws.com/putMedia with result Timeout was reached: Operation too slow. Less than 30 bytes/sec transferred the last 30 seconds
[WARN ] [01-03-2023 00:21:36:499.363 GMT] curlCompleteSync(): HTTP Error 0 : Response: (null)

There is a gap of 30 seconds which is the default timeout for the connection. 

If I were to guess, you have an issue with the camera not producing any frames or the RTSP link is broken.

Suggest removing KVS from the pipeline and stabilizing the camera/RTSP link with a fakesink and then plug in the kvs.

stormon-force, 2023-03-01:
Thanks for the fast response! Is there a way I can get you more information?

I tried this with fakesink using the command
```gst-launch-1.0 -v rtspsrc \
location=<RTSP LINK> \
short-header=TRUE ! rtph264depay ! video/x-h264, format=avc,alignment=au ! fakesink
```
The RTSP link works using VLC, or at least as much as I can visually confirm (no significant frames dropping). What could be different between VLC and the producer example app? 
I'm not sure how to parse these logs. I tried this same command with the public test RTSP link and it was similar.
Log here
```
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Progress: (open) Opening Stream
Pipeline is PREROLLED ...
Prerolled, waiting for progress to finish...
Progress: (connect) Connecting to <RTSP LINK>
Progress: (open) Retrieving server options
Progress: (open) Retrieving media info
Progress: (request) SETUP stream 0
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstRtpBin:manager: latency = 2000
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstRtpBin:manager: ntp-sync = false
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstRtpBin:manager: rfc7273-sync = false
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstRtpBin:manager: ntp-time-source = ntp
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstRtpBin:manager: drop-on-latency = false
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstRtpBin:manager: max-rtcp-rtp-time-diff = 1000
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstRtpBin:manager: max-ts-offset-adjustment = 0
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstRtpBin:manager: buffer-mode = slave
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstUDPSrc:udpsrc0: timeout = 5000000000
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstUDPSrc:udpsrc0: caps = application/x-rtp, media=(string)audio, payload=(int)96, clock-rate=(int)12000, encoding-name=(string)MPEG4-GENERIC, encoding-params=(string)2, profile-level-id=(string)1, mode=(string)AAC-hbr, sizelength=(string)13, indexlength=(string)3, indexdeltalength=(string)3, config=(string)149056e500, a-sdplang=(string)en, ssrc=(uint)1633245032
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstUDPSrc:udpsrc1: caps = application/x-rtcp
Progress: (request) SETUP stream 1
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstUDPSrc:udpsrc3: timeout = 5000000000
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstUDPSrc:udpsrc3: caps = application/x-rtp, media=(string)video, payload=(int)97, clock-rate=(int)90000, encoding-name=(string)H264, packetization-mode=(string)1, profile-level-id=(string)64000C, sprop-parameter-sets=(string)"Z2QADKzZQ8Vv/ACAAGxAAAADAEAAAAwDxQplgA\=\=\,aOvssiw\=", a-framesize=(string)240-160, a-sdplang=(string)en, a-cliprect=(string)"0\,0\,160\,240", a-framerate=(string)24.0, ssrc=(uint)195801152
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstUDPSrc:udpsrc4: caps = application/x-rtcp
Progress: (open) Opened Stream
Setting pipeline to PLAYING ...
New clock: GstSystemClock
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstRtpBin:manager: buffer-mode = slave
Progress: (request) Sending PLAY request
Redistribute latency...
Redistribute latency...
Progress: (request) Sending PLAY request
Redistribute latency...
Redistribute latency...
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstUDPSrc:udpsrc0: caps = application/x-rtp, media=(string)audio, payload=(int)96, clock-rate=(int)12000, encoding-name=(string)MPEG4-GENERIC, encoding-params=(string)2, profile-level-id=(string)1, mode=(string)AAC-hbr, sizelength=(string)13, indexlength=(string)3, indexdeltalength=(string)3, config=(string)149056e500, a-sdplang=(string)en, ssrc=(uint)1633245032, clock-base=(uint)0, seqnum-base=(uint)1, npt-start=(guint64)0, npt-stop=(guint64)634625000000, play-speed=(double)1, play-scale=(double)1, onvif-mode=(boolean)false
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstUDPSrc:udpsrc3: caps = application/x-rtp, media=(string)video, payload=(int)97, clock-rate=(int)90000, encoding-name=(string)H264, packetization-mode=(string)1, profile-level-id=(string)64000C, sprop-parameter-sets=(string)"Z2QADKzZQ8Vv/ACAAGxAAAADAEAAAAwDxQplgA\=\=\,aOvssiw\=", a-framesize=(string)240-160, a-sdplang=(string)en, a-cliprect=(string)"0\,0\,160\,240", a-framerate=(string)24.0, ssrc=(uint)195801152, clock-base=(uint)0, seqnum-base=(uint)1, npt-start=(guint64)0, npt-stop=(guint64)634625000000, play-speed=(double)1, play-scale=(double)1, onvif-mode=(boolean)false
Progress: (request) Sent PLAY request
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstUDPSrc:udpsrc0.GstPad:src: caps = application/x-rtp, media=(string)audio, payload=(int)96, clock-rate=(int)12000, encoding-name=(string)MPEG4-GENERIC, encoding-params=(string)2, profile-level-id=(string)1, mode=(string)AAC-hbr, sizelength=(string)13, indexlength=(string)3, indexdeltalength=(string)3, config=(string)149056e500, a-sdplang=(string)en, ssrc=(uint)1633245032, clock-base=(uint)0, seqnum-base=(uint)1, npt-start=(guint64)0, npt-stop=(guint64)634625000000, play-speed=(double)1, play-scale=(double)1, onvif-mode=(boolean)false
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstUDPSrc:udpsrc0.GstPad:src: caps = application/x-rtp, media=(string)audio, payload=(int)96, clock-rate=(int)12000, encoding-name=(string)MPEG4-GENERIC, encoding-params=(string)2, profile-level-id=(string)1, mode=(string)AAC-hbr, sizelength=(string)13, indexlength=(string)3, indexdeltalength=(string)3, config=(string)149056e500, a-sdplang=(string)en, ssrc=(uint)1633245032, clock-base=(uint)0, seqnum-base=(uint)1, npt-start=(guint64)0, npt-stop= [... cut ...] ount\=\(uint\)0\,\ recv-packet-rate\=\(uint\)0\,\ have-sr\=\(boolean\)false\,\ sr-ntptime\=\(guint64\)0\,\ sr-rtptime\=\(uint\)0\,\ sr-octet-count\=\(uint\)0\,\ sr-packet-count\=\(uint\)0\;" >, rtx-count=(uint)0, recv-rtx-req-count=(uint)0, sent-rtx-req-count=(uint)0;
^Chandling interrupt. (23.1 %)
Interrupt: Stopping pipeline ...
Execution ended after 0:02:27.158819897
Setting pipeline to NULL ...
Freeing pipeline ...
```

I will also mention we are running this on an EC2 instance on us-east-1.

MushMal, 2023-03-01:
It seems that fakesink is getting some frames. I am wondering whether kvssink drops the frames at the source or something. Perhaps you could run kvssink with GSTDEBUG=4 or 5 which could give more information. I would also try to search the issues whether something similar related to EC2 and rtsp stream have been answered in the context of gstreamer.

stormon-force, 2023-03-01:
I think I may of misformatted the command, but here is the log. The log appears to be the same whether I use the public RTSP stream or our RTSP stream link, which is contrary to what I have with the sample app.

Could there be a resource constraint issue? This EC2 instance is type t2.micro.

Command
```
[gst_launch_log_fulldebug.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/10865225/gst_launch_log_fulldebug.txt)

sudo GST_DEBUG=5 GST_PLUGIN_PATH=<PATH> LD_LIBRARY_PATH=<PATH> \
gst-launch-1.0 -v rtspsrc \
location=<RTSP LINK> \
short-header=TRUE ! rtph264depay ! video/x-h264, format=avc,alignment=au ! kvssink stream-name="StormonTest" storage-size=512
```

stormon-force, 2023-03-01:
I think I may of misformatted the command, but here is the log: [gst_launch_log_fulldebug.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/10865234/gst_launch_log_fulldebug.txt). The log appears to be the same whether I use the public RTSP stream or our RTSP stream link, which is contrary to what I have with the sample app.

Could there be a resource constraint issue? This EC2 instance is type t2.micro.

Command
```
sudo GST_DEBUG=5 GST_PLUGIN_PATH=<PATH> LD_LIBRARY_PATH=<PATH> \
gst-launch-1.0 -v rtspsrc \
location=<RTSP LINK> \
short-header=TRUE ! rtph264depay ! video/x-h264, format=avc,alignment=au ! kvssink stream-name="StormonTest" storage-size=512
```

MushMal, 2023-03-01:
It's hard to parse the output but it seems that there is a failure in the caps negotiation. I would recommend looking at the configuration of the stream and see some of the KVS documentation on it. I am not very familiar with it. If the two streams (working and not working) have the same configuration then I don't really understand what's going on. Perhaps someone from AWS can help.

stormon-force, 2023-03-01:
Ok I'll see about getting assistance from them. When you say configuration of the stream do you mean the KVS instance or the sample app generating the pipeline? The two streams are the same RTSP URL, just changing some ports to direct it to a different camera. Would you also be able to describe what part of the process is not completing? I will link this issue, but I'd like to get a basic understanding of what's happening.

MushMal, 2023-03-01:
In this case it would be settings for the encoder and packager - the bitrate, q-values, encoding profile, resolution, frame rate. Making sure the GoP size is up-to 10 seconds long. Not much else comes to mind. From the logs I can see that KVS loads up and configures which means that it doesn't have any issues with memory pressure nor connectivity.

stormon-force, 2023-03-02:
I was also looking for the source of functions like ```writeHeaderCallback()``` and ```postReadCallback()```, are these part of the GStreamer framework or KVS API? I was having trouble finding them in repository.

MushMal, 2023-03-02:
These are defined in C producer layer https://github.com/awslabs/amazon-kinesis-video-streams-producer-c/blob/master/src/source/Response.c#L608

You might want to put a debug statement in the put frame call in PIC which is this: https://github.com/awslabs/amazon-kinesis-video-streams-pic/blob/master/src/client/src/Stream.c#L734 - see if the call is actually made there.

Also, trace the call in the KVS sink here: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/src/gstreamer/gstkvssink.cpp#L1153 see if gstreamer makes a buffer handling call and whether the buffer data is the frame data (as kvs plugin deems).

Sorry I couldn't be of more help..