awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #982: SDK Stuck

Alex-Wenner-FHR, 2023-03-06:
I have published the contianer to an ECS cluster. I have confirmed everything is good (credentials, url, permissions, etc.) and When the container boots up it appears to just stall here: 

```python

2023-03-06T10:53:09.867-06:00	Progress: (open) Retrieving server options

2023-03-06T10:53:09.915-06:00	Progress: (open) Retrieving media info

2023-03-06T10:53:10.137-06:00	Progress: (request) SETUP stream 0

2023-03-06T10:53:10.183-06:00	Progress: (open) Opened Stream

2023-03-06T10:53:10.183-06:00	Setting pipeline to PLAYING ...

2023-03-06T10:53:10.183-06:00	New clock: GstSystemClock

2023-03-06T10:53:10.183-06:00	Progress: (request) Sending PLAY request

2023-03-06T10:53:10.184-06:00	Progress: (request) Sending PLAY request

2023-03-06T10:53:10.335-06:00	Progress: (request) Sent PLAY request
```
 
Nothing is coming into Kinesis. When I run this contianer locally, with the exact config it does not get stuck here. ECS Container Mem & CPU look to be fine. 

Any thoughts on what could be happening?

Alex-Wenner-FHR, 2023-03-06:
@disa6302 @niyatim23 adding you for visibility

Alex-Wenner-FHR, 2023-03-06:
To follow up, here are the logs I see when executing the same container build locally:

```python
Progress: (request) Sending PLAY request
Progress: (request) Sending PLAY request
Progress: (request) Sent PLAY request
WARNING: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0: Could not read from resource.
Additional debug info:
gstrtspsrc.c(5427): gst_rtspsrc_reconnect (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0:
Could not receive any UDP packets for 5.0000 seconds, maybe your firewall is blocking it. Retrying using a tcp connection.
[INFO ] [06-03-2023 17:17:02:096.367 GMT] kinesisVideoStreamFormatChanged(): Stream format changed.
[INFO ] [06-03-2023 17:17:02:960.802 GMT] putStreamResultEvent(): Put stream result event. New upload handle 0
[DEBUG] [06-03-2023 17:17:02:971.928 GMT] Kinesis Video client and stream metrics
        >> Overall storage byte size: 536870912
        >> Available storage byte size: 536158026
        >> Allocated storage byte size: 712886
        >> Total view allocation byte size: 172880
        >> Total streams elementary frame rate (fps): 30
        >> Total streams transfer rate (bps): 33554432 (32768 Kbps)
        >> Current view duration (ms): 0
        >> Overall view duration (ms): 0
        >> Current view byte size: 712838
        >> Overall view byte size: 712838
        >> Current elementary frame rate (fps): 30
        >> Current transfer rate (bps): 33554432 (32768 Kbps)
[DEBUG] [06-03-2023 17:17:03:851.510 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
[INFO ] [06-03-2023 17:17:03:852.244 GMT] writeHeaderCallback(): RequestId: db8f3bb3-7917-14f6-870f-19fce9991fa7
[DEBUG] [06-03-2023 17:17:03:852.654 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [06-03-2023 17:17:03:853.323 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [06-03-2023 17:17:03:854.438 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [06-03-2023 17:17:03:854.688 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [06-03-2023 17:17:03:854.978 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [06-03-2023 17:17:03:856.494 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [06-03-2023 17:17:03:857.620 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [06-03-2023 17:17:03:858.211 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [06-03-2023 17:17:03:860.476 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [06-03-2023 17:17:03:860.926 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [06-03-2023 17:17:04:040.276 GMT] postWriteCallback(): Curl post body write function for stream with handle: Axis-Rail-100 and upload handle: 0 returned: {"EventType":"BUFFERING","FragmentTimecode":1678123022960,"FragmentNumber":"91343852333181447247962534076940227787841398098"}

[DEBUG] [06-03-2023 17:17:04:040.375 GMT] fragmentAckReceivedHandler invoked
[DEBUG] [06-03-2023 17:17:04:249.114 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [06-03-2023 17:17:04:249.605 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [06-03-2023 17:17:04:251.687 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [06-03-2023 17:17:04:252.518 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [06-03-2023 17:17:04:254.430 GMT] postReadCallback(): Wrote 42636 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [06-03-2023 17:17:04:559.635 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [06-03-2023 17:17:04:560.066 GMT] postReadCallback(): Wrote 11068 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [06-03-2023 17:17:04:866.699 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [06-03-2023 17:17:04:867.120 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [06-03-2023 17:17:04:867.559 GMT] postReadCallback(): Wrote 43938 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [06-03-2023 17:17:05:490.365 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [06-03-2023 17:17:05:491.157 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [06-03-2023 17:17:05:491.486 GMT] postReadCallback(): Wrote 32117 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [06-03-2023 17:17:05:633.253 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [06-03-2023 17:17:05:633.684 GMT] postReadCallback(): Wrote 38405 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [06-03-2023 17:17:05:695.903 GMT] postReadCallback(): Wrote 55110 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [06-03-2023 17:17:06:320.539 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [06-03-2023 17:17:06:321.631 GMT] postReadCallback(): Wrote 16425 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [06-03-2023 17:17:06:383.525 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [06-03-2023 17:17:06:384.110 GMT] postReadCallback(): Wrote 33004 bytes to Kinesis Video. Upload stream handle: 0
^Chandling interrupt.
```

Alex-Wenner-FHR, 2023-03-06:
Follow up questions, is there any details on what a VPC _needs_ in able to function properly? Feels like an AWS Environment issue being that the same contianer works locally.

Alex-Wenner-FHR, 2023-03-07:
GST_DEBUG=4 Logs: 

```python

Progress: (open) Opened Stream
--
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Progress: (request) Sending PLAY request
Progress: (request) Sending PLAY request
0:00:00.687806869 [332m    1[00m       0xa7fca0 [36mINFO   [00m [00;01;31m          GST_STATES gstelement.c:2394:gst_element_continue_state:<fakesrc0>[00m committing state from READY to PAUSED, pending PLAYING, next PLAYING
0:00:00.687806869 [332m    1[00m       0xa7fca0 [36mINFO   [00m [00;01;31m          GST_STATES gstelement.c:2324:_priv_gst_element_state_changed:<fakesrc0>[00m notifying about state-changed READY to PAUSED (PLAYING pending)
0:00:00.687880220 [332m    1[00m 0x7f394c0478f0 [36mINFO   [00m [00;01;37;41m    GST_ELEMENT_PADS gstelement.c:913:gst_element_get_static_pad:[00m no such pad 'sink' in element "fakesrc0"
0:00:00.687914988 [332m    1[00m 0x7f394c0478f0 [32;01mFIXME  [00m [00;04m             default gstutils.c:3826:gst_pad_create_stream_id_internal:<fakesrc0:src>[00m Creating random stream-id, consider implementing a deterministic way of creating a stream-id
0:00:00.687914988 [332m    1[00m       0xa7fca0 [36mINFO   [00m [00;01;31m          GST_STATES gstelement.c:2401:gst_element_continue_state:<fakesrc0>[00m continue state change PAUSED to PLAYING, final PLAYING
0:00:00.687914988 [332m    1[00m       0xa7fca0 [36mINFO   [00m [00;01;31m          GST_STATES gstelement.c:2419:gst_element_continue_state:<fakesrc0>[00m completed state change to PLAYING
0:00:00.687914988 [332m    1[00m       0xa7fca0 [36mINFO   [00m [00;01;31m          GST_STATES gstelement.c:2324:_priv_gst_element_state_changed:<fakesrc0>[00m notifying about state-changed PAUSED to PLAYING (VOID_PENDING pending)
0:00:00.688000611 [332m    1[00m 0x7f394c0478f0 [36mINFO   [00m [00;01;34m           GST_EVENT gstevent.c:889:gst_event_new_segment:[00m creating segment event bytes segment start=0, offset=0, stop=-1, rate=1.000000, applied_rate=1.000000, flags=0x00, time=0, base=0, position 0, duration -1
0:00:00.688027715 [332m    1[00m 0x7f394c0478f0 [36mINFO   [00m [00m             basesrc gstbasesrc.c:2847:gst_base_src_loop:<fakesrc0>[00m marking pending DISCONT
0:00:00.688199215 [332m    1[00m 0x7f394c0478f0 [36mINFO   [00m [00m             basesrc gstbasesrc.c:2743:gst_base_src_loop:<fakesrc0>[00m pausing after gst_base_src_get_range() = eos
0:00:00.688235335 [332m    1[00m 0x7f394c0478f0 [36mINFO   [00m [00m                task gsttask.c:318:gst_task_func:<fakesrc0:src>[00m Task going to paused
0:00:00.812082310 [332m    1[00m       0xa7fca0 [36mINFO   [00m [00;01;31m          GST_STATES gstbin.c:2521:gst_bin_element_set_state:<rtpssrcdemux0>[00m current PAUSED pending VOID_PENDING, desired next PLAYING
0:00:00.812200033 [332m    1[00m       0xa7fca0 [36mINFO   [00m [00;01;31m          GST_STATES gstelement.c:2419:gst_element_continue_state:<rtpssrcdemux0>[00m completed state change to PLAYING
0:00:00.812213985 [332m    1[00m       0xa7fca0 [36mINFO   [00m [00;01;31m          GST_STATES gstelement.c:2324:_priv_gst_element_state_changed:<rtpssrcdemux0>[00m notifying about state-changed PAUSED to PLAYING (VOID_PENDING pending)
0:00:00.812325452 [332m    1[00m       0xa7fca0 [36mINFO   [00m [00;01;31m          GST_STATES gstbin.c:2969:gst_bin_change_state_func:<manager>[00m child 'rtpssrcdemux0' changed state to 4(PLAYING) successfully
0:00:00.812354578 [332m    1[00m       0xa7fca0 [36mINFO   [00m [00;01;31m          GST_STATES gstbin.c:2521:gst_bin_element_set_state:<rtpsession0>[00m current PAUSED pending VOID_PENDING, desired next PLAYING
0:00:00.812454744 [332m    1[00m       0xa7fca0 [36mINFO   [00m [00;01;31m          GST_STATES gstelement.c:2419:gst_element_continue_state:<rtpsession0>[00m completed state change to PLAYING
0:00:00.812461252 [332m    1[00m       0xa7fca0 [36mINFO   [00m [00;01;31m          GST_STATES gstelement.c:2324:_priv_gst_element_state_changed:<rtpsession0>[00m notifying about state-changed PAUSED to PLAYING (VOID_PENDING pending)
0:00:00.812540562 [332m    1[00m       0xa7fca0 [36mINFO   [00m [00;01;31m          GST_STATES gstbin.c:2969:gst_bin_change_state_func:<manager>[00m child 'rtpsession0' changed state to 4(PLAYING) successfully
0:00:00.812547880 [332m    1[00m       0xa7fca0 [36mINFO   [00m [00;01;31m          GST_STATES gstelement.c:2419:gst_element_continue_state:<manager>[00m completed state change to PLAYING
0:00:00.812567295 [332m    1[00m       0xa7fca0 [36mINFO   [00m [00;01;31m          GST_STATES gstelement.c:2324:_priv_gst_element_state_changed:<manager>[00m notifying about state-changed PAUSED to PLAYING (VOID_PENDING pending)
0:00:00.812653441 [332m    1[00m       0xa7fca0 [36mINFO   [00m [00m             basesrc gstbasesrc.c:1343:gst_base_src_do_seek:<udpsrc1>[00m seeking: time segment start=0:00:00.000000000, offset=0:00:00.000000000, stop=99:99:99.999999999, rate=1.000000, applied_rate=1.000000, flags=0x00, time=0:00:00.000000000, base=0:00:00.000000000, position 0:00:00.000000000, duration 99:99:99.999999999
0:00:00.812714158 [332m    1[00m       0xa7fca0 [36mINFO   [00m [00m                task gsttask.c:456:gst_task_set_lock:[00m setting stream lock 0xa5b4f0 on task 0xa5c710
[... cut ...]
0:00:05.492519512 [332m    1[00m 0x7f394c0479e0 [33;01mWARN   [00m [00m     rtpjitterbuffer gstrtpjitterbuffer.c:2960:gst_rtp_jitter_buffer_chain:<rtpjitterbuffer0>[00m 141 pending timers > 120 - resetting
0:00:06.448855233 [332m    1[00m 0x7f394c0479e0 [36mINFO   [00m [00m     rtpjitterbuffer rtpjitterbuffer.c:778:rtp_jitter_buffer_calculate_pts:[00m resync to time 0:00:05.762485567, rtptime 4:56:57.078877777
0:00:08.450885123 [332m    1[00m 0x7f39380036d0 [36mINFO   [00m [00m     rtpjitterbuffer gstrtpjitterbuffer.c:3800:do_deadline_timeout:<rtpjitterbuffer0>[00m got deadline timeout

```

Alex-Wenner-FHR, 2023-03-08:
This has been resolved. After chatting with AWS KVS support, the issue that I believe is going on here is that it seems UDP connection is a known issue within a container. After forcing TCP by using `do-rtcp=TRUE rtspt://...` params in the gstreamer command seen [here](https://urldefense.com/v3/__https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/397*issuecomment-610191954__;Iw!!HKYIif90!zFUwaKaarZUgzTkVElmty8jfYd0YkUO45vsK97xag5ndN3ee6Mf2Fkx8LKBw4jCF9ucR2WmJC72S_3_a-XTreTM$) fixed the issue!

closed 2023-03-08 by Alex-Wenner-FHR

disa6302, 2023-03-09:
Glad it worked for you! As always, feel free to reach out if you have any more questions.