awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #991: [BUG] All frames are dropped for h264

Rizhiy, 2023-03-25:
**Logging**
```
Setting pipeline to PAUSED ...
log4cplus:ERROR could not open file ../kvs_log_configuration
INFO - createKinesisVideoClient(): Creating Kinesis Video Client
2023-03-25 01:47:58 [140025934083904] INFO - heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
2023-03-25 01:47:58 [140025934083904] INFO - heapInitialize(): Creating AIV heap.
2023-03-25 01:47:58 [140025934083904] INFO - heapInitialize(): Heap is initialized OK
2023-03-25 01:47:58 [140025934083904] DEBUG - getSecurityTokenHandler invoked
2023-03-25 01:47:58 [140025934083904] DEBUG - Refreshing credentials. Force refreshing: 0 Now time is: 1679708878570485134 Expiration: 0
2023-03-25 01:47:58 [140025934083904] INFO - createDeviceResultEvent(): Create device result event.
2023-03-25 01:47:58 [140025934083904] DEBUG - clientReadyHandler invoked
2023-03-25 01:47:58 [140025934083904] INFO - try creating stream
2023-03-25 01:47:58 [140025934083904] INFO - Creating Kinesis Video Stream test-stream
2023-03-25 01:47:58 [140025934083904] INFO - createKinesisVideoStream(): Creating Kinesis Video Stream.
2023-03-25 01:47:58 [140025934083904] INFO - logStreamInfo(): SDK version: 4abb7cd5e78098e1e5c7e217f77b61e9911e272a
2023-03-25 01:47:58 [140025934083904] DEBUG - logStreamInfo(): Kinesis Video Stream Info
2023-03-25 01:47:58 [140025934083904] DEBUG - logStreamInfo():  Stream name: test-stream
2023-03-25 01:47:58 [140025934083904] DEBUG - logStreamInfo():  Streaming type: STREAMING_TYPE_REALTIME
2023-03-25 01:47:58 [140025934083904] DEBUG - logStreamInfo():  Content type: video/h264
2023-03-25 01:47:58 [140025934083904] DEBUG - logStreamInfo():  Max latency (100ns): 600000000
2023-03-25 01:47:58 [140025934083904] DEBUG - logStreamInfo():  Fragment duration (100ns): 20000000
2023-03-25 01:47:58 [140025934083904] DEBUG - logStreamInfo():  Key frame fragmentation: Yes
2023-03-25 01:47:58 [140025934083904] DEBUG - logStreamInfo():  Use frame timecode: Yes
2023-03-25 01:47:58 [140025934083904] DEBUG - logStreamInfo():  Absolute frame timecode: Yes
2023-03-25 01:47:58 [140025934083904] DEBUG - logStreamInfo():  Nal adaptation flags: 0
2023-03-25 01:47:58 [140025934083904] DEBUG - logStreamInfo():  Average bandwith (bps): 4194304
2023-03-25 01:47:58 [140025934083904] DEBUG - logStreamInfo():  Framerate: 25
2023-03-25 01:47:58 [140025934083904] DEBUG - logStreamInfo():  Buffer duration (100ns): 1200000000
2023-03-25 01:47:58 [140025934083904] DEBUG - logStreamInfo():  Replay duration (100ns): 400000000
2023-03-25 01:47:58 [140025934083904] DEBUG - logStreamInfo():  Connection Staleness duration (100ns): 600000000
2023-03-25 01:47:58 [140025934083904] DEBUG - logStreamInfo():  Store Pressure Policy: 1
2023-03-25 01:47:58 [140025934083904] DEBUG - logStreamInfo():  View Overflow Policy: 1
2023-03-25 01:47:58 [140025934083904] DEBUG - logStreamInfo():  Segment UUID: NULL
2023-03-25 01:47:58 [140025934083904] DEBUG - logStreamInfo():  Frame ordering mode: 0
2023-03-25 01:47:58 [140025934083904] DEBUG - logStreamInfo(): Track list
2023-03-25 01:47:58 [140025934083904] DEBUG - logStreamInfo():  Track id: 1
2023-03-25 01:47:58 [140025934083904] DEBUG - logStreamInfo():  Track name: kinesis_video
2023-03-25 01:47:58 [140025934083904] DEBUG - logStreamInfo():  Codec id: V_MPEG4/ISO/AVC
2023-03-25 01:47:58 [140025934083904] DEBUG - logStreamInfo():  Track type: TRACK_INFO_TYPE_VIDEO
2023-03-25 01:47:58 [140025934083904] DEBUG - logStreamInfo():  Track cpd: NULL
2023-03-25 01:47:58 [140025934083904] INFO - writeHeaderCallback(): RequestId: c5cd1b71-244d-4cff-a78e-bbe3c26f7185
2023-03-25 01:47:58 [140025746413120] DEBUG - describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.677106777651E9,"DataRetentionInHours":720,"DeviceName":null,"IngestionConfiguration":null,"KmsKeyId":"arn:aws:kms:eu-west-1:108686862818:a
lias/aws/kinesisvideo","MediaType":null,"Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:eu-west-1:108686862818:stream/test-stream/1677106777651","StreamName":"test-stream","Version":"otr4buTHDSb5KnNZAHlb"}}
2023-03-25 01:47:58 [140025746413120] INFO - describeStreamResultEvent(): Describe stream result event.
2023-03-25 01:47:58 [140025746413120] WARN - describeStreamResult(): Retention period returned from the DescribeStream call doesn't match the one specified in the StreamInfo
2023-03-25 01:47:58 [140025746413120] WARN - describeStreamResult(): Content type returned from the DescribeStream call doesn't match the one specified in the StreamInfo
2023-03-25 01:47:58 [140025746413120] INFO - writeHeaderCallback(): RequestId: 79897b24-3830-4160-b171-ecc14c323330
2023-03-25 01:47:58 [140025664632384] DEBUG - getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-5ac677c0.kinesisvideo.eu-west-1.amazonaws.com"}
2023-03-25 01:47:58 [140025664632384] INFO - getStreamingEndpointResultEvent(): Get streaming endpoint result event.
2023-03-25 01:47:58 [140025664632384] DEBUG - getStreamingTokenHandler invoked
2023-03-25 01:47:58 [140025664632384] DEBUG - Refreshing credentials. Force refreshing: 1 Now time is: 1679708878744400484 Expiration: 18446744073709551615
2023-03-25 01:47:58 [140025664632384] INFO - getStreamingTokenResultEvent(): Get streaming token result event.
2023-03-25 01:47:58 [140025664632384] DEBUG - streamReadyHandler invoked
2023-03-25 01:47:58 [140025664632384] Stream is ready
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
/GstPipeline:pipeline0/GstV4l2Src:v4l2src0.GstPad:src: caps = video/x-h264, width=(int)1920, height=(int)1080, stream-format=(string)byte-stream, alignment=(string)au, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, interlace-mode=(string)progressive, colorimetr
y=(string)2:4:7:1
/GstPipeline:pipeline0/GstH264Parse:h264parse0.GstPad:sink: caps = video/x-h264, width=(int)1920, height=(int)1080, stream-format=(string)byte-stream, alignment=(string)au, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, interlace-mode=(string)progressive, color
imetry=(string)2:4:7:1
/GstPipeline:pipeline0/GstH264Parse:h264parse0.GstPad:src: caps = video/x-h264, width=(int)1920, height=(int)1080, stream-format=(string)avc, alignment=(string)au, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, interlace-mode=(string)progressive, colorimetry=(s
tring)2:4:7:1, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true, profile=(string)main, level=(string)4.1, codec_data=(buffer)014d0029ffe10019674d00298d8d403c0113f2cdc0404050000070800015f9004001000468ee3c80
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = video/x-h264, width=(int)1920, height=(int)1080, stream-format=(string)avc, alignment=(string)au, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, interlace-mode=(string)progressive, colorimetry=
(string)2:4:7:1, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true, profile=(string)main, level=(string)4.1, codec_data=(buffer)014d0029ffe10019674d00298d8d403c0113f2cdc0404050000070800015f9004001000468ee3c80
INFO - kinesisVideoStreamFormatChanged(): Stream format changed.
2023-03-25 01:47:58 [140025664632384] /GstPipeline:pipeline0/GstKvsSink:sink.GstPad:video_0: caps = video/x-h264, width=(int)1920, height=(int)1080, stream-format=(string)avc, alignment=(string)au, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, interlace-mode=(
string)progressive, colorimetry=(string)2:4:7:1, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true, profile=(string)main, level=(string)4.1, codec_data=(buffer)014d0029ffe10019674d00298d8d403c0113f2cdc0404050000070800015f900
[... 20 lines cut ...]
When I try to stream from h264 directly, it drops all frames.

**To Reproduce**
Build on Ubuntu 22.04, run command from FAQ (`gst-launch-1.0 -v v4l2src device=/dev/video0 ! h264parse ! video/x-h264,stream-format=avc,alignment=au ! kvssink stream-name="$STREAM_NAME" access-key="$AWS_ACCESS_KEY_ID" secret-key="$AWS_SECRET_ACCESS_KEY"`)

**Expected behavior**
Stream propperly

**SDK version number**
v3.3.1

**Open source building**
Build went fine

**Screenshots**
N/A

**Platform (please complete the following information):**
 - OS: Ubuntu
 - Processor: AMD Ryzen 9 5950X
 - Version 22.04

**Additional context**
The stream from `video/x-raw` works fine, but then I'm limited to either low resolution or low framerate.
It was working fine on Ubuntu 20.04

sirknightj, 2023-03-25:
Hi @Rizhiy,

Could you please provide the working gstreamer pipeline with low resolution/framerate?

Also, if you're okay with it, could you share your camera's output?
```
gst-discoverer-1.0 rtsp://your_rtsp_url
```

disa6302, 2023-04-01:
Closing due to no response. Feel free to reopen if you have more details.

closed 2023-04-01 by disa6302