awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #993: Using the develop branch to build on M1, but log4cplus headers have deprecated auto_ptr

jkartik, 2023-04-03:
I am using the develop branch as suggested from this discussion below. This helped me build the dependencies. However during make, I run into this compile error

```
amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/include/log4cplus/appender.h:186:43: error: no template named 'auto_ptr' in namespace 'std'
        virtual void setErrorHandler(std::auto_ptr<ErrorHandler> eh);
                                     ~~~~~^

```
auto_ptr is deprecated with cxx17. How do I work around this issue? Any ideas?

### Discussed in https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/discussions/968

<div type='discussions-op-text'>

<sup>Originally posted by **Rizhiy** February 22, 2023</sup>
I want stream my webcam to KVS, this repo was one of recommendations I came across.
I tried to build following the instructions, but got an error.
I don't develop in C/C++, so don't really understand what is missing.

I'm on Apple M1 Pro, macOS Monterey 12.6.3

** Logging **
```
[ 66%] Performing build step for 'project_libopenssl'
In file included from apps/app_rand.c:10:
In file included from apps/apps.h:13:
In file included from ./e_os.h:16:
In file included from include/openssl/e_os2.h:243:
/Library/Developer/CommandLineTools/usr/lib/clang/14.0.0/include/inttypes.h:21:15: fatal error: 'inttypes.h' file not found
#include_next <inttypes.h>
              ^~~~~~~~~~~~
1 error generated.
make[4]: *** [apps/app_rand.o] Error 1
make[3]: *** [all] Error 2
make[2]: *** [build/src/project_libopenssl-stamp/project_libopenssl-build] Error 2
make[1]: *** [CMakeFiles/project_libopenssl.dir/all] Error 2
make: *** [all] Error 2
CMake Error at dependency/libkvscproducer/kvscproducer-src/CMake/Utilities.cmake:93 (message):
  CMake step for libopenssl failed: 2
Call Stack (most recent call first):
  dependency/libkvscproducer/kvscproducer-src/CMakeLists.txt:69 (build_dependency)
```

** Any design considerations/constraints **
I just want to stream webcam to KVS for a demo.
</div>

disa6302, 2023-04-03:
Have you tried clean build? If your open-source folder has the wrong version of log4cplus, it is likely for this to happen. We did update the log4cplus version in develop. 

You need to remove the dependency, open-source and build folders from project root and re-build

jkartik, 2023-04-03:
Excellent, that worked! Thanks.