awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #998: [BUG] Missing codec private data error

quintene, 2023-04-12:
**Logging**

INFO - createKinesisVideoClient(): Creating Kinesis Video Client
2023-04-12 16:58:28 [547737588768] INFO - heapInitialize(): Initializing native heap with limit size 1077936128, spill ratio 0% and flags 0x00000001
2023-04-12 16:58:28 [547737588768] INFO - heapInitialize(): Creating AIV heap.
2023-04-12 16:58:28 [547737588768] INFO - heapInitialize(): Heap is initialized OK
2023-04-12 16:58:28 [547737588768] DEBUG - getSecurityTokenHandler invoked
2023-04-12 16:58:28 [547737588768] DEBUG - Refreshing credentials. Force refreshing: 0 Now time is: 1681311508561040558 Expiration: 0
2023-04-12 16:58:28 [547737588768] INFO - createDeviceResultEvent(): Create device result event.
2023-04-12 16:58:28 [547737588768] DEBUG - clientReadyHandler invoked
2023-04-12 16:58:28 [547737588768] INFO - try creating stream
2023-04-12 16:58:28 [547737588768] INFO - Creating Kinesis Video Stream CAM_1_stream
2023-04-12 16:58:28 [547737588768] INFO - createKinesisVideoStream(): Creating Kinesis Video Stream.
2023-04-12 16:58:28 [547737588768] INFO - logStreamInfo(): SDK version: e99e7e94d897f309c773cb3bb677cab5c9b5f587
2023-04-12 16:58:28 [547737588768] DEBUG - logStreamInfo(): Kinesis Video Stream Info
2023-04-12 16:58:28 [547737588768] DEBUG - logStreamInfo():     Stream name:CAM_1_stream 
2023-04-12 16:58:28 [547737588768] DEBUG - logStreamInfo():     Streaming type: STREAMING_TYPE_REALTIME 
2023-04-12 16:58:28 [547737588768] DEBUG - logStreamInfo():     Content type: video/h265 
2023-04-12 16:58:28 [547737588768] DEBUG - logStreamInfo():     Max latency (100ns): 600000000
2023-04-12 16:58:28 [547737588768] DEBUG - logStreamInfo():     Fragment duration (100ns): 20000000
2023-04-12 16:58:28 [547737588768] DEBUG - logStreamInfo():     Key frame fragmentation: Yes
2023-04-12 16:58:28 [547737588768] DEBUG - logStreamInfo():     Use frame timecode: Yes
2023-04-12 16:58:28 [547737588768] DEBUG - logStreamInfo():     Absolute frame timecode: Yes
2023-04-12 16:58:28 [547737588768] DEBUG - logStreamInfo():     Nal adaptation flags: 0
2023-04-12 16:58:28 [547737588768] DEBUG - logStreamInfo():     Average bandwith (bps): 4194304
2023-04-12 16:58:28 [547737588768] DEBUG - logStreamInfo():     Framerate: 30
2023-04-12 16:58:28 [547737588768] DEBUG - logStreamInfo():     Buffer duration (100ns): 1200000000
2023-04-12 16:58:28 [547737588768] DEBUG - logStreamInfo():     Replay duration (100ns): 400000000
2023-04-12 16:58:28 [547737588768] DEBUG - logStreamInfo():     Connection Staleness duration (100ns): 600000000
2023-04-12 16:58:28 [547737588768] DEBUG - logStreamInfo():     Store Pressure Policy: 1
2023-04-12 16:58:28 [547737588768] DEBUG - logStreamInfo():     View Overflow Policy: 1
2023-04-12 16:58:28 [547737588768] DEBUG - logStreamInfo():     Segment UUID: NULL
2023-04-12 16:58:28 [547737588768] DEBUG - logStreamInfo():     Frame ordering mode: 0
2023-04-12 16:58:28 [547737588768] DEBUG - logStreamInfo(): Track list
2023-04-12 16:58:28 [547737588768] DEBUG - logStreamInfo():     Track id: 1
2023-04-12 16:58:28 [547737588768] DEBUG - logStreamInfo():     Track name: kinesis_video
2023-04-12 16:58:28 [547737588768] DEBUG - logStreamInfo():     Codec id: V_MPEGH/ISO/HEVC
2023-04-12 16:58:28 [547737588768] DEBUG - logStreamInfo():     Track type: TRACK_INFO_TYPE_VIDEO
2023-04-12 16:58:28 [547737588768] DEBUG - logStreamInfo():     Track cpd: NULL
2023-04-12 16:58:28 [547737588768] INFO - writeHeaderCallback(): RequestId: 951f22bf-c01f-4cb0-9648-d84e9a600129
2023-04-12 16:58:28 [546454143472] DEBUG - describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.68053761733E9,"DataRetentionInHours":24,"DeviceName":"CAM_1","IngestionConfiguration":null,"KmsKeyId":"arn:*:alias/aws/kinesisvideo","MediaType":null,"Status":"ACTIVE","StreamARN":"arn:aws:*CAM_1_stream","StreamName":"CAM_1_stream","Version":"*"}}
2023-04-12 16:58:28 [546454143472] INFO - describeStreamResultEvent(): Describe stream result event.
2023-04-12 16:58:28 [546454143472] WARN - describeStreamResult(): Retention period returned from the DescribeStream call doesn't match the one specified in the StreamInfo
2023-04-12 16:58:28 [546454143472] WARN - describeStreamResult(): Content type returned from the DescribeStream call doesn't match the one specified in the StreamInfo
2023-04-12 16:58:28 [546454143472] INFO - writeHeaderCallback(): RequestId: 
2023-04-12 16:58:28 [546445750768] INFO - getStreamingEndpointResultEvent(): Get streaming endpoint result event.
2023-04-12 16:58:28 [546445750768] DEBUG - streamReadyHandler invoked
2023-04-12 16:58:28 [546445750768] Stream is ready
Opening in BLOCKING MODE 
0:00:00.486312395 17657   0x55b52a0cd0 WARN                    v4l2 gstv4l2object.c:2388:gst_v4l2_object_add_interlace_mode:0x55b5287d00 Failed to determine interlace mode
0:00:00.486400886 17657   0x55b52a0cd0 WARN                    v4l2 gstv4l2object.c:2388:gst_v4l2_object_add_interlace_mode:0x55b5287d00 Failed to determine interlace mode
0:00:00.486446147 17657   0x55b52a0cd0 WARN                    v4l2 gstv4l2object.c:2388:gst_v4l2_object_add_interlace_mode:0x55b5287d00 Failed to determine interlace mode
0:00:00.486495210 17657   0x55b52a0cd0 WARN                    v4l2 gstv4l2object.c:2388:gst_v4l2_object_add_interlace_mode:0x55b5287d00 Failed to determine interlace mode
0:00:00.486600212 17657   0x55b52a0cd0 WARN                    v4l2 gstv4l2object.c:4476:gst_v4l2_object_probe_caps:<nvv4l2h265enc0:src> Failed to probe pixel aspect ratio with VIDIOC_CROPCAP: Unknown error -1
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
0:00:00.489048060 17657   0x55b52a0cd0 WARN                     bin gstbin.c:2787:gst_bin_do_latency_func:<pipeline0> did not really configure latency of 0:00:00.000000000
0:00:00.489105978 17657   0x55b52820a0 FIXME                default gstutils.c:3981:gst_pad_create_stream_id_internal:<nvarguscamerasrc0:src> Creating random stream-id, consider implementing a deterministic way of creating a stream-id
New clock: GstSystemClock
NvMMLiteOpen : Block : BlockType = 8 
Redistribute latency...
===== NVMEDIA: NVENC =====
0:00:00.492740354 17657   0x55b52a0cd0 WARN                     bin gstbin.c:2787:gst_bin_do_latency_func:<pipeline0> did not really configure latency of 0:00:00.000000000
NvMMLiteBlockCreate : Block : BlockType = 8 
0:00:00.534902014 17657   0x55b52820a0 WARN          v4l2bufferpool gstv4l2bufferpool.c:1087:gst_v4l2_buffer_pool_start:<nvv4l2h265enc0:pool:src> Uncertain or not enough buffers, enabling copy threshold
GST_ARGUS: Creating output stream
CONSUMER: Waiting until producer is connected...
GST_ARGUS: Available Sensor modes :
GST_ARGUS: 4032 x 3040 FR = 29,999999 fps Duration = 33333334 ; Analog Gain range min 1,000000, max 22,250000; Exposure Range min 13000, max 683709000;

[... 69 lines cut ...]

nvv4l2h265enc insert-sps-pps=1 insert-aud=1 insert-vui=1 idrinterval=15  ! h265parse config-interval=1 !

**To Reproduce**
Steps to reproduce the behavior:

run gstreamer command: 
GST_DEBUG=3 gst-launch-1.0 nvarguscamerasrc ! 'video/x-raw(memory:NVMM),width=4032, height=3040, framerate=29/1, format=NV12' ! nvvidconv  ! nvv4l2h265enc  ! h265parse config-interval=1 ! 'video/x-h265, stream-format=(string)byte-stream' ! kvssink stream-name='CAM_1_stream' content-type='video/h265' codec-id='V_MPEGH/ISO/HEVC' storage-size=1028 framerate=30 ..[credential properties]

I also tried adding:  insert-sps-pps=1 insert-aud=1 insert-vui=1 idrinterval=15

Thanks in advance! 

**Expected behavior**
Showing h265 video encoded content in the console..

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Platform (please complete the following information):**
 - OS: Linux
 - Processor: Nvidia Jetson Nano Developmentkit 4GB 
 - Version 18.04
- GStreamer Core Library version 1.14.5
- IMX477 camera

benwkim, 2023-04-14:
You can view the CPD used by kvssink by using GST_DEBUG=4 instead of GST_DEBUG=3.  You should be able to see a caps event like the following:
```
0:00:06.141501000 18781 0x6000026082d0 INFO               GST_EVENT gstevent.c:918:gst_event_new_caps: creating caps event video/x-h265, stream-format=(string)hvc1, alignment=(string)au, codec_data=(buffer)0101600000009000000000005af000fcfdf8f800000f03200001001840010c01ffff01600000030090000003000003005a959809210001002e42010101600000030090000003000003005aa0050201e165959a4932bffc0004000404000003000400000300782022000100074401c172b42240, width=(int)640, height=(int)480, framerate=(fraction)30/1, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true, pixel-aspect-ratio=(fraction)1/1, profile=(string)main, tier=(string)main, level=(string)3
```

The CPD is the `codec_data`.  Can you re-run your pipeline and see if you can get the `codec_data` provided in the caps event?

quintene, 2023-04-18:
> [benwkim](/benwkim)

Thanks benwkim, seems like nvv4l2h265enc does not sends the codec_data property, eventually I used the omxh265enc element in gstreamer which does work!

closed 2023-04-18 by quintene