awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #1010: Not working with AXIS IP Camera

openedhardware, 2023-05-03:
I tried to stream video from my IP camera to the AWS KVS, but got this error: https://pastebin.pl/view/132d32f3

Could you guys help me figure out why this doesn't work?

Sees 1280x960 resolution is not supported?

niyatim23, 2023-05-03:
Hi @openedhardware, can you please attach the log file instead of this link as I can't access it?

openedhardware, 2023-05-03:
Sure, here you go - 
[kvs_log.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/11390002/kvs_log.txt)

Thank you!

niyatim23, 2023-05-03:
Are we talking about the `Bus error`? The log seems to be cut midway. Can you please upload the complete logs? 
Thanks!

openedhardware, 2023-05-03:
Yes, that is what I see as the last reply:
[image]

Thank you!

sirknightj, 2023-05-03:
1280x960 is supported.
```
gst-inspect-1.0 kvssink
...
video/x-h264
          stream-format: avc
              alignment: au
                  width: [ 16, 2147483647 ]
                 height: [ 16, 2147483647 ]
      video/x-h265
              alignment: au
                  width: [ 16, 2147483647 ]
                 height: [ 16, 2147483647 ]
```

Is #686 related?

openedhardware, 2023-05-03:
I tried with `export DALIGNED_MEMORY_MODEL=TRUE`, but same result... :(

sirknightj, 2023-05-03:
It's a cmake flag, so you need to pass it into cmake:

```bash
cmake .. -DALIGNED_MEMORY_MODEL=TRUE <other cmake flags>
```

Then run `make` again.

sirknightj, 2023-05-03:
May be safest to delete the `build` folder and do a fresh build in case of cmake cache not taking the updated values.

openedhardware, 2023-05-03:
That fixed, thank you!

closed 2023-05-03 by openedhardware