awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #1018: [BUG] OSX M1 Build not working

ageekymonk, 2023-05-26:
**Logging**
Add relevent SDK logging.  IMPORTANT NOTE:  Please make sure to NOT share AWS access credentials under any circumstance!  Please make sure they are not in the logs.

ec2-user@ip-10-22-226-58 build % cmake .. -DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_JNI=TRUE
ec2-user@ip-10-22-226-58 build % rm * -rf
zsh: sure you want to delete all 3 files in /Users/ec2-user/amazon-kinesis-video-streams-producer-sdk-cpp/build [yn]? y
rm: CMakeFiles: is a directory
rm: dependency: is a directory
rm: -rf: No such file or directory
ec2-user@ip-10-22-226-58 build % 
ec2-user@ip-10-22-226-58 build % cmake .. -DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_JNI=TRUE
-- The C compiler identification is AppleClang 14.0.0.14000029
-- The CXX compiler identification is AppleClang 14.0.0.14000029
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Kinesis Video Cpp Producer path is /Users/ec2-user/amazon-kinesis-video-streams-producer-sdk-cpp
-- dependencies install path is /Users/ec2-user/amazon-kinesis-video-streams-producer-sdk-cpp/open-source
-- Begin building dependencies.
-- autoconf already built
-- automake already built
-- log4cplus already built
-- Finished building dependencies.
-- Configuring done (0.0s)
-- Generating done (0.0s)
-- Build files have been written to: /Users/ec2-user/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer
[ 11%] Performing update step for 'libkvscproducer-download'
[ 22%] No patch step for 'libkvscproducer-download'
[ 33%] No configure step for 'libkvscproducer-download'
[ 44%] No build step for 'libkvscproducer-download'
[ 55%] No install step for 'libkvscproducer-download'
[ 66%] No test step for 'libkvscproducer-download'
[ 77%] Completed 'libkvscproducer-download'
[100%] Built target libkvscproducer-download
-- Kinesis Video Producer path is /Users/ec2-user/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src
-- Begin building dependencies.
-- Configuring done (0.0s)
-- Generating done (0.0s)
-- Build files have been written to: /Users/ec2-user/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/libopenssl
[ 11%] Creating directories for 'project_libopenssl'
[ 22%] Performing download step (git clone) for 'project_libopenssl'
Cloning into 'project_libopenssl'...
HEAD is now at e2e09d9fba Prepare for 1.1.1g release
Submodule 'boringssl' (https://boringssl.googlesource.com/boringssl) registered for path 'boringssl'
Submodule 'krb5' (https://github.com/krb5/krb5) registered for path 'krb5'
Submodule 'pyca.cryptography' (https://github.com/pyca/cryptography.git) registered for path 'pyca-cryptography'
Cloning into '/Users/ec2-user/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/libopenssl/build/src/project_libopenssl/boringssl'...
Cloning into '/Users/ec2-user/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/libopenssl/build/src/project_libopenssl/krb5'...
Cloning into '/Users/ec2-user/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/libopenssl/build/src/project_libopenssl/pyca-cryptography'...
Submodule path 'boringssl': checked out '2070f8ad9151dc8f3a73bffaa146b5e6937a583f'
Submodule path 'krb5': checked out '890ca2f401924cdcb88f493950b04700bbe52db3'
Submodule path 'pyca-cryptography': checked out '09403100de2f6f1cdd0d484dcb8e620f1c335c8f'
[ 33%] Performing update step for 'project_libopenssl'
[ 44%] No patch step for 'project_libopenssl'
[ 55%] Performing configure step for 'project_libopenssl'
Operating system: i686-apple-darwinDarwin Kernel Version 22.3.0: Mon Jan 30 20:39:35 PST 2023; root:xnu-8792.81.3~2/RELEASE_ARM64_T8103
Configuring OpenSSL version 1.1.1g (0x1010107fL) for darwin-i386-cc
Using os-specific seed configuration
Creating configdata.pm
Creating Makefile

**********************************************************************
***                                                                ***
[... 130 lines cut ...]
make[3]: *** [all] Error 2
make[2]: *** [build/src/project_libopenssl-stamp/project_libopenssl-build] Error 2
make[1]: *** [CMakeFiles/project_libopenssl.dir/all] Error 2
make: *** [all] Error 2
CMake Error at dependency/libkvscproducer/kvscproducer-src/CMake/Utilities.cmake:93 (message):
  CMake step for libopenssl failed: 2
Call Stack (most recent call first):
  dependency/libkvscproducer/kvscproducer-src/CMakeLists.txt:72 (build_dependency)

-- Configuring incomplete, errors occurred!

**Describe the bug**
when cmake is triggered with
cmake .. -DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_JNI=TRUE

we get the error shown above

**To Reproduce**
1. clone the repo
2. install all brew dependencies specified. 
3. cmake

**Platform (please complete the following information):**
 - OS: MAC
 - Processor: ARM

sirknightj, 2023-05-27:
I noticed that you only deleted the `build` folder. Please also remove the `open-source` and `dependency` folders (located in the project root directory) before running cmake again.

ageekymonk, 2023-05-30:
After removing open-source and dependency folders and rebuilding again, it worked. Thanks for your support

closed 2023-05-30 by hassanctech